wills106/homeassistant-solax-modbus #967: [Bug]: no entities after update

MenosChris, 2024-07-25:
### Describe the bug

All entities gone after update....
Such a mess...

### Integration Version

2023097

### Homeassistant core version

2024.7.3

### Inverter brand

SOLAX power

### Plugin used

plugin_solax.py

### Serial prefix

H34a15

### Inverter firmware versions

ARM131 dsp133

### Connection Method

Solax wifi 3

### Dongle firmware

301016

### Detailed Error Log

No entities 

### Additional context

After update

wills106, 2024-08-01:
Have a look at #872

closed 2024-08-01 by wills106

MenosChris, 2024-08-01:
Hi wills106, that did not help.
BUT here is the ultimate solution which always works:  (Video in German only, unfortunately for the english speaking guy)

1) Youtube video: https://youtu.be/hp5TRVzfUYw?si=yx58RToJ8IbEVjlg ( Video in German only, unfortunately for the english speaking guy )

2) Install modbuy proxy: https://github.com/Akulatraxas/ha-modbusproxy

3) Solution also works for SOLAX.

4) From minute 4:31- 5:23 --> I ignored that, since I do not have these settings.

5 ) Important:
In case you use modbus and evcc i.a. , than you have 2 instances grabbing data from modbus. Modbus accepts only 1 instance.
So the proxy grabbs the data.
Make sure you configure your modbus and evcc ( config.yaml) IP adresses, like the ip address of your home assistant instance.

Then restart evrything. Just follow the video. Headtip to [smart-live.net](https://www.youtube.com/@smartlivenet)  for this solution.

MenosChris, 2024-08-01:
Please make this post available in reopening the issue. Thank you.