wills106/homeassistant-solax-modbus #968: [Bug]: 'Growatt' - 'SPF5000EU' crash on add integration "KeyError: 'support-battery'"

Evka2k, 2024-07-25:
### Describe the bug

Add integration -> Select Interface:Serial, Inverter type: growatt or any other -> Serial Interface Parameters: Submit

[image]

<pre>
home-assistant  |     next_step_id_or_end_flow = await form_step.next_step(self._options)
home-assistant  |                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
home-assistant  |   File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
home-assistant  |     if user_input["support-battery"]:
home-assistant  |        ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
home-assistant  | KeyError: 'support-battery'
</pre>

### Integration Version

2024.07.0b3

### Homeassistant core version

2024.7.3

### Inverter brand

growatt

### Plugin used

plugin_growatt.py

### Serial prefix

113.02

### Connection Method

RS485-USB

### Detailed Error Log

[crash.log](https://github.com/user-attachments/files/16382329/crash.log)

### Additional context

This fix prevents from crash.

<pre>
--- custom_components/solax_modbus/config_flow.py
+++ custom_components/solax_modbus/config_flow.py
@@ -190,6 +190,8 @@ async def _next_step_modbus(user_input: Any) -> str:
 
 async def _next_step_battery(user_input: Any) -> str:
     _LOGGER.debug(f"_next_step_battery: returning data: {user_input}")
+    if "support-battery" not in user_input.keys():
+        return None
     if user_input["support-battery"]:
         return "battery"
     return None
</pre>

Beunhaaz, 2024-08-03:
I'm having the same problem on my Growatt SPH8000 but unfortunately your fix didn't work for me.

Now I get:
`  File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
    if "support-battery" not in user_input.keys():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'support-battery'`

It seems to trip over the first support-battery it encounters.

MrRickwi, 2024-08-04:
Same for me with a Solax Inverter. Updated to latest version and lost battery level. Therefore I deinstalled the complete integration, rebooted HA and tried to install again - with the same arror as above. Will try the fix now.

MrRickwi, 2024-08-04:
File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
    if user_input["support-battery"]:
       ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 'support-battery'

MrRickwi, 2024-08-04:
Fix doesn't help for me, same as Beunhaaz reported

MrRickwi, 2024-08-04:
Got it up and running for me with this hack:

    user_input["support-battery"] = true # plugin.plugin_instance.BATTERY_CONFIG is not None

This of course is only valid inverter supports a battery

MrRickwi, 2024-08-04:
After getting the fix up and running I'm missing some entities:

sensor.solax_battery_input_energy_total
sensor.solax_battery_output_energy_total
sensor.solax.grid_status

Please make them also available again. Especially the total energies are crucial for the energy dashboard to work.

MrRickwi, 2024-08-04:
X3-Hybrid
von SolaX Power
Firmware: DSP v1.29 ARM v1.31
Hardware: Gen4

siebeclaes, 2024-08-04:
Untested but a suggestion to try:

at "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery

replace
`if user_input["support-battery"]:`
With
`if user_input.get("support-battery", False):`

MrRickwi, 2024-08-05:
Latest version is working for me. Missing entities have been found, are disabled by default.

wills106, 2024-09-07:
Can this be closed down now?

wills106, 2024-09-15:
Closing down as no further response.
If there are still issues either reopen or raise a new issue.

closed 2024-09-15 by wills106

noessi5, 2024-09-15:
Just wanted to let you know that the issue is not fixed in the latest version.
When looking at the code, I can see that the "support-battery" setting is only set up in the `_validate_host` function (which is only called when setting up the connection via TCP). When setting up an inverter-connection via serial, the variable is never set, so it should always fail at the next step.

The change that @siebeclaes suggested should be applied to master in my opinion to fix this.

noessi5, 2024-09-15:
Opened a PR now with this change: https://github.com/wills106/homeassistant-solax-modbus/pull/1054

wills106, 2024-09-17:
Just released [2024.09.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.3) with this change.