wills106/homeassistant-solax-modbus #976: [Bug]: 'SolaX' - 'Gen3' crash on add integration "KeyError: 'support-battery'"

MrRickwi, 2024-08-04:
### Describe the bug

Same like Growatt bug. Reported here to have different label on it. Additionally, some entities are missing after getting the integration up and running with a hack. See also other reported bug.

### Integration Version

2024.08.02

### Homeassistant core version

2024.7.4

### Inverter brand

SolaX Gen3

### Plugin used

plugin_solax.py

### Serial prefix

H34C06

### Inverter firmware versions

DSP v1.29 ARM v1.31

### Connection Method

USB Modbus Adapter

### Dongle firmware

None

MrRickwi, 2024-08-05:
Latest version is working for me. Missing entities have been found, are disabled by default.

wills106, 2024-08-06:
I'll close this down then.

closed 2024-08-06 by wills106

ruimmsimoes, 2024-08-10:
I’m having the same issue when adding a SolaX inverter after upgrading to 2024.08.08

wills106, 2024-09-17:
Just released [2024.09.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.3) which hopefully fixes this error.