wills106/homeassistant-solax-modbus #977: After upgrade to new version - unable to add invterter - unknown error

perjmolsen, 2024-08-04:
### Describe the bug

Hi
I have upgraded to the newest version  of SolarX modbus, now I am unable to add my invterter.
When I try to add it, I get "unknown" error, I am adding the inverter withh the same information as earlier.

mic3000tl-xe
/dev/ttyUSB0
9600

I have removed the old invterter - rebooted HA og redownloaded the newest add on - still the same issue

Please help

### Integration Version

2024.8.6

### Homeassistant core version

2024.7.4

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

QUH7CEQ0CW

### Connection Method

Serial

### Detailed Error Log

No logs available 
[image]

FloCodeRian, 2024-08-04:
some here - other inverter brand but some method used ... usb to serial

perjmolsen, 2024-08-04:
How did you solve it?

wuschi1983, 2024-08-05:
Same problem here

granthgh, 2024-08-06:
> ### Describe the bug
> Hi I have upgraded to the newest version of SolarX modbus, now I am unable to add my invterter.
> 
> When I try to add it, I get "unknown error," I am adding the inverter withh the same information as earlier.
> 
> mic3000tl-xe /dev/ttyUSB0 9600
> 
> I have removed the old invterter - rebooted HA og redownloaded the newest add on - still the same issue
> 
> Please help
> 
> ### Integration Version
> 2024.8.6
> 
> ### Homeassistant core version
> 2024.7.4
> 
> ### Inverter brand
> Growatt
> 
> ### Plugin used
> plugin_growatt.py
> 
> ### Serial prefix
> QUH7CEQ0CW
> 
> ### Connection Method
> Serial
> 
> ### Detailed Error Log
> No logs available ![error](https://private-user-images.githubusercontent.com/126662740/354911936-714c3f54-35f3-47c5-9436-4ce594776cf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI5MzUxMjYsIm5iZiI6MTcyMjkzNDgyNiwicGF0aCI6Ii8xMjY2NjI3NDAvMzU0OTExOTM2LTcxNGMzZjU0LTM1ZjMtNDdjNS05NDM2LTRjZTU5NDc3NmNmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNlQwOTAwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMWVhOTRiYjI2ODNjMzY3YjUwMGMxYTIwN2I2ZWIwNTZlM2QyODcyMGYzNDMzMmMzZjgyMjYxODBkNWFiMGZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5lJQhykYEprau8S89wZPg9j0QOGoYS-gykUBFbCSrOE)
> 
> ### Additional context
> _No response_

Same problem here with Growatt SPH6000 .

FloCodeRian, 2024-08-06:
Workaround would be - downgrading via zip download of version 2024.05.07 

https://github.com/wills106/homeassistant-solax-modbus/tree/2024.05.7

mfcltd, 2024-08-06:
Same problem too.

rhlavienka, 2024-08-06:
the same here
Solax G4

granthgh, 2024-08-06:
> Workaround would be - downgrading via zip download of version 2024.05.07
> 
> https://github.com/wills106/homeassistant-solax-modbus/tree/2024.05.7
‐------------------------

I've tested downgrading to this version and it does get round the issue.

wills106, 2024-08-06:
@perjmolsen 

You can get more detailed error logs following:
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log

@Zvolli 

Do you get "unknown error occurred" when adding your Inverter, or just your entities have gone to unknown since updating to 2024.08.x?

For everyone else who's saying "same here" can you clarify what the actual issue is.

Either
"unknown error occurred" like @perjmolsen 
Or all entities are showing "unknown"

granthgh, 2024-08-06:
My original issue was that after upgrading to 2024.8.6 some of the entities became unknown.
I then deleted the inverter and tried re-adding it, but when I got to the box with the COM port / data rate I got the unknown error when actioning it.
Since then I've downgraded to 2024.05.07 I was able to add the inverter and the entities are now working. 
There are no errors in the logs.

Zvolli, 2024-08-07:
@***@***.***> After updating to 2024.08.x all entities went to unknown state. If i delete inverter and solax, reinstall I can add my inverter with success but of course with unknown state for all entities.

https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log

rhlavienka, 2024-08-07:
@wills106 
I'm sorry, I thought all of us have the same issue.
I removed inverter from integration and after upgrade to 2024.8.7 
is not possible add it back.
I get "unknown error occurred" when adding Inverter.

Zvolli, 2024-08-07:
Sure but downgrading in docker container is for me a unknown territory 🙂

https://github.com/wills106/homeassistant-solax-modbus/tree/2024.05.7
‐------------------------

mfcltd, 2024-08-07:
> Same problem too.

@wills106 - When re-adding the inverter after upgrading, the unknown error occurs.

HaareistNase, 2024-08-07:
Seems to have something to do with the MIC

After the Upgrade I can add my SPH4000 without problem, but my Mic3000 shows "unknown" on all values EXEPT the Firmware-Version^^

Mike5229, 2024-08-08:
> > Workaround would be - downgrading via zip download of version 2024.05.07
> > https://github.com/wills106/homeassistant-solax-modbus/tree/2024.05.7
> > ‐------------------------
> 
> I've tested downgrading to this version and it does get round the issue.

Hi, 

can you please briefly explain how to perform the downgrade?

toszym, 2024-08-08:
I can confirm, downgrading to version 2024.05.07 works.
Steps to follow: remove current version from HACS and then follow manual installation steps ( download tar.gz or zip archive, upload to homeassistant, unzip and move solax_modbus folder to /homeassistant/custom_components/)

Zvolli, 2024-08-08:
I can also confirm 2024.05.7 works but by doing this maneuver it does not appear in the HACS section.
In my case it's in Synology docker container.

rhlavienka, 2024-08-08:
I waited 2 days if by chance updating the entire HA doesn't solve the problem. No it didn't.
So I also downgraded.

Using the File Editor, I deleted the content of the folder /custom_components/solax_modbus and replaced it with the content from version 2024.5.7. I can't delete folder or multiple files at once in the File Editor. It is necessary individually, but there is not much of it.

See Manual installation
https://homeassistant-solax-modbus.readthedocs.io/en/latest/installation/

shaarkys, 2024-08-08:
Same here, trying to add Solax X3 G4 after removal due to the unknown entities and duplicated inverters (even I do have only one) and can't get over this : 
[image]

Time to manual downgrade and re-adding back inverter.

ruimmsimoes, 2024-08-10:
I have a SolaX inverter and I’m having the same issue.
I upgraded to version 2024.08.08.
When adding the inverter after the upgrade I’m also getting the same error message.
In the logs I am seeing an error message that is the same as the one reported in this 2 issues:
#988 - Solax X3 Hybrid Raspberry RS485 USB -> Unknown error occurred
#976 - 'SolaX' - 'Gen3' crash on add integration "KeyError: 'support-battery'"

ralfna, 2024-08-12:
Same issue with Qcells/Solax, in addition to downgrade I have to reset the WiFi dongle to get the integration back to work.

martinpichlo, 2024-08-14:
> I can confirm, downgrading to version 2024.05.07 works. Steps to follow: remove current version from HACS and then follow manual installation steps ( download tar.gz or zip archive, upload to homeassistant, unzip and move solax_modbus folder to /homeassistant/custom_components/)

Is the downgrading also working on HA 2024.8.1? I don't get it working. I'm able to select the integration. But instead of "unknown error" a got an spinning waiting icon all the time.

[image]

@wills106 : How could we all support, find the problem?

wills106, 2024-08-24:
I have released [2024.08.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.11) it may fix this error, but no error logs in this issue thread so can't check.

granthgh, 2024-08-24:
> I have released [2024.08.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.11) it may fix this error, but no error logs in this issue thread so can't check.

Still got the same unknown error has occurred, with 2924.08.11. Attached the only entries in the logs.
[image]
[image]
[image]

wills106, 2024-08-25:
@perjmolsen I don't know if [2024.08.12](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.12) will help with your issue, but it should resolve @granthgh issue. Where it can't read "Inverter Module"

wills106, 2024-09-07:
Has this been resolved so I can close this issue down?

granthgh, 2024-09-07:
Issue still exists, but workaround, setting up as tcp and then changing to serial, gets round the issue.

wills106, 2024-09-07:
I didn't introduce the serial support, are you able to try and fix the issue?
I don't use or own any RS485 equipment, so going to be difficult to fix from my end.

granthgh, 2024-09-08:
The issue is that there's no errors recorded, so it needs someone with an offline system, as I need mine up and running to manage my inverter. I presume there have been significant changes since the working 2024.7, so difficult to identify the changes to the code that broke it. As there's a workaround then I think it can be put on the back burner  and If you want I'm happy to look at the code to see if I can spot an issue.

wills106, 2024-09-17:
Just released [2024.09.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.3) which hopefully fixes this error.

rafalkur, 2024-09-22:
Hello, 
Im using 2024.09.4 and trying to configure it with my growatt (via waveshare eth converter) for the first time. Initialization phase last forever, after configuring again with same settings im getting success box, but device is not added.
in logs:
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.051 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 73, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.052 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 134, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.052 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 109, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.053 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 136, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.053 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 184, in async_unload_entry
    hass.data[DOMAIN].pop(entry.options["name"])
KeyError: 'growatt'
2024-09-22 21:55:25.654 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 64, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=6)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 546, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 6 retries
2024-09-22 21:55:25.655 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: reading firmware number from address 0xbb9 failed; other address may succeed
2024-09-22 21:55:25.655 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read growatt 0xbb9 firmware number before potential swap: None
2024-09-22 21:55:25.655 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: cannot find firmware version, even not for other Inverter
2024-09-22 21:55:25.655 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : unknown
2024-09-22 21:55:25.655 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2024-09-22 21:55:35.656 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: trying to determine inverter type

closed 2025-10-03 by github-actions[bot]