wills106/homeassistant-solax-modbus #980: Entities now unknown after update to 2024.08.all[Bug]:

mnh637, 2024-08-06:
### Describe the bug

Hi,
Solax X1 Boost Gen3 (2 inverters) after update to 2024.08. all versions (on 08.06 now)
these two entities I was using so I know they worked before, full list shown in additional. Most work ok :-)
SolaX-Garage Inverter Inverter Frequency
Unknown
SolaX-Garage Inverter Inverter Voltage
Unknown
Thanks
Mark

### Integration Version

2024.08.06

### Homeassistant core version

2024.7.4

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

XB3350

### Inverter firmware versions

Firmware: DSP v2.231 ARM v1.148

### Connection Method

WIFI 3.0

### Dongle firmware

3.001.02

### Additional context

See Below all that work (or don't)
SolaX-Garage Inverter Active Power Limit
0 W
SolaX-Garage Inverter Grid Export
1,483 W
SolaX-Garage Inverter Grid Import
0 W
SolaX-Garage Inverter Inverter Current
5.9 A
SolaX-Garage Inverter Inverter Frequency
Unknown
SolaX-Garage Inverter Inverter Power
Unknown
SolaX-Garage Inverter Inverter Voltage
Unknown
SolaX-Garage Inverter Measured Power
1,483 W
SolaX-Garage Inverter Measured Power 2
Unknown
SolaX-Garage Inverter PV Current 1
3.4 A
SolaX-Garage Inverter PV Current 2
3.0 A
SolaX-Garage Inverter PV Power 1
787 W
SolaX-Garage Inverter PV Power 2
714 W
SolaX-Garage Inverter PV Power Total
1,501 W
SolaX-Garage Inverter PV Total Power
1,501 W
SolaX-Garage Inverter PV Voltage 1
230.9 V
SolaX-Garage Inverter PV Voltage 2
236.5 V
SolaX-Garage Inverter Run Mode
Normal Mode
SolaX-Garage Inverter Today's Yield
3.7 kWh
SolaX-Garage Inverter Total Grid Export
Unknown
SolaX-Garage Inverter Total Grid Import
Unknown
SolaX-Garage Inverter Total Yield
6,556.3 kWh

alna-laurentiu, 2024-08-06:
Same for me.

2024.08.6
X3-Hybrid
Firmware: DSP v1.39 ARM v1.38
Hardware: Gen4

MJP-76, 2024-08-06:
SK-SU5000E inverter

Almost all my sensors now have "Inverter Inverter Inverter Inverter Inverter" in their name. Yes 5 times

Only cosmetic as the actual sensor is correct

[image]

MJP-76, 2024-08-06:
There is a change to some of the sensor

Today's Solar Energy
Was: sensor.solax_today_s_solar_energy
Now: sensor.solax_today_solar_energy

Total Solar Energy
Was: sensor.solax_total_solar_energy
Now: solax_total_solar_yield

Hope it helps

mnh637, 2024-08-06:
I deleted the inverter as the instructions before and then reloaded, the sensor names seem to be correct just no data.
sensor.solax_garage_inverter_voltage
 it just shows unknown
Regards
Mark

wills106, 2024-08-06:
> There is a change to some of the sensors
    Today's Solar Energy
    Total Solar Energy

@MJP-76 corrected in [2024.08.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.7)

MJP-76, 2024-08-06:
@wills106 

I can confirm the "multiple" inverter in the names is gone and "Today's Solar Energy" & "Total Solar Energy" are back after updating to 2024.08.07 - Cheers

When you have time, Can you look at the last missing sensor I have
sensor.solax_charger_use_mode

mnh637, 2024-08-08:
Hi,
I've updated to 2024.08.08 the values below are still unknown
SolaX-Garage Inverter Inverter Frequency
Unknown
SolaX-Garage Inverter Inverter Voltage
Unknown
Thanks
Mark

mnh637, 2024-08-08:
Sorry, closed by mistake, now reopened. Ta.

Kars-de-Jong, 2024-08-19:
Ah, same for me (SolaX X3 MIC MU802T).

mnh637, 2024-08-20:
Updated to 2024.08.9 no change to unknown entities.
Regards
Mark

mnh637, 2024-08-24:
Updated to 2024.08.11 no change to unknown entities.
Regards
Mark

mnh637, 2024-08-29:
Updated to 2024.08.13 no change to unknown entities.
Regards
Mark

KestasG, 2024-09-03:
Same issue here with MIC-X3. Debugged modpy and see data being read succesfully:
`
2024-09-03 10:56:52.825 DEBUG (MainThread) [pymodbus.logging] Adding transaction 25
2024-09-03 10:56:52.825 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 
2024-09-03 10:56:52.825 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x19 0x0 0x0 0x0 0x6 0x1 0x4 0x4 0x0 0x0 0x27
2024-09-03 10:56:52.933 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x19 0x0 0x0 0x0 0x51 0x1 0x4 0x4e 0xe 0x5a 0xe 0x2c 0x0 0x4d 0x0 0x4f 0x9 0x53 0x9 0x4d 0x9 0x71 0x13 0x8a 0x13 0x8a 0x13 0x8c 0x0 0x46 0x0 0x47 0x0 0x49 0x0 0x2f 0x15 0xc 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0x2d 0xb 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4a 0x48 0x0 0x2 0x0 0x5b 0x0 0x0 old_data:  addr=None
2024-09-03 10:56:52.933 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x19 0x0 0x0 0x0 0x51 0x1 0x4 0x4e 0xe 0x5a 0xe 0x2c 0x0 0x4d 0x0 0x4f 0x9 0x53 0x9 0x4d 0x9 0x71 0x13 0x8a 0x13 0x8a 0x13 0x8c 0x0 0x46 0x0 0x47 0x0 0x49 0x0 0x2f 0x15 0xc 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0x2d 0xb 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4a 0x48 0x0 0x2 0x0 0x5b 0x0 0x0
2024-09-03 10:56:52.933 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadInputRegistersResponse': 4]
2024-09-03 10:56:52.933 DEBUG (MainThread) [pymodbus.logging] Getting transaction 25
2024-09-03 10:56:52.934 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-09-03 10:56:52.934 DEBUG (MainThread) [pymodbus.logging] [3674, 3628, 77, 79, 2387, 2381, 2417, 5002, 5002, 5004, 70, 71, 73, 47, 5388, 2, 0, 0, 0, 0, 2861, 2871, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 19016, 2, 91, 0]
2024-09-03 10:56:52.934 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x02', b'JH']
2024-09-03 10:56:52.934 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00[']
`
Voltage is 238.7 238.1 241.7 and frequency 50.02 50.02 50.04

KestasG, 2024-09-03:
Dig a bit seeper into this. inverter_voltage and inverter_frequency has  sleepmode = SLEEPMODE_LASTAWAKE. Because of  this, additional check performed before using the value
` self.plugin.isAwake`
which is doing
`return (datadict.get('run_mode', None) == 'Normal Mode')`
the problem here is that run_mode register is processed after voltage and frequency, and run_mode is None until this. This is visible in additonal logging I've added:
2024-09-03 14:32:29.122 DEBUG (MainThread) [custom_components.solax_modbus] SolaX modbus input block start: 0x400 end: 0x427  len: 39 
regs: [1024, 1025, 1026, 1027, 1028, 1029, 1030, 1031, 1032, 1033, 1034, 1035, 1036, 1037, 1038, 1039, 1044, 1045, 1059, 1061]
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x400 : pv_voltage_1
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] Read raw value : 3531
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] Updating data!
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] Expiry data: None
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] descr.sleepmode: 1
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: None
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] self.plugin.isAwake: False
2024-09-03 14:32:29.232 DEBUG (MainThread) [custom_components.solax_modbus] Read return_value : 353.1
.........................................
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus] Read raw value : 2377
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: None
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus] Expiry data: None
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus] descr.sleepmode: 2
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: None
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus] self.plugin.isAwake: False
2024-09-03 14:32:29.233 DEBUG (MainThread) [custom_components.solax_modbus] Read return_value : 237.7
.................................

2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: None
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] self.plugin.isAwake: False
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read return_value : 49
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x40e : measured_power
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read raw value : 5648
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Updating data!
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Expiry data: None
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] descr.sleepmode: 1
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: None
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] self.plugin.isAwake: False
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read return_value : 5648
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x40f : run_mode
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read raw value : 2
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Updating data!
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Expiry data: None
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] descr.sleepmode: 1
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: Normal Mode
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] self.plugin.isAwake: True
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read return_value : Normal Mode
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] skipping bytes 8
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x414 : pv_power_1
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Read raw value : 3012
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Updating data!
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] Expiry data: None
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus] descr.sleepmode: 1
2024-09-03 14:32:29.235 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] run mode from isAwake: Normal Mode

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this change.

mnh637, 2024-09-06:
Hi,
This has restored the missing voltage and frequency.
I'm happy that this is fixed.
Many Thanks
mark

closed 2024-09-06 by mnh637