wills106/homeassistant-solax-modbus #987: [Bug]: Entity names are “SolaX Inverter Inverter Inverter Inverter Inverter …”

freddanastrom, 2024-08-07:
### Describe the bug

After updating to 2024.08.7 all entities except the type sensor have the  “SolaX Inverter Inverter Inverter Inverter Inverter …” in the friendly names. Entity id’s seems to be correct.

All “sensor.” has the correct names.
“number.”, “button.”, “input.” etc has the inverter inverter inverter included in the names.

A lot of inverter inverter:
[image]

Example of entity with correct id, but wrong friendly name:
[image]

### Integration Version

2024.08.7

### Homeassistant core version

2024.7.4

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H3BC25

### Inverter firmware versions

DSP 20 ARM 15.14

### Connection Method

Waveshare

### Dongle firmware

V1.452

wills106, 2024-08-07:
Do you have multiple Inverters setup. Or just the one?

I know if you have two Inverters, the second one get's a single Inverter added to the front of the name.

But not experienced this on a single setup.

First mention was https://github.com/wills106/homeassistant-solax-modbus/issues/980#issuecomment-2271260169 but then the issue went https://github.com/wills106/homeassistant-solax-modbus/issues/980#issuecomment-2271832361

freddanastrom, 2024-08-07:
Just one single inverter.
I also removed the integration prior to updating, as recommended in the release notes.

freddanastrom, 2024-08-07:
Just noticed that 2024.08.8 was released while I was filling the form and updating to this version seems to solve this naming issue.

LubikR, 2024-08-07:
@wills106 , @freddanastrom , I have tested this with new version 2024.08.8 and don't think that this issue is resolved. If you want add "Inverter" to entity names just reload the integration. Everytime you reload integration, new text "Inverter" is added

wills106, 2024-08-07:
There wasn't any change between 2024.08.1 to 2024.08.8 to alter this behaviour.

So I think restarting HA reset it back to no added "Inverter"

I'll have a proper look into this over the next few days.

freddanastrom, 2024-08-07:
Sorry for closing the issue before making sure updating was the solution.

I tried to reloading the integration a couple of times without any naming issues.

However, when editing the configuration I erased the name suffix option and saved it blank.
Suffix was still there when looking at the entity names.
Editing the settings once again and adding a suffix does as expected add a the suffix to the name, but also keeps the previous.
Repeat and get more added to the entity names.

wills106, 2024-08-07:
> Sorry for closing the issue before making sure updating was the solution.

It's ok.

> However, when editing the configuration I erased the name suffix option and saved it blank.
> Suffix was still there when looking at the entity names.
> Editing the settings once again and adding a [...]

That's handy to know then, when editing the config it seems to add all of the duplicates?

MiJo112, 2024-08-14:
I still have the same bug with 2024.08.8. Every time I change the name, it is only added, the old one is not deleted.  I would like to do without the name Inverter completely
[image]

closed 2025-10-03 by github-actions[bot]