wills106/homeassistant-solax-modbus #992: [Bug]: upgraded to 2024.08.8 Now unable to control inverter through modbus

UK-JustinSmall, 2024-08-09:
### Describe the bug

I have problem with inverter not saving values when they are updated through HA

Values change but inverter does not take action

E.g. I change 
select.solax_charger_use_mode to "Manual Mode" and

select.solax_manual_mode_select to "Force Discharge"

I expect the battery to force discharge, but nothing happens, the values change in the UI, but battery does not force discharge!!!

I have waveshare Modbus connection to Solax Inverter and can see the values of the entities OK.

### Integration Version

2024.08.8

### Homeassistant core version

2024.8.0

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H4752A

### Inverter firmware versions

DSP v1.42 ARM v1.39

### Connection Method

Waveshare Adapter / Wired Ethernet. 

### Dongle firmware

3.015.02

### Additional context

I have attempted a fresh install by removing the previous intigration custom_components/solax_modbus folder on server, restarting HA, and doing fresh install from HACS this fixed a issue I was having with it giving a error about "Failed to call service select/select option Modbus connection client not connected"

But as I say it's no longer controlling the inverter, but values now change and don't get the above modbus connection error.

wills106, 2024-08-20:
Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1

wills106, 2024-09-07:
@UK-JustinSmall or @fastsonic any feedback? Can this be closed now?

UK-JustinSmall, 2024-09-07:
From my side can be closed thank you :-)Sent from my iPad

closed 2024-09-07 by wills106