wills106/homeassistant-solax-modbus #993: unable to control inverter following upgrade [Bug]:

luke3579, 2024-08-09:
### Describe the bug

after upgrading to 2024.08.** i am unable to control the inverter, selecting inverter charger use modes, adjusting the charging times etc. error comes up when adjusting the nightcharge upper SOC option for example is "failed to perform action number/set_value. Modbus error: [Connection] Client is not connected" 
HA can ping the WiFi adapter on the inverter, no dropouts etc. 
- tried unplugging and replugging the dongle.
- tried removing inverter, removing the integration, restarting HA, redownloading the integration, restarting HA, readding the inverter... still the same

### Integration Version

2024.08.8

### Homeassistant core version

2024.8.0

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H4372A

### Inverter firmware versions

DSP v1.36 ARM v1.34

### Connection Method

Pocket WiFi 3.0

### Dongle firmware

3.015.02

### Detailed Error Log

2024-08-09 09:40:09.321 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-09 09:40:09.321 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139676680027392] Unexpected exception
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
response = await hass.services.async_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 120, in async_set_value
await entity.async_set_native_value(native_value)
File "/config/custom_components/solax_modbus/number.py", line 151, in async_set_native_value
await self._hub.async_write_register(
File "/config/custom_components/solax_modbus/__init__.py", line 579, in async_write_register
return await self.async_lowlevel_write_register(unit, address, payload)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 571, in async_lowlevel_write_register
resp = await self._client.write_register(address, payload[0], **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
resp = await asyncio.wait_for(
^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
return await fut
^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-09 09:40:14.371 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-09 09:40:19.381 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-09 09:40:24.395 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!

Codiwr, 2024-08-09:
Same issue here

Duncik, 2024-08-10:
Same. Looks like it is caused by HA Update. With version 2024.07.x it worked okay.

Codiwr, 2024-08-10:
Broken for me on 2024.8.0

ralfna, 2024-08-11:
Broken also with 2024.8.1

Duncik, 2024-08-11:
I got some logs. 
It looks like there is some problem with writing modbus. 
Reading looks okay but everytime there is request for writing it disconnects. 

pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-10 22:21:17.424 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-10 22:21:17.425 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.76:502
2024-08-10 22:21:17.435 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.1.76:502
2024-08-10 22:21:17.435 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-10 22:21:23.353 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-08-10 22:21:32.426 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-10 22:21:33.234 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-08-10 22:21:47.427 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-10 22:21:48.135 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-08-10 22:22:00.058 INFO (MainThread) [custom_components.solax_modbus.number] writing SolaX export_control_user_limit number register 66 value 45 after div by readscale 10 scale 1
2024-08-10 22:22:00.058 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.76:502
2024-08-10 22:22:00.131 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.1.76:502
2024-08-10 22:22:02.428 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-10 22:22:02.428 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.76:502
2024-08-10 22:22:02.591 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.1.76:502
2024-08-10 22:22:02.591 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-10 22:22:05.134 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-10 22:22:05.135 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-10 22:22:05.135 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.76:502
2024-08-10 22:22:05.138 ERROR (MainThread) [homeassistant.components.automation.povoleni_dodavky_do_site] Povolení dodávky do sítě: Error executing script. Unexpected error for call_service at pos 1: Modbus Error: [Connection] Client is not connected
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 525, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 120, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/solax_modbus/number.py", line 151, in async_set_native_value
    await self._hub.async_write_register(
  File "/config/custom_components/solax_modbus/__init__.py", line 579, in async_write_register
    return await self.async_lowlevel_write_register(unit, address, payload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 571, in async_lowlevel_write_register
    resp = await self._client.write_register(address, payload[0], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-10 22:22:05.152 ERROR (MainThread) [homeassistant.components.automation.povoleni_dodavky_do_site] While executing automation automation.povoleni_dodavky_do_site
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/automation/__init__.py", line 755, in async_trigger
    return await self.action_script.async_run(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/config/custom_components/solax_modbus/__init__.py", line 571, in async_lowlevel_write_register
    resp = await self._client.write_register(address, payload[0], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^

Codiwr, 2024-08-11:
Reading ok for me. Also upgraded to 8.1. with no effect.

Can usually get it to write a single parameter e.g. Self use to Manual, but
the subsequent steps in the automation fail (e.g
Change to force Charge, Set Charge limit).

Temporary fix has been to leave Force Charge set all the time (and cancel
the automations for for discharge at 10pm) so when it flips between self
use and manual with the intention of charging I don't end up discharging by
accident

Not ideal, as I lose the automated export to grid at the end of the day,
but can avoid the battery dumping into the car.

vansteedens, 2024-08-11:
Same as all those above, can read everything but not change so now automations don't work such as changing discharge times when EV is charging. Hopefully can be fixed quickly.

Duncik, 2024-08-12:
So i was able to resolve the issue by manually downgrading the Home Assistant core and integration. 
Mine working combo : HASS Core : 2024.7.1 And integration version : 2024.05.7

fastsonic, 2024-08-13:
Same for me. my EW11 will also freeze and i have to restart the EW11.

spetrie13, 2024-08-14:
> Same for me. my EW11 will also freeze and i have to restart the EW11.

Same for me. Solis inverter with EW11. Only started freezing after update to latest version of integration and HA.
Have to restart the solar integration to get data flowing again.

heyiamluke, 2024-08-16:
Writing to modbus doesn't work. So same for me since upgrading

wills106, 2024-08-20:
Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1

Codiwr, 2024-08-20:
Thanks for the update.

Unfortunately the problem persists (Modbus not connected) following the
update.

Many thanks

wills106, 2024-08-20:
Is that with removing the Integration, restart HA and adding it in again?
To make sure it's using the new code and not the old `client.connect` method.

vansteedens, 2024-08-20:
> Is that with removing the Integration, restart HA and adding it in again? To make sure it's using the new code and not the old `client.connect` method.

I'm having the same with a Solax Inverter. I installed the latest Home Assistant update as had reverted, then removed inverter and restarted, updated the integration, added the integration back and configured the settings and still doesn't allow me to write to it, saying not connected. How can I check it's not still the old 'client.connect' method? Can I change something manually?

The logs show this:

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:540
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12:53:58 (27 occurrences)
Last logged: 16:49:37

Cancel send, because not connected!

wills106, 2024-08-20:
@vansteedens Do you get more info if you show full error logs https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log remember to search for SolaX in the box provided.

02curls, 2024-08-20:
@wills106 I deleted the inverter, deleted the integration, retarted HA, installed integration2024.08.9, the writing problem remains, or actually sometimes it works, sometimes it does not, please have a look at my screen recording illustrating that unpredictable behavior: https://www.dropbox.com/s/nvol95x6sohyxiy/modbuserror.mov?dl=0 
it feels like it works at first try, but then if trying to write the entity again within a short period of time, it displays the error, as if changing the value could only be done once in a certain period of time, but not too often, or too soon again

vansteedens, 2024-08-20:
> @vansteedens Do you get more info if you show full error logs https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log remember to search for SolaX in the box provided.

I searched for Solax in full log and this is what I get: 
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
File "/config/custom_components/solax_modbus/__init__.py", line 570, in async_write_register
File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_lowlevel_write_register

wills106, 2024-08-20:
Looks like it's correctly setup then.

lhorak, 2024-08-20:
I believe it might be a race condition in https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/__init__.py#L560-L563 where we actually await the `self._check_connection()` but it's not further checked if the connection is available or not, so even if it returns false, it proceeds to call the `self._client.write_registers(...)`

this doesn't resolve the issue of not having the client connected, but I think it could be hadeled there (or at least the call shouldn't proceed if the client is not connected)

lhorak, 2024-08-20:
Actually, I'll defer to your knowledge @wills106 but could it be that due to https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/__init__.py#L567 calling `self.async_connect()` and right after the call to `self.async_lowlevel_write_register(unit, address, payload)` might try to connect again in `self._check_connection()` causing the client to connect twice to the master modbus resulting in issue described in https://github.com/wills106/homeassistant-solax-modbus/tree/main?tab=readme-ov-file#multiple-connections ?

wills106, 2024-08-20:
I was thinking of removing all of the check connections. Once the first connection has been made pyModbus is meant to automatically reconnect anyway.

The strange thing is, I can't recreate any of the issues people are having in general.

I think the Gen3 Hybrid SolaX's with the built in Ethernet just aren't bothered what you do to them, they still just work somehow.

lhorak, 2024-08-20:
Yea it looks like the issues are mostly happening with the pocket wifi modbus connections (I'm in the same boat)

lhorak, 2024-08-20:
@wills106 I just tested my PR #1006 and I can confirm (at least in my instance) that it fixes the consecutive writes mentioned by @02curls in https://github.com/wills106/homeassistant-solax-modbus/issues/993#issuecomment-2299226925

lhorak, 2024-08-20:
@wills106 I believe the `_check_connection` overall is fine (doesn't hurt to have that check in there), the only issue was that with `async_write_register` it tried to establish connection without checking if there is already a previous open connection to the modbus, probably causing issues

vansteedens, 2024-08-21:
> @wills106 I just tested my PR #1006 and I can confirm (at least in my instance) that it fixes the consecutive writes mentioned by @02curls in [#993 (comment)](https://github.com/wills106/homeassistant-solax-modbus/issues/993#issuecomment-2299226925)

I tested your change by commenting out the line you removed and can confirm this fixes it. My automations worked fine also. Thanks for working that out until a full fix is pushed out. Let's hope future updates don't require re-setting up the inverter as my automations are set up using device ID and have to keep re-doing them haha

Codiwr, 2024-08-21:
Great detective work guys! Hopefully it can be incorporated into the next update very soon

wills106, 2024-08-21:
Does writing to multiple registers work ok?
(`client.write_registers`) Ie updating RTC?
Or does that need modification as well?

lhorak, 2024-08-21:
@wills106 I believe the issue was isolated to the `async_write_register` calls, I haven't found other place that would initiate the modbus client connection outside of the `_check_connection` (which would case the race condition)

wills106, 2024-08-21:
Ok thanks, I just wanted to check.

closed 2024-08-21 by wills106

wills106, 2024-08-21:
Just released [2024.08.10](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.10) with the change.

lhorak, 2024-08-21:
@02curls glad to hear it's working!

Codiwr, 2024-08-21:
Brilliant work guys, working on latest update!

Thankyou so much!

PR #1006 (2024-08-21): Remove unnecessary client connection in async_write_register
The client connection happens later in `self.async_lowlevel_write_register` call, if the client is not connected already

Potentially resolves #993