wills106/homeassistant-solax-modbus #994: [Bug]: Broken Export Control setting for SolaX X3-Ultra

KOPRajs, 2024-08-09:
### Describe the bug

We are running 2x SolaX X3-Ultra 30K inverters in master-slave parallel mode. Each inverter has a maximum power of 30 kW, but by default I can only set 2 500 as a maximum _Export Control Limit_ value. Also by default the displayed _Export Control Limit_ value is 10x smaller than what is actually set in the inverter.

I've found these diagnostics entities in the device configuration:
__config_export_control_limit_readscale_
__config_max_export_

I can fix the scale by setting the __config_export_control_limit_readscale_ or by adding _H3BC_ to the _EXPORT_LIMIT_SCALE_EXCEPTIONS_ in _plugin_solax.py_. Both options seem to have the same effect. Now the displayed value in HA is the same as what I set in the inverter (up to 50 000 W, which is the maximum we are allowed to export by the operator).

The issue is that I am unable to set the _Export Control Limit_ from the HA.
By default, I can only set 2 500. I understand that this is the default limit for the unknown inverter. When I change the __config_max_export_ to a higher value, say 10 000, I can set up to 10 000 by the slider control, however it seems I can't set more than 2 000 by typing the value to the text box in the entity view. When I set the __config_max_export_ to 25 000, I can still set up to 25 000 by the slider, but not directly by typing the value to the text box. When I set the __config_max_export_ above 25 000 to the desired 50 000, the slider disappears and there is only a text box. Now I can't set anything above 2 000 by the text box. When trying to set more than 2 000 to the text box, I get the following error: "Failed to perform the action number/set_value. Modbus Error: [Connection] Client is not connected". The device is connected for sure and I can still get values from it.

So here is the summary of the issues:
- 0.1x scale of the Export Control Limit for X3-Ultra 30K (can be fixed by __config_export_control_limit_readscale_)
- no support for a maximum of 30 000 _Export Control Limit_ for the X3-Ultra 30K by default
- broken number/set _value for values above 2 000 for _Export Control Limit_
- no slider for _Export Control Limit_ above 25 000 of __config_max_export_
- hard coded maximum of 50 000 in the __config_max_export_ (2x X3-Ultra 30K actually have a maximum of 60 000)
- no automatic support for a maximum _Export Control Limit_ in parallel mode (should be twice as big for 2 inverters)

### Integration Version

2024.08.8

### Homeassistant core version

2024.8.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H3BC30

### Inverter firmware versions

DSP 80 ARM 14.8

### Connection Method

USB Dongle v4 LAN+WiFi

### Dongle firmware

1.002.09

wills106, 2024-08-20:
I have added max_currents for X3-Ultra also increased "Config Max Export" to 60kW in [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9)

KOPRajs, 2024-08-20:
Great, thank you.

FYI: It seems that the main issue we have with 2024.08.8 is the modbus error "Client not connected" when trying to change any setting on the inverter. It does not happen everytime (sometimes the setting is actually changed), but it happens very frequently, so it is mostly useble only for reading. Also it seems that the write operation takes a lot more time in 2024.08.8 then before. Currently we are back on 2024.05.7 and everything seems to work smooth, fast and stable with the limits set to 50 kW and scale to 10 via the diagnostic entities.

wills106, 2024-08-20:
> "Client not connected"

That should be corrected in 2024.08.9 as it's reverted client.connect back to 2024.05.7 method.

wills106, 2024-09-07:
Any updates or feedback on this issue?

KOPRajs, 2024-09-08:
Sorry, we are currently stuck on 2024.05.7 version, as it is mostly working fine, while there are much bigger issues to solve directly with SolaX for the master-slave dual X3 Ultra setup (firmware issues).

magd1978, 2025-04-09:
Hello.

I have the same issue with a SolaX X1-Hybrid-G4, connected by RS-485 adapter.

I tried to control max export power by "SolaX Inverter Export Control User Limit" entity, but the inverter real value is 10x smaller than entity display, and entity range is limited from 0 to 9200 W., but it sets 0 to 920 W. in inverter register.

¿Could you fix entity value scale to fit inverter values? ¿Could you permit 60000 W. as maximum entity value to fit inverter limit (60000 = unlimited).

Thanks!

magd1978, 2025-04-09:
Hello again.

I'm answering myself. :-D

> ¿Could you fix entity value scale to fit inverter values?
I managed to correct the scaling by setting the "SolaX Inverter Config Export Control Limit Readscale" entity to 1.

> ¿Could you permit 60000 W. as maximum entity value to fit inverter limit (60000 = unlimited)?
As for equating the maximum value supported by the inverter (60,000 W) with the maximum allowed by the "SolaX Inverter Export Control User Limit" entity, it's no longer necessary; it's a cosmetic issue. Now that the scaling is correct, the value of 9200 W is perfectly valid for me, as my inverter is 6000 W.

closed 2025-10-01 by github-actions[bot]