wills106/homeassistant-solax-modbus #995: [Bug]: 2024.8.8 Cannot add inverter again via serial after upgrade to 2024.8.8

jimbaxter, 2024-08-10:
### Describe the bug

With new  I removed the integration first, upgraded the solax-modbus from HACS and rebooted the raspberry PI. Now I cannot add it again and I get "unknown error," on the screen to define the serial address.
I am adding the inverter with the same information as earlier.

/dev/ttyUSB0
19200

I upgraded from [2024.05.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.7)

### Integration Version

2024.8.8

### Homeassistant core version

2024.8.0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H4502A

### Inverter firmware versions

DSP v1.41 ARM v1.40

### Connection Method

Generic RS485 - USB, as listed in compatible RS485 Adaptors

### Dongle firmware

N/A

### Detailed Error Log

2024-08-10 12:54:24.907 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-10 12:54:48.676 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting repository information
2024-08-10 12:54:49.131 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Running checks against 2024.08.8
2024-08-10 12:54:49.621 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting documentation for version=2024.08.8,filename=README.md
2024-08-10 12:54:49.621 DEBUG (MainThread) [custom_components.hacs] Trying to download https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/2024.08.8/README.md
2024-08-10 12:54:51.514 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Houseload from grid (and PV)' trigger: In 'numeric_state' condition: unknown entity sensor.solax_battery_capacity
2024-08-10 12:55:21.098 INFO (SyncWorker_8) [homeassistant.loader] Loaded solax from homeassistant.components.solax
2024-08-10 12:55:29.086 INFO (ImportExecutor_0) [custom_components.solax_modbus.config_flow] detected HA core version 2024 8
2024-08-10 12:55:29.086 INFO (ImportExecutor_0) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-08-10 12:56:48.593 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base: {'name': 'SolaX', 'interface': 'serial', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 15, 'scan_interval_medium': 15, 'scan_interval_fast': 15, 'inverter_name_suffix': 'Inverter', 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2024-08-10 12:56:48.594 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base config for SolaX: pre: {'name': 'SolaX', 'interface': 'serial', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 15, 'scan_interval_medium': 15, 'scan_interval_fast': 15, 'inverter_name_suffix': 'Inverter', 'read_eps': False, 'read_dcb': False, 'read_pm': False}
File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
In 'state' condition: unknown entity select.solax_remotecontrol_power_control
In 'state' condition: unknown entity select.solax_remotecontrol_power_control
2024-08-10 13:12:18.075 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting repository information
2024-08-10 13:12:18.773 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Running checks against 2024.08.8
2024-08-10 13:12:19.254 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting documentation for version=2024.08.8,filename=README.md
2024-08-10 13:12:19.255 DEBUG (MainThread) [custom_components.hacs] Trying to download https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/2024.08.8/README.md
2024-08-10 13:12:34.045 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting repository information
2024-08-10 13:12:34.600 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Running checks against 2024.08.8
2024-08-10 13:12:35.100 DEBUG (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Getting documentation for version=2024.08.8,filename=README.md
2024-08-10 13:12:35.101 DEBUG (MainThread) [custom_components.hacs] Trying to download https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/2024.08.8/README.md

### Additional context

This is the same issue as https://github.com/wills106/homeassistant-solax-modbus/issues/977

wills106, 2024-08-12:
Closing down as like you say a duplicate of #977

closed 2024-08-12 by wills106

jimbaxter, 2024-08-12:
Thanks, I agree.

I was going to just mention I had issues in #977 and only created a new ticket because it was non Solax and the Solax bug reporter appears to have extra fields that I thought might be useful.