wills106/homeassistant-solax-modbus #996: [Bug]: Unable to add Solax X3-Ultra 15k with version 2024.8.8

Hetzi74, 2024-08-11:
### Describe the bug

When installing the Solax Modbus adapter, I only get the error messages from the log. I have tried a complete reinstallation with the result shown in the log file.

### Integration Version

2024.8.8

### Homeassistant core version

2024.8.1

### Inverter brand

Solax X3-Ultra 15k

### Plugin used

plugin_solax.py

### Serial prefix

H3BD15

### Inverter firmware versions

cant find out right now

### Connection Method

Solax Wifi+LAN

### Dongle firmware

1.002.09

### Detailed Error Log

2024-08-11 18:00:18.618 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-11 18:00:18.619 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-11 18:00:22.017 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/frontend", "/config/custom_components/hacs/hacs_frontend_experimental", False)])`; This function will be removed in 2025.7 at custom_c
omponents/hacs/frontend.py, line 49: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:00:22.021 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/iconset.js", "/config/custom_components/hacs/iconset.js", True)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 57: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:00:22.024 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:00:22.349 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 642: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:00:57.305 WARNING (MainThread) [homeassistant.util.json] json_loads was called from hacs, this is a deprecated function which will be removed in HA Core 2025.8. Use homeassistant.util.json.json_loads instead, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:01:09.632 WARNING (MainThread) [homeassistant.util.json] json_loads was called from hacs, this is a deprecated function which will be removed in HA Core 2025.8. Use homeassistant.util.json.json_loads instead, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:01:10.793 WARNING (MainThread) [homeassistant.util.json] json_loads was called from hacs, this is a deprecated function which will be removed in HA Core 2025.8. Use homeassistant.util.json.json_loads instead, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:01:12.938 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles", "/config/www/community", True)])`; This function will be removed in 2025.7 at custom_components/hacs/base.py, line 1162: self.hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-11 18:02:15.584 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-11 18:02:15.584 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 549, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-11 18:02:15.589 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-08-11 18:02:25.593 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-11 18:02:25.594 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 549, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-11 18:02:25.594 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-08-11 18:02:45.599 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None

### Additional context

Originally I had set up the Raspberry pi 5 about 2 weeks ago with the then current versions. With these versions (Solax Modbus version 2024.05.7 and Homeasitant Core 2024.07.4) the Solax Modbus integration with the Solax X3 Ultra had already worked. 

I then went through the ongoing HA updates and already with HA Core version 2024.08.1, which I installed this morning, I got the error messages of unreadable serial number. (Presumably a problem with something not working with the TCP Modbus connection)

Then I tried various combinations of upgrades and downgrades of both Solax Modbus and HA Core, unfortunately all without success.

My last attempt was to start again with a completely fresh SD card with HA in the Raspi, but I had no success with that either, as the attached log file shows.

How do I get a connection from HA to the Solax inverter again?

butvoo, 2024-08-12:
I have the same problem, 
1. I have NOT remvoed  Solax inverter from Solax Modbus integration
2.  upgraded Solax Modbus to 2024.8.8
3. I am not able to add Solax inverter, it error "Unknow error occurred"

[image]

butvoo, 2024-08-12:
Workaround that works 👍  is 
1. add (fake) integration as TCP/ Ethernet
[image]

2.  any temporary IP address can be used:
[image]

3. inside integration > 3dots > "Configure" > Change to "Serial" > Submit> Serial port name> Baudrate > Submit> ShutDown or Restart HA

[image]
[image]
[image]

After restarting it loads Solax with the use of Serial
[image]

wills106, 2024-08-12:
@butvoo you have a different issue to this issue.
Yours is a duplicate of #995 & #977

Hetzi74, 2024-08-12:
It's working again!

Disconnecting and reconnecting the Pocket Wifi+Lan adapter (is this the same as PocketWiFi 3.0 from the FAQ? Mine is connected via LAN) finally worked.

But first after adding the Solax Modbus adapter there was no data under the sensors but the serial number was visible and about 215 hidden entities.

Some time later I deleted the Solax Modbus adapter and created it again. There were some entries in the log that the serial number could not be found, but it finally worked and I saw the current data of the inverter!

I then imported the backup of the data from the previous SD card and again saw the old data and the current inverter data with Solax Modbus version 2024.05.7. Then I deleted the Solax adapter according to the upgrade description, carried out updates and added Solax again and all the data was there again. 

But now there are a few small problems with non-functioning automation and renamed sensors, especially the SOC of the battery has been renamed, is now called "sensor.solax_battery_total_capacity_charge". Previously the last "_charge" was not appended. But these are small things that I can easily adjust!