wills106/homeassistant-solax-modbus #998: [Bug]: Can not control values from HA

toko-dotcom, 2024-08-17:
### Describe the bug

Since update 2024.08.08 I am not able to change values in "SolaX Inverter Export Control User Limit" via HA. It is strange behavior, because I test it several times and in non expected period some of request change value. But after that next 20 times nothing happend and only get back error Modbus erroc connection. Client is not connected. But all values I see in overview actively chaning so modbus is connected all time. 

### Integration Version

2024.08.08

### Homeassistant core version

2024.8.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

Dsp  v1.41 arm 1.40

### Connection Method

Solax pocket wifi 3.0

### Dongle firmware

3.006.04

spetrie13, 2024-08-17:
I have similar issues with Solis inverter and EW11 modbus device. Was working perfectly for over 1 year with setup until the latest updates.

mobyline, 2024-08-18:
The same issues here with X3 G4. 
A few month ago it worked fine.

toko-dotcom, 2024-08-18:
I also have x3 g4 10kw. I use wifi pocket 3.0  modbus because I have EVC connected via Comm port. As i read in different issue report, there is not solution to have EVC and Modbus TCP RS485 on one Comm port. So my only one solution is to stay Wifi Pocket 3.0  modbus and EVC via Comm port. 

I can confirm that some weeks ago I was able set inverter configuration from HA via Wifi pocket 3.0 modbus.

Thank you for help in advance

Btw: what already tried:
Uninstall Solax inverter and modbus from HA and delete all related setting
Install solax modbus via hacks and updated repositories
Setup automatization with continue on error (no help, all time stops on failure which I wrote upper)
Setup loop in automatization until value will be changed ( no help, all time stops on failure which i wrote upper)

The strangest is sometimes by manual updating via HA configuration is accepted and written to inverter. But it is not possible to say that for example each 20 times it is ok or something like that.

Read entity information from Inverter to HA works fine without any issue. Still with wifi pocket 3.0 modbus.

Hetzi74, 2024-08-18:
On my inverter Solax X3-Ultra 15k i can change the value for number.solax_export_control_user_limit and sometimes i get that modbus error too. 

But the value i seen in HA are 10-times off. So a value of 500W in HA is 5000W on the inverter itself, so there is allways a faktor of 10 difference between the visible HA value and the value at the inverter.

lhorak, 2024-08-19:
I'm having the same issues, reading values is fine, but writing anything fails with `Modbus Error: [Connection] Client is not connected` This happens any time I try to change any value, including the lock state. The lock state says unlocked, but when I try to change to Unlocked Advanced, fails with the same client not connected

Inverter: Solax X3 Hybrid G4
DSP v1.39
ARM v1.38
HA Version: 2024.8.1
Integration version: 2024.08.8

Error logs:
```
2024-08-19 10:55:33.445 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!

2024-08-19 10:55:33.445 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [130051770470400] Unexpected exception

Traceback (most recent call last):

File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service

response = await hass.services.async_call(

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call

response_data = await coro

^^^^^^^^^^

File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service

return await target(service_call)

^^^^^^^^^^^^^^^^^^^^^^^^^^

File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call

single_response = await _handle_entity_call(

^^^^^^^^^^^^^^^^^^^^^^^^^^

File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call

result = await task

^^^^^^^^^^

File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option

await self.async_select_option(option)

File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option

await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=payload)

File "/config/custom_components/solax_modbus/__init__.py", line 579, in async_write_register

return await self.async_lowlevel_write_register(unit, address, payload)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "/config/custom_components/solax_modbus/__init__.py", line 571, in async_lowlevel_write_register

resp = await self._client.write_register(address, payload[0], **kwargs)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute

resp = await asyncio.wait_for(

^^^^^^^^^^^^^^^^^^^^^^^

File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for

return await fut

^^^^^^^^^

pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

2024-08-19 10:55:38.451 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
```

wills106, 2024-08-20:
Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1

GeordiePorker, 2024-08-20:
> Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1

I've updated to 2024.08.9 but I'm getting the same issue: Modbus Error: [Connection] Client is not connected

The integration works great until I change a setting.  When I try to change the setting back, I get this error.

I'll try to grab the logs for both success and failure, but I'm new to this!

Thanks for the great work - I hate to complain when it's a personal effort...sorry :(

GeordiePorker, 2024-08-20:
> > Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1
> 
> I've updated to 2024.08.9 but I'm getting the same issue: Modbus Error: [Connection] Client is not connected
> 
> The integration works great until I change a setting. When I try to change the setting back, I get this error.
> 
> {{I'll try to grab the logs for both success and failure, but I'm new to this! - Edit1: Failure Logs Below}}
> 
> Error Log for Failure:
> 
> ```
> 2024-08-20 11:43:22.121 INFO (MainThread) [custom_components.solax_modbus.select] writing SolaX select register 0 value 2014
> 2024-08-20 11:43:22.122 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
> 2024-08-20 11:43:22.128 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
> 2024-08-20 11:43:24.216 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
> 2024-08-20 11:43:24.217 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
> 2024-08-20 11:43:24.221 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
> 2024-08-20 11:43:24.221 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
> 2024-08-20 11:43:27.129 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
> 2024-08-20 11:43:27.130 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
> 2024-08-20 11:43:27.130 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
> 2024-08-20 11:43:27.131 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139798177090720] Unexpected exception
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
>     response = await hass.services.async_call(
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
>     response_data = await coro
>                     ^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
>     return await target(service_call)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
>     single_response = await _handle_entity_call(
>                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
>     result = await task
>              ^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
>     await self.async_select_option(option)
>   File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
>     await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=payload)
>   File "/config/custom_components/solax_modbus/__init__.py", line 570, in async_write_register
>     return await self.async_lowlevel_write_register(unit, address, payload)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_lowlevel_write_register
>     resp = await self._client.write_register(address, payload[0], **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
>     resp = await asyncio.wait_for(
>            ^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
>     return await fut
>            ^^^^^^^^^
> pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
> 2024-08-20 11:43:27.146 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
> 2024-08-20 11:43:32.163 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
> 2024-08-20 11:43:32.163 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Connection] Client is not connected : SolaX cannot read holding registers at device 1 position 0x1d
> NoneType: None
> 2024-08-20 11:43:32.164 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
> 2024-08-20 11:43:32.164 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
> ```
> 
> Edit2: Oddly, the 'live' sensors still work. So I'm still seeing the solar production and export which all tie up to the native SolaX sensors I can view.
> 
> Edit3: It appears this issue is related to how long the system has been up for rather than "work once then fail"; if I reload HA, it works then it fails. If I reload HA then get distracted before trying something, it fails.
> 
> Thanks for the great work - I hate to complain when it's a personal effort...sorry :(

OK, I've been doing some more digging.  Building on the description of the fix in issue#1003 [https://github.com/wills106/homeassistant-solax-modbus/pull/1003], I changed the timeouts in init.py to 10s.

Oddly, I still get a near immediate 'client not connected' error - I would have expected the error to wait at least until 1 'unit of timeout' had passed if not all 6. {BTW, this stops HA automations in their tracks.}
BUT, after 10-15seconds, the command I had passed to the inverter is implemented.

I have zero idea whether this information is helpful, feel free to tell me to hush 🤣

lhorak, 2024-08-20:
@GeordiePorker I don't think your issue is the same, as from the logs it looks like the client succeeds to connect, but you get the exception `SolaX cannot read holding registers at device 1 position 0x1d NoneType: None` - just my opinion, I haven't been using the integration for long...

What inverter do you own and how is it connected? I think it would help to give the basic info (inverter brand/type, FW version, HA version, Modbus type connection...)

lhorak, 2024-08-20:
Also might be good to mention what setting are you trying to change, and also if your inverter is in the unlocked state (just to make sure 🙂 )

GeordiePorker, 2024-08-20:
All fair points!  I've been using this integration for a while and it's great - never had any issues so I've never had cause to post before...

Inverter: Solax X1 Hybrid
DSP v1.42
ARM v1.39
Hardware: Gen4
HA Version: 2024.8.2
Integration version: 2024.08.9

Connected using a device SolaX Cloud calls "WiFi3.0" (which I believe from various images is commonly called a PocketWiFi)

My system is set to "Unlocked - Advanced"

The setting I'm tyring to change is "any of them", but most commonly I'm tyring to put the system into:
`Charger Use Mode: Manual Use`
`Manual Mode Control: On`
`Manual Mode Select: Force Discharge`

This, clearly, should force the system to discharge to the grid which give me an easy test of the controls which I can verify from both SolaX Cloud and from my electricity meter.

wills106, 2024-08-20:
@GeordiePorker Is that with removing the Integration, restart HA and adding it in again?
To make sure it's using the new code and not the old `client.connect` method.

GeordiePorker, 2024-08-20:
@wills106 I didn't fully remove the integration.  I deleted my inverter from the integration first using the delete button in the attached.
[image]

Then, I updated from 2024.05.7 (I think!) to 2024.08.8 and was immediately advised that there was a 2024.08.09, so I installed that.

I then re-added my inverter using the same name as before (and I can indeed still see all of the history).

I had to re-enable a few sensors (such as Total Grid Export) and all seemed fine.

However obviously the device id in Home Assistant has changed, so I was re-configuring all of my automations and it was the testing of these automations which initially highlighted the issue.

I have looked at init.py and all of the client.connect code is the same as in the #1003 issue/pull request.

GeordiePorker, 2024-08-20:
I'll try again. I didn't restart HA between removing the inverter and updating the integration.

GeordiePorker, 2024-08-20:
@wills106
OK, so I deleted the integration from HACS and looked on the file system - the folder 'solax_modbus' was gone.
Restarted HA
Fresh install of the integration via HACS
Restarted HA
Ran through the setup; all good.  I can see the 'live' data (essentially the read-only aspects) fine.
Cannot force the inverter into Manual Mode.  Same fault.
Looked on the file system - 'solax_modbus' is back!

Logs:
```
2024-08-20 15:18:34.238 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-20 15:18:34.238 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
2024-08-20 15:18:34.244 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
2024-08-20 15:18:34.245 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-20 15:18:34.245 DEBUG (MainThread) [custom_components.solax_modbus] SolaX modbus holding block start: 0x1d end: 0x7e len: 97
regs: [29, 30, 125]
2024-08-20 15:18:34.555 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-20 15:18:34.555 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-20 15:18:34.555 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
2024-08-20 15:18:34.556 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139652033598592] Unexpected exception
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
response = await hass.services.async_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 999, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1071, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 188, in async_handle_select_option
await self.async_select_option(option)
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=payload)
File "/config/custom_components/solax_modbus/__init__.py", line 570, in async_write_register
return await self.async_lowlevel_write_register(unit, address, payload)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_lowlevel_write_register
resp = await self._client.write_register(address, payload[0], **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
resp = await asyncio.wait_for(
^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
return await fut
^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-08-20 15:18:34.569 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
2024-08-20 15:18:49.238 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-08-20 15:18:49.239 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 10.19.5.80:502
2024-08-20 15:18:49.248 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.19.5.80:502
2024-08-20 15:18:49.248 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-08-20 15:18:49.248 DEBUG (MainThread) [custom_components.solax_modbus] SolaX modbus holding block start: 0x1d end: 0x7e len: 97
regs: [29, 30, 125]
2024-08-20 15:18:49.576 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
```

mobyline, 2024-08-20:
Version 2024.08.9 works fine for me, but I had to reboot my (cable) router before!?!

Thanks to the developer

toko-dotcom, 2024-08-20:
> Just released [2024.08.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.9) reverting the `client.connect` change from 2024.08.1

Thanks @wills106  Very much.
It works fully for me. Entity information and also able to change values via HA.

My steps:
Delete solax modbus integration
Reboot HA
Delete HACS Solax modbus 
Reboot HA
Check if any files are in components folder
Download solax modbus to hacs
Add solax modbus integration woth different name
Reboot
And test with possitive result.

Thank you very much

GeordiePorker, 2024-08-20:
Last bit of spam, I promise!

I've now reverted to 2024.05.7 and so far I have had no issues with changing settings.  I had to do a manual install, of course, so I look forward to the merry hell of talking HA into polling for updates!

Also, note to self: use backups before major installs! 🤦

toko-dotcom, 2024-08-21:
> Thanks @wills106 Very much. It works fully for me. Entity information and also able to change values via HA.
>
> My steps: Delete solax modbus integration Reboot HA Delete HACS Solax modbus Reboot HA [...]

@wills106  I am sorry but have to report that now without any changing setting or installing updates I ha e same problem. 

Chyba: Modbus Error: [Connection] Client is not connected

Again during chaning values via HA. Connection as I have written is by Modbus Wifi Pocket 3.0...

lhorak, 2024-08-21:
@toko-dotcom try updating to [2024.08.10](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.10) there was another issue with the write call, which could lead to race condition causing the Client not connected error (especially when making fast consecutive changes)

The new version should hopefully fix that.

lhorak, 2024-08-21:
> > @toko-dotcom try updating to [2024.08.10](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.10) there was another issue with the write call, which could lead to race condition causing the Client not connected error (especially when making fast consecutive changes)
> > The new version should hopefully fix that.
> 
> @lhorak thank you, updated and looks again fine all. Diky moc :)

Není zač 🙂 🇨🇿

GeordiePorker, 2024-08-22:
> @toko-dotcom try updating to [2024.08.10](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.10) there was another issue with the write call, which could lead to race [...]

@lhorak @wills106 Thank you!  Just re-updated from 2024.05.7 to 2024.08.10 and all appears fine including changing settings.

Great integration and thanks for all your efforts!

wills106, 2024-08-24:
I'll close this down as resolved.

closed 2024-08-24 by wills106