wills106/homeassistant-solax-modbus #999: [Bug]: 'Solis' - 'RAI-3K-48ES-5G ' Solar Power values not returned

andygodber, 2024-08-18:
### Describe the bug

Values for the solar readings e.g solax_pv_power_1 (or voltage, current etc) are not returned. All other functions seem ok.
This inverter is AC Coupled, so solar values are obtained via a clamp onto separate inverter. Solis cloud is reading the values correctly.

### Integration Version

2024.08.8

### Homeassistant core version

2024.8.1

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

010262

### Connection Method

RS485 Waveshare

### Detailed Error Log

No errors generated in logs :( all other data being collected

viking2010, 2025-04-02:
Hi @andygodber Is this still an issue for you, or can this now be closed off?

andygodber, 2025-04-02:
> Hi [@andygodber](https://github.com/andygodber) Is this still an issue for you, or can this now be closed off?

Hi - still there on 2025.02.1 Im afraid.

viking2010, 2025-04-02:
I think this is actually the same issue that I'm experiencing. I just hadn't put 2 & 2 together!! I'm raising a ticket about it now but think I'll put all the details into this ticket if that's ok?

viking2010, 2025-04-02:
@wills106, I was going to log a new ticket for this, but I'm adding to Andy's issue which I believe is the same:

At sporadic times during daylight hours, the PV1 and PV2 Power entities report 0 watts. The PV1 and PV2 Current and Voltage have no issues, just the power entities. 

No issue:
[image]

Issue Present:
[image]

You can see the impact in the chart below:
[image]

Here some more info:
Integration Version: 2025.03.4
HA Core Version: 2025.2.2
Inverter Brand: Solis
Inverter Model: RHI-3.6K-48ES-5G
Plugin Used: plugin_solis.py
Serial Prefix: 160F52
Connection Method: S2-WL-ST

When the issue occurs, the debug logs show the following information:
2025-04-02 12:22:29.457 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-04-02 12:22:32.904 ERROR (MainThread) [pymodbus.logging] Exception response 134 / 0
2025-04-02 12:22:32.904 INFO (MainThread) [custom_components.solax_modbus] read_error : solis cannot read holding registers at device 1 position 0xa7ff

NoneType: None
2025-04-02 12:22:32.904 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2025-04-02 12:22:32.904 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-04-02 12:22:47.612 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)

When it works ok, it shows the following:
2025-04-02 12:25:09.467 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-04-02 12:25:12.868 DEBUG (MainThread) [custom_components.solax_modbus] device group read done

If there's anything I can do to provide any more information, please let me know.

andygodber, 2025-04-02:
Just saw there was an update to 2025.04.1 but the bug is still present there.

Query whether @viking2010  is the same issue however - looks like you at least get some values during the day - I get nothing!

@viking2010 looks like your inverter is hybrid, whereas mine is a/c coupled only, pulling solar data from a clamp.

viking2010, 2025-04-02:
@andygodber Yeah mine is a hybrid inverter. Must be trying to pull from the same registers though, so whilst it may not be exactly the same issue, the two could be related.  I apologise if it is a different issue and for putting it alongside yours. I will see what @wills106 has to suggest but if needs be I'll log a separate ticket. 👍

andygodber, 2025-04-02:
I dont get any readings from those metrics with PV in them unfortunately. However, the core inverter W,V,A values all seem to be present.

viking2010, 2025-04-02:
ok, well I'll raise my issue as a separate case then as it does appear to be different. Good to discuss this with you though and figure out if it is different or not.