wills106/homeassistant-solax-modbus #1000: Broken in the latest HA version 2024.8.2 on Solis RHI 6k

shavermcspud, 2024-08-19:
### Describe the bug

 Core 2024.8.2
Supervisor 2024.08.0
Operating System 13.0
Frontend 20240809.0

Add on no longer works, entities blank.
If removed and reinstalled an unknown error occurs. 

### Integration Version

2024.08.8

### Homeassistant core version

2024.8.2

### Inverter brand

Solis RHI 6k 48ES

### Plugin used

plugin_solis.py

### Serial prefix

00000

### Connection Method

Direct Serial RS485

### Detailed Error Log

There are no errors in the Log file

wills106, 2024-08-24:
If there is nothing in the error logs you might not have the correct logging set have a look at the [FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)

I have just released [2024.08.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.11) which fixes another Serial issue.

wills106, 2024-09-20:
Have the newer Integration versions resolved this?