wills106/homeassistant-solax-modbus #1007: [Bug]: Solis - S5-EH1P3.6K-L "Timed Charge / Discharge Current" update breaks communication

AlexKAPD, 2024-08-21:
### Describe the bug

When changing "Timed Charge Current" or "Timed Discharge Current", the Modbus communication stops. 
On integration reload / home assistant restart, the communication is up and running again, and the current value actually changes / updates to the value set prior to restart. 

### Integration Version

2024.08.9 (previous as well)

### Homeassistant core version

2024.8.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603102

### Connection Method

Waveshare RS485 to WiFi/Ethernet Module

### Detailed Error Log

Nothing in Home Assistant Core logs. 

### Additional context

I am aware that the WiFi version of Waveshare is not recommended. 
However, I have had 0 issues with reading data - connection is stable, I have tried various polling rates (1s, 5s, 15s, 20s) - they all work as intended and I don't think the issue is related to the hardware, but might be wrong.
Currently, the Waveshare is connected to Home Assistant via Ethernet connection (rather than WiFi). Issues persist in both connection arrangements.

AKlamaznikovs, 2024-08-28:
Haven't experienced this since 2024.08.10 release, will keep monitoring

jevburchell, 2024-08-29:
I'd like to second this bug - I am seeing similar behaviour.  I lose Modbus connection, with the only way to restart data flow being an HA reboot.  I see this behaviour when making alterations to number input entities in Solax - it seems to happen on quite a few of them, but I have not tested them all.  Confirmed Modbus loss after altering the following:
- Solax Solis Timed Charge Current
- Solax Solis Timed Discharge Current
- Solax Solis Energy Storage Control Switch

When Modbus loss is encountered, I see the following error:
[image]

During this time, if I try to reload the integration, it displays "initialising" but it never reloads.  All entities become unavailable.  

Reloading all YAML has nil effect.  A full HA reboot allows everything to restart normally.

I'm running HA on an Odroid N2+ 4GB with a networked wireless connection to an Elfin EW11.

Thank you in advance for considering this issue

AKlamaznikovs, 2024-08-29:
> I'd like to second this bug - I am seeing similar behaviour. I lose Modbus connection, with the only way to restart data flow being an HA reboot. I see this behaviour when making alterations to number input entities in Solax - it seems to happen on quite a few of them, but I have not tested them all. Confirmed Modbus loss after altering the following:
> 
>     * Solax Solis Timed Charge Current
> 
>     * Solax Solis Timed Discharge Current
> 
>     * Solax Solis Energy Storage Control Switch
> 
> 
> When Modbus loss is encountered, I see the following error: ![Screenshot 2024-08-29 111935](https://private-user-images.githubusercontent.com/128035411/362637414-0859e339-749b-42cd-8289-d608a1ab3e39.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ5Mjc0ODUsIm5iZiI6MTcyNDkyNzE4NSwicGF0aCI6Ii8xMjgwMzU0MTEvMzYyNjM3NDE0LTA4NTllMzM5LTc0OWItNDJjZC04Mjg5LWQ2MDhhMWFiM2UzOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyOVQxMDI2MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OTBjMTg0YjZlMjMzMDMzNjIzZDU3ZGE0NjFmOTlkOTdlZjViYTMzYjQxMTFkOWJmMWQ3MjZiMjFmZDIyMmQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iKapnG1J1CNmBoB7o4o1k1FfJMnynjtx90i2gRCLHag)
> 
> During this time, if I try to reload the integration, it displays "initialising" but it never reloads. All entities become unavailable.
> 
> Reloading all YAML has nil effect. A full HA reboot allows everything to restart normally.
> 
> I'm running HA on an Odroid N2+ 4GB with a networked wireless connection to an Elfin EW11.
> 
> Thank you in advance for considering this issue

Just to check, what version of the Solax Integration are you using? As mentioned above, this has been resolved for me since 2024.08.10. From release notes, looks like the removal of `await self.async_connect()` has fixed it.

jevburchell, 2024-08-29:
> Just to check, what version of the Solax Integration are you using? As mentioned above, this has been resolved for me since 2024.08.10. From release notes, looks like the removal of `await self.async_connect()` has fixed it.

Ok, I have just updated now to 2024.08.13, (was on .08).  Let me also monitor and report back.  Thanks for the heads up!

wills106, 2024-09-07:
Can I close this issue down now?

jevburchell, 2024-09-07:
> Can I close this issue down now?

I've seen no more errors of this type, integration behaviour normal.  Thank you Wills for your fine work 👌🏼

closed 2024-09-07 by wills106