wills106/homeassistant-solax-modbus #1008: [Bug]: Unable to add Solax X3 Hybrid G4 15.0 with version 2024.8.9

Mike5229, 2024-08-21:
### Describe the bug

I can't manage to read out my Solax X3 Hybrid G4 via USB Modbus adapter using my Raspberry.

When configuring the plugin, there is no longer an error and it is displayed as successfully added.
But after trying to initialize, the following error message always appears under integrations/integration/solax_modbus: 

SolaX
No devices or entities
Error during setup:
Check the protocols

The complete error log is inserted below.

I am asking for help as I would really like to read out the data via Raspberry.

[image]

### Integration Version

2024.8.9

### Homeassistant core version

2024.7.4

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM 2.03 and DSP 2.07

### Connection Method

RS485 USB on Raspberry

### Dongle firmware

43 09 42

### Detailed Error Log

2024-08-21 08:55:13.816 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-21 08:55:13.819 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-21 08:55:14.511 WARNING (MainThread) [aiohttp_fast_zlib] zlib_ng and isal are not available, falling back to zlib, performance will be degraded.
2024-08-21 08:55:15.121 ERROR (ImportExecutor_0) [homeassistant.components.camera.img_util] Error loading libturbojpeg; Camera snapshot performance will be sub-optimal
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/site-packages/homeassistant/components/camera/img_util.py", line 100, in __init__
    TurboJPEGSingleton.__instance = TurboJPEG()
                                    ^^^^^^^^^^^
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/turbojpeg.py", line 298, in __init__
    self.__find_turbojpeg() if lib_path is None else lib_path)
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/turbojpeg.py", line 928, in __find_turbojpeg
    raise RuntimeError(
RuntimeError: Unable to locate turbojpeg library automatically. You may specify the turbojpeg library path manually.
e.g. jpeg = TurboJPEG(lib_path)
2024-08-21 08:55:15.548 ERROR (MainThread) [haffmpeg.core] FFmpeg fails [Errno 2] No such file or directory: 'ffmpeg'
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/haffmpeg/core.py", line 130, in open
    self._proc = await asyncio.create_subprocess_exec(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/asyncio/subprocess.py", line 224, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/asyncio/base_events.py", line 1744, in subprocess_exec
    transport = await self._make_subprocess_transport(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/asyncio/unix_events.py", line 211, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/asyncio/unix_events.py", line 820, in _start
    self._proc = subprocess.Popen(
                 ^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'
2024-08-21 08:55:15.561 WARNING (MainThread) [haffmpeg.tools] Error starting FFmpeg.
2024-08-21 08:55:17.876 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.12.3/lib/python3.12/site-packages/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/dietpi_userdata/homeassistant/custom_components/solax_modbus/__init__.py", line 159, in async_setup_entry
    hub = SolaXModbusHub(
          ^^^^^^^^^^^^^^^
  File "/mnt/dietpi_userdata/homeassistant/custom_components/solax_modbus/__init__.py", line 237, in __init__
    self._client = AsyncModbusSerialClient(
                   ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: AsyncModbusSerialClient.__init__() got an unexpected keyword argument 'method'
2024-08-21 08:55:19.391 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/frontend", "/mnt/dietpi_userdata/homeassistant/custom_components/hacs/hacs_frontend", False)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 54: hass.http.register_static_path(f"{URL_BASE}/frontend", locate_dir(), cache_headers=False), please create a bug report at https://github.com/hacs/integration/issues
2024-08-21 08:55:19.408 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/iconset.js", "/mnt/dietpi_userdata/homeassistant/custom_components/hacs/iconset.js", True)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 57: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-21 08:55:19.425 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-21 08:55:19.790 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 630: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-21 08:55:24.073 ERROR (SyncWorker_3) [aiodhcpwatcher] Cannot watch for dhcp packets without a functional packet filter: libpcap is not available. Cannot compile filter !

Mike5229, 2024-08-23:
Same Problem with 2024.8.10...

Any ideas?

lhorak, 2024-08-23:
Hey @Mike5229 I've just raised a PR which theoretically should fix your problem, based on the logs you provided (I don't use serial connection, so can't confirm).

@wills106 please take a look when you have some time 🙂 (is it something that was supported in previous versions of pymodbus and was removed?)

lhorak, 2024-08-23:
BTW @Mike5229 you could test the fix locally by modifying the `__init__.py` file locally on your HA instance according to this change: https://github.com/wills106/homeassistant-solax-modbus/pull/1013

lhorak, 2024-08-23:
> @lhorak
> 
> Hi,
> 
> thank you very much. Unfortunately, I am an absolute beginner in this topic. Can you tell me exactly what I need to do to implement your advice?

Ah ok, no worries, in that case, I would suggest to wait for a new version to be released with the fix once @wills106 gets to it 🙂 

But if you do feel adventurous:
1. You need to go to the file system where your HomeAssistant config is located.
2. Go to `custom_components` -> `solax_modbus` folder
3. There will be a file called __init__.py
4. Open the file in a text editor and remove line 238 (`method="rtu",`) and save
5. Delete the `__pycache__` folder which is also in the `custom_components` -> `solax_modbus` folder
6. Restart Home Assistant
7. Try adding your Inverter again 🙂

wills106, 2024-08-24:
> is it something that was supported in previous versions of pymodbus and was removed?

Possibly, the serial part was added in years ago by someone else. The integration started off TCP only as I have no way of testing RS485 as my Inverter doesn't support it.

wills106, 2024-08-24:
I have just released [2024.08.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.11) I'll close the issue down once you have confirmed it works.

Mike5229, 2024-08-25:
@lhorak  
Thank you very much for your help! This procedure has helped

@wills106 
I have now completely reinstalled version 2024.08.12 and the error no longer occurs. So this version works.

What does not work, however, is the direct addition of the serial interface.
Here you still have to take the detour via adding via TCP / Ethernet and then changing to Serial.

But basically it works now. Thank you very much!!!

closed 2024-08-25 by Mike5229