wills106/homeassistant-solax-modbus #1010: [Bug]:unable to add Solax X3 MIC G2

RSX2613, 2024-08-22:
### Describe the bug

Unable to add Solax X3 MIC G2 in Hubs.
Inverter is not found.
[image]

### Integration Version

2024.08.10

### Homeassistant core version

2024.8.2

### Inverter brand

Solax 

### Plugin used

plugin_solax.py

### Serial prefix

MC205T

### Inverter firmware versions

ARM 1.15 DSP 1.18

### Connection Method

Wifi 3.0

### Dongle firmware

3.006.04

wills106, 2024-08-24:
Can you provide the error log? If there is nothing in the error logs you might not have the correct logging set have a look at the [FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)

RSX2613, 2024-08-27:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:69
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 25. August 2024 um 11:48:19 (4946 Vorkommnisse)
Zuletzt protokolliert: 16:38:31

SolaX West: attempt to read serialnumber failed at 0x0 data: None
SolaX West: attempt to read serialnumber failed at 0x300 data: None
SolaX West: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 540, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 6 retries

wills106, 2024-11-02:
If you unplug the Pocket WiFi 3 and connect it again does it start working again?
It might be worth contacting SolaX to update the firmware of the Pocket WiFi, there are two newer Firmware versions since yours.