wills106/homeassistant-solax-modbus #1012: [Bug]: 'ISOFAR' - 'HYD10' device group check not success

hubertozog, 2024-08-23:
### Describe the bug

After last update - probably from version 2024.08.7 to 2024.08.10. There is no any issues. Just error.

### Integration Version

2024.08.10

### Homeassistant core version

2024.8.2

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES110N

### Connection Method

Waveshare adapter RS485 to ETH

### Detailed Error Log

Rejestrator: custom_components.solax_modbus
Źródło: custom_components/solax_modbus/__init__.py:877
integracja: SolaX Inverter Modbus ([dokumentacja](https://homeassistant-solax-modbus.readthedocs.io/), [Problemy](https://github.com/wills106/homsassistant-solax-modbus/issues))
Pierwsze zdarzenie: 22 sierpnia 2024 12:42:22 (4194 zdarzenia)
Ostatnio zalogowany: 06:11:57

device group check not success

Zwer2k, 2024-09-06:
This warning can occur in the event of communication problems.
Are you using the new function to read out individual battery modules? In this case, the warning will occur more often, as the batteries often provide incorrect results. Or the selection of battery modules does not work reliably.

hubertozog, 2024-09-10:
@Zwer2k Yes, i am using it. It looks likes its read second battery pack data from time to time but usually one battery pack data its outdated.

Zwer2k, 2024-09-10:
Does the displayed serial number of the battery pack match the one in the inverter menu? (Also the order)

hubertozog, 2024-09-12:
@Zwer2k Yeah both are good, but the update time for second its preatty long... more than 2 days right now ;) it would be greate to have acces to both battteries live.

Zwer2k, 2024-09-15:
Please test it with the following settings. There may be disruptions in communication. How long is your RS485 cable?
[image]

hubertozog, 2024-09-18:
@Zwer2k  I changed setting to yours, right now all looks good! Gona give feedback after some time! Thanks!

cschlipf, 2024-11-17:
@hubertozog Can this be closed?

cc: @wills106

hubertozog, 2024-11-17:
@cschlipf @wills106 We can close for sure, all works perfect!

closed 2024-11-17 by wills106