wills106/homeassistant-solax-modbus #1015: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

stefsims, 2024-08-25:
### Describe the bug

When adding Inverter, submit gives error: "Unknown error occurred".
Works with pre-2024.8 

### Integration Version

2024.08.12

### Homeassistant core version

2024.8.2

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

YA1.0ZDAA

### Connection Method

Serial

### Detailed Error Log

2024-08-25 15:42:43.111 DEBUG (MainThread) [custom_components.solax_modbus.config_flow] _next_step_battery: returning data: {'name': 'Growatt', 'interface': 'serial', 'read_modbus_addr': 1, 'plugin': 'growatt', 'scan_interval': 10, 'read_eps': True, 'read_dcb': False, 'read_pm': True, 'read_serial_port': '/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0', 'baudrate': '9600', 'scan_interval_medium': 15, 'scan_interval_fast': 15, 'inverter_name_suffix': 'power'}
2024-08-25 15:42:43.112 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 456, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 285, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 366, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 413, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 516, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 454, in _async_step
    return await self._common_handler.async_step(step_id, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 147, in async_step
    return await self._async_form_step(step_id, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 196, in _async_form_step
    return await self._show_next_step_or_create_entry(form_step)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 228, in _show_next_step_or_create_entry
    next_step_id_or_end_flow = await form_step.next_step(self._options)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
    if user_input["support-battery"]:
       ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 'support-battery'
2024-08-25 15:42:44.030 DEBUG (MainThread) [custom_components.solax_modbus] device group read done

RKOvlesenMerkle, 2024-08-27:
I have the exact same problem. I have not been able to use any 2024.8 versions and have to stay at 2024.7.5

aen42, 2024-08-27:
Same/very similar problem adding a SolaX X3 Hybrid (which worked fine and was removed before the upgrade from 2024.05.7 as per the release note instructions in the early 2024.08 releases). On HomeAssistant OS 13.1 and core 2024.8.3.

the 'support-battery' seem to be the problem for the config flow when adding the integration according to the logs.

...
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 366, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 413, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 516, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 361, in _async_step
    return await self._common_handler.async_step(step_id, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 147, in async_step
    return await self._async_form_step(step_id, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 196, in _async_form_step
    return await self._show_next_step_or_create_entry(form_step)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 228, in _show_next_step_or_create_entry
    next_step_id_or_end_flow = await form_step.next_step(self._options)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/config_flow.py", line 193, in _next_step_battery
    if user_input["support-battery"]:
       ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 'support-battery'

RKO, 2024-08-27:
I have the exact same problem. I have not been able to use any 2024.8 versions and have to stay at 2024.5.7

stefsims, 2024-08-27:
Workaround is to add inverter as tcp/ip (with some make believe ip address) and not as serial. When integration is initializing, use the three dots and change to serial. Then restart Home Assistant 
That worked for me.

stefsims, 2024-08-27:
Reopen, because it is still a bug, even though I found a workaround

wills106, 2024-09-17:
Just released [2024.09.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.3) which hopefully fixes this error.