wills106/homeassistant-solax-modbus #1016: Solis - entity name changes after upgrading to 2024.8.12

funks81, 2024-08-25:
### Describe the bug

I've managed to upgrade the integration to 2024.8.12 from a much older version 2024.5.7. 
Unfortunately a number of the entities seem to have been renamed in the latest version - mainly to do with charge/discharge times - leading to my automations no longer working.
Examples include Timed Charge Start Hours - previously number.solisdirect_timed_charge_start_hours
has become
number.solisdirect_timed_charge_start_h

Couldn't see this mentioned anywhere - and wonder if they could be reverted to their old names to save rewriting a number of automations and dashboards.
Many Thanks

### Integration Version

2024.8.12

### Homeassistant core version

2024.8.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603105

### Connection Method

RS485

wills106, 2024-08-28:
The key needs updating to match the name. See https://github.com/wills106/homeassistant-solax-modbus/issues/949#issuecomment-2206445859 for details.

funks81, 2024-08-29:
Thanks for this - just to clarify - I need to go into config/custom_components/solax_modbus/plugin_solis.py and manually change the keys to match the names for all the entities that I'm having issues with?
Don't want to break everything as I don't really know what I'm doing!!!!

funks81, 2024-08-29:
[plugin_solis.py.zip](https://github.com/user-attachments/files/16798613/plugin_solis.py.zip)
Have been through and changed keys in plugin_solis.py to match names - have attached the file in case its useful to others.  Seems to be working okay!!!

wills106, 2024-08-29:
Thanks, I'll merge it into the branch and release it for others.

jevburchell, 2024-09-08:
> ### Describe the bug
> I've managed to upgrade the integration to 2024.8.12 from a much older version 2024.5.7. Unfortunately a number of the entities seem to have been renamed in the latest version - [...]

I had rewritten all of my automations and dashboards given the entity name changes mentioned here - specifically the timed charge hours/minutes input numbers.  After updating to 2024.09.1 this evening, I see they have all changed back to their original entity names, so my edits will need to be reversed.

viking2010, 2025-04-02:
@funks81 can you confirm if this can now be closed off?

closed 2025-09-28 by github-actions[bot]