wills106/homeassistant-solax-modbus #1017: [Bug]: "SolaX Inverter Battery Charge Upper SOC" can't be set by the Integration

mobyline, 2024-08-26:
### Describe the bug

"SolaX Inverter Battery Charge Upper SOC" can't be set by the SolaX Modbus Integration. After a few seconds it always falls back to the value set in the SolaX cloud app.

### Integration Version

2024.08.12

### Homeassistant core version

2024.8.3

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A12

### Inverter firmware versions

DSP v1.29 ARM v1.27

### Connection Method

SolaX Pocket WiFi 3.0

### Dongle firmware

3.008.10

olafseidel, 2024-08-30:
I have the same problem.
Wonder wether this is a problem from Solax or within Home Assistant.
Would be nice to be able to change SolaX Battery Charge Upper SOC as it is positive for the health of the bettery to keep the SOC low over the lifetime (this is also true for LFP batteries).

olafseidel, 2024-08-30:
Same Bug is described (but not solved) in https://github.com/wills106/homeassistant-solax-modbus/issues/713

mobyline, 2024-09-01:
Yes, my intention is indeed building up a 'battery charging management flow'. Therefore I need to set 'SolaX Battery Charge Upper SOC' from HA (node-red).

johny-mnemonic, 2024-09-21:
As a workaround, you can try setting `number.solax_battery_charge_max_current` to 0 when you want to stop battery charging.

MrJanD, 2024-09-21:
> As a workaround, you can try setting `number.solax_battery_charge_max_current` to 0 when you want to stop battery charging.

Yes, I know. But as described in https://homeassistant-solax-modbus.readthedocs.io/en/latest/automation-examples/ this comes with some drawbacks…

wills106, 2024-09-21:
Has anyone tried writing to this value when the Inverter is in "Unlocked - Advanced" instead of just the normal "Unlocked" ?
Might need a higher level of access?

wills106, 2024-09-21:
@infradom has this ever worked on your Inverter?

Xoffroad, 2024-09-21:
> Has anyone tried writing to this value when the Inverter is in "Unlocked - Advanced" instead of just the normal "Unlocked" ? Might need a higher level of access?

Yes, my inverter is set to Advanced according to your integration. But I cannot change the value solax_battery_charge_upper_soc. If I change the value in the manufacturer app, the correct value is displayed in your integration.

johny-mnemonic, 2024-09-21:
> Yes, I know. But as described in https://homeassistant-solax-modbus.readthedocs.io/en/latest/automation-examples/ this comes with some drawbacks…

Yes. I am searching for a way how to stop charging using the PowerControl feature as [described here](https://github.com/wills106/homeassistant-solax-modbus/discussions/586), but so far I was unable to persuade the inverter to do so😞

I have moved almost all of my automations to Power Control, so setting this single parameter to EEPROM once a day at max should not cause significant wear🤞

steps56, 2024-09-21:
Some month ago I've asked SolaX/Qcells about Eeprom write Limits.
They told me, its 1000000 (10x as stated by always just "assumed" values here)

I use a slider of
number.solax_battery_discharge_max_current
to cut battery draln by wallbox charging with Evcc via HA.
Same could be used by an automation and
number.solax_battery_charge_max_current
with:
If SOC is 79%, set
number.solax_battery_charge_max_current
to fixed value 0
Second automation needed, to switch back to a save 10 or 15 A charge by setting
number.solax_battery_charge_max_current
to f. i. 15 A if SOC is reaching f. i. min. SOC.
You decide at the 2. automation, which value may be your favorit min. SOC....50, 30, 20%... your choice.

This will write mostly no more as 2x a day, that's very safe to the Eeprom... if you don't use many other writing automations too 😉.

mobyline, 2024-09-21:
> As a workaround, you can try setting `number.solax_battery_charge_max_current` to 0 when you want to stop battery charging.

thx for the hint. Will give it a try.

mobyline, 2024-09-21:
> Has anyone tried writing to this value when the Inverter is in "Unlocked - Advanced" instead of just the normal "Unlocked" ? Might need a higher level of access?

Yes, but didn' work :(

DavidAntonin, 2024-09-23:
I have the same problem with my Solax X3-Hybrid G4 inverter. Ulocked mode, when I change setting it is back to value set by cloud in few seconds

richie241, 2024-09-26:
same problem with 'upper SoC' on X3 Ultra. All other stuff works well :)

Henry1-github, 2024-10-02:
I have the same problem with "Upper SOC". Please try to solve it. @PatrikTrestik

PatrikTrestik, 2024-10-02:
I'll look on it. But I have only short time 13.10-22.10 for hobby work. If not solved by that time, please someone else look on it.

wills106, 2024-10-02:
I don't think it's an integration issue nor a firmware issue.

Either the docs are incorrect, but less likely as it's the same for Gen4 & Gen5. Or it's not the register we think we need, or it needs to be used in combination with another register / function.

viktorp75, 2024-10-06:
I was able to change the value of Battery Charge Upper SOC using local API through registry value 200. To set it to 0 for example you need to send following data to inverter, where $INV_SN is serial of your inverter.

```'optType=setReg&pwd=$INV_SN&data={"num":1,"Data":[{"reg":200,"val":"0"}]}'```

Perhaps it will help to fix the issue in the integration? If not then you can make that number selector using the restful integration instead.

PatrikTrestik, 2024-10-06:
@viktorp75 this bug is for Modbus integration.
All API rated stuff should be written to https://github.com/PatrikTrestik/homeassistant-solax-http

API has completely different mapping than Midbus.

johny-mnemonic, 2024-10-06:
Did anyone try to report it as a bug to Solax? I remember in the past we had an issue with other variable on the Modbus and when I reported it to Solax I got reply, that it is known and fixed bug, which will be released in the next FW version.
So in case this really doesn't look like an issue on the integration side, we can try it. Maybe no one reported this one yet.

viktorp75, 2024-10-06:
>this bug is for Modbus integration. All API rated stuff should be written to https://github.com/PatrikTrestik/homeassistant-solax-http
> 
> API has completely different mapping than Midbus.

@PatrikTrestik 
I do understand, however I posted here so affected users can use it as a workaround before its fixed in Modbus. And also you know that it works somewhere else than cloud which would confirm that the issue is in the Modbus, perhaps vs. firmware. Btw. mine is  DSP v1.43 ARM v1.42 and I have the same problem.
I am not sure why you asked to report to your local api repo if it is not a bug?

steps56, 2024-10-06:
> @viktorp75 this bug is for Modbus integration. All API rated stuff should be written to https://github.com/PatrikTrestik/homeassistant-solax-http
> 
> API has completely different mapping than Midbus.

Totally confusing and unlogic as your api http thing is about the SolaX Charger aka Wallbox and has nothing about SolaX System Battery SOC settings.
What do you want to be asked, solved or whatever at your link, about max. SOC entity bug/not working regarding to yout Wallbox stuff ???

It's much more logic, like wills already asumed before...

steps56, 2024-10-06:
> Did anyone try to report it as a bug to Solax? I remember in the past we had an issue with other variable on the Modbus and when I reported it to Solax I got reply, that it is known and fixed bug, [...]

Hmm, may be it's walking at thin ice... as I belive, SolaX has no real interest in its open modbus connection to be used widely by anybody and may perhaps close it in future 🤷🏼‍♂️.
So I'm always one step to offline usage of everything (by just hitting one button in HA), if anything will be forced to unfree status or unwanted updates, implementing some "cut off" stuff.

I assume, its bug/error/wrong data in the modbus documentation, in conjunction by chained other entiies, like wills already assumed too.

insipiens, 2024-10-10:
changing it works on my X4, I’m on a slightly older version of this integration but probably more relevant is I did have this issue a few months back, exactly as described…. so stopped changing it but now it works. Perhaps there is some limitation Solax have imposed as suggested above

steps56, 2024-10-10:
> changing it works on my X4, I’m on a slightly older version of this integration but probably more relevant is I did have this issue a few months back, exactly as described…. so stopped changing it but now it works. Perhaps there is some limitation Solax have imposed as suggested above

Nope... still jumping back to always same value set by the Qcells/SolaX App.
Changed setting f. i. from 100 % to 80% jumps back to 100% some seconds later as always before.
Bug still exists.

olafseidel, 2024-10-10:
> Nope... still jumping back to always same value set by the Qcells/SolaX App. Changed setting f. i. from 100 % to 80% jumps back to 100% some seconds later as always before. Bug still exists.

Same with me. Always jumps back... :-(

wills106, 2024-10-12:
I have marked this as "wont fix" as it's not an Integration issue.
Someone needs to report that this register is not working to SolaX, or ask how it's meant to be used.

OrustEngineering, 2024-10-12:
from Solax: Yes, out team has confirmed that this is a known issue and we are working on fixing this ASAP. The temporary firmware will be available next Monday and the formal version will be published by the end of Oct.

Evarra, 2025-01-03:
[image]
Good News, got new firmware SOLAX X3-HYBRID G4 from solax support today.

[image]
[image]

Setting battery Charger Upper SOC works!

wills106, 2025-01-03:
Thanks @Evarra do you what Firmware you are on?
Than others know what to ask for.

steps56, 2025-01-03:
Hi,
YES its working now.

SolaX Hyb G3 at my place.
Firmware newest Support Update Version:
1.48 / (Bootloader 0.9) / 1.45

Thanks a lot, now last known "bug" is solved.

Evarra, 2025-01-03:
> Thanks @Evarra do you what Firmware you are on? Than others know what to ask for.

ARM 1.45 DSP 1.48

steps56, 2025-01-03:
> ![image](https://private-user-images.githubusercontent.com/47775538/400033111-a3f4cbd5-8c2a-4dd5-969d-29addd1bee1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I assume, you know how to update by usb in case of which one first an so on.
Why not asking for OTA Update from support to be at the save garantie side ?
Took just to next morning, till being updated by the german support, in my case.

Evarra, 2025-01-03:
> I assume, you know how to update by usb in case of which one first an so on. Why not asking for OTA Update from support to be at the save garantie side ? Took just to next morning, till being [...]

I know, but I like when my sys is not expose to China :) Full offline sys.

steps56, 2025-01-03:
> I know, but I like when my sys is not expose to China :) Full offline sys.

Okay, I fully understand.
I'm able to switch off internet access of the System/Wlan Dongle by just a switch set at HA, see screen:

[image]

Evarra, 2025-01-03:
> > > > ![image](https://private-user-images.githubusercontent.com/47775538/400033111-a3f4cbd5-8c2a-4dd5-969d-29addd1bee1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5MzI3MjIsIm5iZiI6MTczNTkzMjQyMiwicGF0aCI6Ii80Nzc3NTUzOC80MDAwMzMxMTEtYTNmNGNiZDUtOGMyYS00ZGQ1LTk2OWQtMjlhZGRkMWJlZTFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAzVDE5MjcwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA4ZWNlNTQ3ZmYxZTAyZmZkOGZmNmU2OTBiYmIwZTkzZGM5ZDNhOGUzOGQ5MWJmMTc0ZjQ5ZTE1NjgyMzVlNDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MWu1t9jsazQ8YzpyfgTYrOuRg2DWTYdwpiOVlCoLtqI) Good News, got new firmware SOLAX X3-HYBRID G4 from solax support today.
> > > > ![image](https://private-user-images.githubusercontent.com/47775538/400033431-19c01f3d-c2ea-443b-b530-a071b7d6bf8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5MzI3MjIsIm5iZiI6MTczNTkzMjQyMiwicGF0aCI6Ii80Nzc3NTUzOC80MDAwMzM0MzEtMTljMDFmM2QtYzJlYS00NDNiLWI1MzAtYTA3MWI3ZDZiZjhlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAzVDE5MjcwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYxNmNiNzhlNTY5NTM5ZjIyYjIyNGFjYmQxOGJhMDgyMWQ5MDNiMGQ2ODZlNmQyMzE5MzdlNGMwNmM1ZmRmY2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cRkJjvyWXePCs_OUP-NooocwKsHDDu1ITb9roXI5JU8) ![image](https://private-user-images.githubusercontent.com/47775538/400033685-6bea93f6-8aa7-4041-9d45-4139f1e464ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5MzI3MjIsIm5iZiI6MTczNTkzMjQyMiwicGF0aCI6Ii80Nzc3NTUzOC80MDAwMzM2ODUtNmJlYTkzZjYtOGFhNy00MDQxLTlkNDUtNDEzOWYxZTQ2NGFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAzVDE5MjcwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0MTI2YTExZDYzMzUzNzZhNjhmMDIyMzFmZjJiNTVkYmI3NjIyODNmNzdhY2M4NWVkZmExZjI2NTlhZGE3NjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nbSuelATp4zKogXYv4rlBrYQDeDRRSu0GzBLQhiQ8OM)
> > > > Setting battery Charger Upper SOC works!
> > > 
> > > 
> > > I assume, you know how to update by usb in case of which one first an so on. Why not asking for OTA Update from support to be at the save garantie side ? Took just to next morning, till being updated by the german support, in my case.
> > 
> > 
> > I know, but I like when my sys is not expose to China :) Full offline sys.
> 
> Okay, I fully understand. I'm able to switch off internet access of the System/Wlan Dongle by just a switch set at HA, see screen:
> 
> ![Screenshot_20250103_203922_Home Assistant](https://private-user-images.githubusercontent.com/67542183/400047706-ccb81491-783a-4e40-b6c1-418b30a2b5f8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5MzM1MDgsIm5iZiI6MTczNTkzMzIwOCwicGF0aCI6Ii82NzU0MjE4My80MDAwNDc3MDYtY2NiODE0OTEtNzgzYS00ZTQwLWI2YzEtNDE4YjMwYTJiNWY4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAzVDE5NDAwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJiY2Y1NTg2NmY0ZTBkZjViOWVkNWY4NTAyM2NjNTRhOGUwYTE1OGJmNzM2YmYxODUyOWU5MGM5N2U2NWFjZTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.i9WREIQt10lfx5EuF00O6F272tdUa1_tDOSwby46zeE)

[image]
I Don't think Solax Cloud like my Batterys anyway :)

steps56, 2025-01-03:
Whow... how much kWh capacity are those 3 E-car batteries in sum ?
And do you get it charged even in winter time ?
(I don't get even 6 kWh Accu getting charged fully most days since Nov. 24...)

dneuhaeuser, 2025-01-03:
where can I get the new firmware files?

steps56, 2025-01-03:
> where can I get the new firmware files?

Up to 1.42/1.41 here:
https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/79352264617?page=1

Usually Firmware will be set up OTA by the Support... if yor use the official way as you should do like 99%.

johny-mnemonic, 2025-02-16:
> > Thanks [@Evarra](https://github.com/Evarra) do you what Firmware you are on? Than others know what to ask for.
> 
> ARM 1.45 DSP 1.48

I can confirm this combination works, but my experience with the upgrade was a bit hair raising.

I asked the support to upgrade my inverter, then got no answer for multiple days and then out of the blue the inverter started to behave weird and stopped reporting all the values except those measured by the energy meter.

When I checked the Inverter physically, it reported "DSP Boot Fault IE109" and than it took us multiple days to actually bring it to healthy state. They used some intermediary DSP, with develop version, then some way older DSP/ARM combo than I originally had and then I had to completely power off the inverter shutting down whole house. Everything we tried failed for them to successfully upgrade the inverter remotely, so in the end they sent me the update DSP file to flash it via USB, which finally did it...

steps56, 2025-02-16:
Support sent you the Firmware File.. whouw 👍.
As its still now outdatet by 1.50/1.47 , I don't ask for sharing ( but maaaany would, because stuck at much older Firmware).
But if you achive to get 1.50 /1.47 by files, I will kiss your feet by sharing it here 🤩.

wills106, 2025-02-23:
Closing down as I believe setting "Battery Charge Upper SOC" is now working through a Firmware update.

closed 2025-02-23 by wills106