wills106/homeassistant-solax-modbus #1027: [Bug]: Entities unavailable after Upgrade to 2024.08.14 (IP integration)

rgrae81, 2024-08-31:
### Describe the bug

After Upgrade to 2024.08.14 from 2024.08.13 all entities went offline/unavailable. After Checking integration It only shows me the Bus-ID even it's choosen TCP. No posibility to change IP  or Port, its also not shown.

### Integration Version

2024.08.14

### Homeassistant core version

2024.8.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H3BC25

### Inverter firmware versions

DSP 1.33

### Connection Method

Solax Wifi+Lan

### Dongle firmware

1.002.09

Emlben, 2024-09-01:
The exakt same thing happened to me yeasterday - but I _did not_ upgrade hass. I am on 2024.6.2

PatrikTrestik, 2024-09-03:
Please enable debug for integration logging and post log here. 

You can disable debug logging after cca 16min (to cover one full reading cycle) and it will automaticaly offer file save.

closed 2025-09-27 by github-actions[bot]