wills106/homeassistant-solax-modbus #1028: [Bug]: Entity name changes in 2024.08.14 break update buttons for Solis inverters

jshiell, 2024-09-01:
### Describe the bug

Hi,

After upgrading to 2024.08.14 a bunch of entity names have changed, e.g. `timed_charge_start_h` became `timed_charge_start_hours`. However, this appears to have broken the update buttons (e.g. `solis_update_charge_discharge_times`). When clicking on these you now get an error referring to the old IDs, e.g. "Failed to perform the action button/press. 'timed_charge_start_h'". I presume something just got missed with the name changes.

[This appears to be the commit that caused the changes.](https://github.com/wills106/homeassistant-solax-modbus/commit/0d98722496d519ec0f5e9d9cbdd6492c8a6fdbb5)

Thanks very much for all of your work!

### Integration Version

2024.08.14

### Homeassistant core version

2024.8.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603101

### Connection Method

Waveshare

jshiell, 2024-09-01:
Updating the IDs in the button functions appears to fix it.

SzosszeNET, 2024-09-01:
Moreover it appears that change from hours/minutes to m/h been back and forth a couple times in the last couple updates...

timstephenson20, 2024-09-02:
Simple enough fix for Solis from the looks of things - have updated the parameter names in the button functions and added a PR.

closed 2024-09-05 by wills106

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this change.

simondrake, 2024-09-06:
Thanks @wills106

PR #1032 (2024-09-05): Fix for #1028 Solis button functions
Fixes to Solis value_function_timingmode(), value_function_timingmode2() & value_function_timingmode3() to reference the updated entity naming from 2024.08.14 for the Timed Charge / Discharge hours/minutes parameters. 

Closes #1028 based on my own testing with a Solis inverter.