wills106/homeassistant-solax-modbus #1029: [Bug]: setting charge/discharge times/current is temperamental

SzosszeNET, 2024-09-01:
### Describe the bug

I'm using predbat to automate charge/discharge and since the last couple updates seeing inconsistencies when predbat is setting schedules resulting in discharging the battery or not charging when expected as the time windows overlap.
Sometimes when manually settings schedules through the solax integration (and predbat disabled) the schedule changes are discarded (despite pressing the set button within seconds ) but no errors nothing that would point into the direction of some possible issues.

### Integration Version

2024.8.13

### Homeassistant core version

2024.8.3

### Inverter brand

Solis 

### Plugin used

plugin_solis.py

### Serial prefix

603102

### Connection Method

modbus

bryanajwells, 2024-09-02:
I'm seeing the exact same thing.  Sometimes the times are passed through 100% no issues, othertimes bits are missed.
For me my setup is as per https://github.com/alienatedsec/solis-ha-modbus-cloud

Method of reproducing:
I add    timed_charge_start_hours, timed_charge_start_minuts, timed_charge_end_hours and timed_charge_end_minutes to a dash board along with the button Inverter Update Charge/Discharge Times

I set the start/end time to   01:02  03:04
press the update button

Then dialing into the soliscloud interver control page I check the charge date/times match what was set.

Then set the times to something else (usually 00:00 / 00:00), check the soliscould again

Repeat, eventually one or more of the 4x digits doesn't set.

I think there may be a corrolation to the 0/5/10/15 etc. minute intervals of the hour, but there is no real pattern.

Note: not related to predbat, just really exposed by it.

timstephenson20, 2024-09-03:
Our setup here is different in that I have HA talking Modbus direct to the datalogger and ditched SolisCloud but when getting automations running I did notice that unless I included a delay of at least 2 seconds between setting timed charge hour/minute/current values and triggering the update button, it didn't always work.  Sounds like a similar problem to what you're seeing in that times wouldn't always be set properly. 

Don't know if you immediately trigger the update after setting values but might be worth trying adding a delay to see if that helps. 

[image]

bryanajwells, 2024-09-03:
@timstephenson20 - My controls are on a dashboard 
[image]
so there would be a natural delay between setting all 4 numbers and pressing the update button.  However how much time between the last number and the update would be quite small I guess.  I'll experiment with adding a delay.

timstephenson20, 2024-09-03:
Ah, was working on the basis you were running this in a script/automation somewhere.  It was quite fussy if memory serves - think I'd assumed at the time the issue was down to the natural latency between submitting values in HA, HA sending them to the Datalogger and the Datalogger then writing them to the inverter - give it a couple of seconds for everything to happen and the problem went away.

SzosszeNET, 2024-09-03:
Yes but bottom line it became an issue with the latest releases, went back all the way I could 2024.04.3 (potentially could go higher) and not experiencing the issue.

j47996, 2024-09-10:
Also seeing this; used to work before a recent version update

bryanajwells, 2024-10-08:
@wills106 
This is still happening on 2024.09.5
Attached is an image of predbat setting the start hours to 23:30, but the inverter not getting this value and triggering a charge.  The 3rd picture shows the hours as recorded by this integration which initially shows 23, then a few mins later changes to 0.
The minutes was uploaded successfully.
Note: I manually cancelled the charge.

To me it seems like the values are accepted by the integration, but some how this occasionally fails to be pushed to the inverter.

I have enabled debugging and hope to catch a future event. It appears from my recent leccy bill that this occurs fairly fequently
[image]
[image]
[image]

bryanajwells, 2024-10-08:
> Also seeing this; used to work before a recent version update

Do you know which version it was deffo working in?

simondrake, 2024-10-09:
I've not seen this issue since I moved over to using helpers and a script. I had to create eight helpers in the format of `input_number.solar_{charge,discharge}_{end,start}_{hours,minutes}`; for example `input_number.solar_charge_start_hours`. 

I then created a `Set Charge/Discharge` script that uses those helpers, to set the integration entities and then "presses" the `update charge/discharge times` button:

```yaml
alias: Set Charge/Discharge
sequence:
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_charge_start_hours') | int)}}"
    target:
      entity_id: number.solis_timed_charge_start_hours
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_charge_start_minutes') | int)}}"
    target:
      entity_id: number.solis_timed_charge_start_minutes
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_charge_end_hours') | int)}}"
    target:
      entity_id: number.solis_timed_charge_end_hours
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_charge_end_minutes') | int)}}"
    target:
      entity_id: number.solis_timed_charge_end_minutes
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_discharge_start_hours') | int)}}"
    target:
      entity_id: number.solis_timed_discharge_start_hours
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_discharge_start_minutes') | int)}}"
    target:
      entity_id: number.solis_timed_discharge_start_minutes
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_discharge_end_hours') | int)}}"
    target:
      entity_id: number.solis_timed_discharge_end_hours
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ (states('input_number.solar_discharge_end_minutes') | int)}}"
    target:
      entity_id: number.solis_timed_discharge_end_minutes
  - action: button.press
    metadata: {}
    data: {}
    target:
      entity_id: button.solis_update_charge_discharge_times
description: ""
```

I've then changed my dashboard to use this mechanism and have included the integration entities so I can keep a close eye and ensure they don't reset to the zero value.

[image]

This obviously only enables the first charging/discharging to be used, but if/when I need the second/third variants I'll create more helpers and another script to handle it.

Like I said, I've been using this for a couple days now and haven't seen a single issue of resetting. Whether that's indicative of the **actual** issue, perhaps it needs to be set in a specific amount of time or it fails, I don't know at this time.

simondrake, 2024-10-11:
Just a note that I've still seen no recurrence of the issue since using the helper/script method. I've now created helpers so I can control the `_2` and `_3` variants and have added some colour conditions to the **actual** solis entities such that, if they don't match the helpers, they turn red so alert me that something has gone wrong.

[image]
[image]

bryanajwells, 2024-10-12:
Hi @wills106 
Had a major issue this morning where the minutes didn't set properly.  I had debug on. Below is the trace from the relavent time.
The charge time was set as 00:46 -> 00:00.  I manually set it to 00:00 -> 00:00.  But this didn't stick.

This log seems to agree.  I've boldened the interesting numbers.

2024-10-12 07:04:20.131 INFO (MainThread) [custom_components.solax_modbus.number] *** **local data written timed_charge_start_minutes: 0**
2024-10-12 07:04:20.172 INFO (SyncWorker_0) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'sync_rtc_offset': 0, 'timed_charge_start_hours': 0, **'timed_charge_start_minutes': 0.0,** 'timed_charge_end_hours': 0, 'timed_charge_end_minutes': 0, 'timed_discharge_start_hours': 0, 'timed_discharge_start_minutes': 0, 'timed_discharge_end_hours': 0, 'timed_discharge_end_minutes': 0, 'timed_charge_start_hours_2': 0, 'timed_charge_start_minutes_2': 0, 'timed_charge_end_hours_2': 0, 'timed_charge_end_minutes_2': 0, 'timed_discharge_start_hours_2': 0, 'timed_discharge_start_minutes_2': 0, 'timed_discharge_end_hours_2': 0, 'timed_discharge_end_minutes_2': 0, 'timed_charge_start_hours_3': 0, 'timed_charge_start_minutes_3': 0, 'timed_charge_end_hours_3': 0, 'timed_charge_end_minutes_3': 0, 'timed_discharge_start_hours_3': 0, 'timed_discharge_start_minutes_3': 0, 'timed_discharge_end_hours_3': 0, 'timed_discharge_end_minutes_3': 0}
2024-10-12 07:04:20.176 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-10-12 07:04:21.081 INFO (MainThread) [custom_components.solax_modbus.button] writing solax_solis button register 43143 value [('timed_charge_start_hours', 0), **('timed_charge_start_minutes', 46),** ('timed_charge_end_hours', 0), ('timed_charge_end_minutes', 0), ('timed_discharge_start_hours', 0), ('timed_discharge_start_minutes', 0), ('timed_discharge_end_hours', 0), ('timed_discharge_end_minutes', 0)]
2024-10-12 07:04:21.081 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0xa887: [0, 46, 0, 0, 0, 0, 0, 0]
2024-10-12 07:04:23.537 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-10-12 07:04:25.175 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-10-12 07:04:28.538 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter

bryanajwells, 2024-10-12:
Another instance
Seems the same accepting the new values then a read happens causing the eventual write to fail.

2024-10-12 05:30:07.905 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written timed_charge_start_hours: 23
2024-10-12 05:30:09.485 WARNING (MainThread) [homeassistant.components.media_player] Updating soundtouch media_player took longer than the scheduled update interval 0:00:10
2024-10-12 05:30:11.922 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written timed_charge_start_minutes: 30
2024-10-12 05:30:12.530 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-10-12 05:30:15.067 INFO (SyncWorker_5) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'sync_rtc_offset': 0, 'timed_charge_start_hours': 23.0, 'timed_charge_start_minutes': 30.0, 'timed_charge_end_hours': 4, 'timed_charge_end_minutes': 30, 'timed_discharge_start_hours': 0, 'timed_discharge_start_minutes': 0, 'timed_discharge_end_hours': 0, 'timed_discharge_end_minutes': 0, 'timed_charge_start_hours_2': 0, 'timed_charge_start_minutes_2': 0, 'timed_charge_end_hours_2': 0, 'timed_charge_end_minutes_2': 0, 'timed_discharge_start_hours_2': 0, 'timed_discharge_start_minutes_2': 0, 'timed_discharge_end_hours_2': 0, 'timed_discharge_end_minutes_2': 0, 'timed_charge_start_hours_3': 0, 'timed_charge_start_minutes_3': 0, 'timed_charge_end_hours_3': 0, 'timed_charge_end_minutes_3': 0, 'timed_discharge_start_hours_3': 0, 'timed_discharge_start_minutes_3': 0, 'timed_discharge_end_hours_3': 0, 'timed_discharge_end_minutes_3': 0}
2024-10-12 05:30:15.071 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-10-12 05:30:17.530 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-10-12 05:30:19.487 WARNING (MainThread) [homeassistant.components.media_player] Updating soundtouch media_player took longer than the scheduled update interval 0:00:10
2024-10-12 05:30:19.569 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-10-12 05:30:19.965 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written timed_charge_end_hours: 23
2024-10-12 05:30:22.531 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-10-12 05:30:23.983 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written timed_charge_end_minutes: 59
2024-10-12 05:30:27.532 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-10-12 05:30:28.003 INFO (MainThread) [custom_components.solax_modbus.button] writing solax_solis button register 43143 value [('timed_charge_start_hours', 0), ('timed_charge_start_minutes', 46), ('timed_charge_end_hours', 23.0), ('timed_charge_end_minutes', 59.0), ('timed_discharge_start_hours', 0), ('timed_discharge_start_minutes', 0), ('timed_discharge_end_hours', 0), ('timed_discharge_end_minutes', 0)]
2024-10-12 05:30:28.004 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0xa887: [0, 46, 23, 59, 0, 0, 0, 0]
2024-10-12 05:30:29.487 WARNING (MainThread) [homeassistant.components.media_player] Updating soundtouch media_player took longer than the scheduled update interval 0:00:10
2024-10-12 05:30:29.567 INFO (SyncWorker_40) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'sync_rtc_offset': 0, 'timed_charge_start_hours': 0, 'timed_charge_start_minutes': 46, 'timed_charge_end_hours': 23.0, 'timed_charge_end_minutes': 59.0, 'timed_discharge_start_hours': 0, 'timed_discharge_start_minutes': 0, 'timed_discharge_end_hours': 0, 'timed_discharge_end_minutes': 0, 'timed_charge_start_hours_2': 0, 'timed_charge_start_minutes_2': 0, 'timed_charge_end_hours_2': 0, 'timed_charge_end_minutes_2': 0, 'timed_discharge_start_hours_2': 0, 'timed_discharge_start_minutes_2': 0, 'timed_discharge_end_hours_2': 0, 'timed_discharge_end_minutes_2': 0, 'timed_charge_start_hours_3': 0, 'timed_charge_start_minutes_3': 0, 'timed_charge_end_hours_3': 0, 'timed_charge_end_minutes_3': 0, 'timed_discharge_start_hours_3': 0, 'timed_discharge_start_minutes_3': 0, 'timed_discharge_end_hours_3': 0, 'timed_discharge_end_minutes_3': 0}

bryanajwells, 2024-10-12:
Additional note:  I have the integration set with the polling interval(s) all set to 5.  When it was 15 it was more reliable but not 100%.
So I guess as a workaround a bigger polling interval will reduce the chance of a clash.  
I'd say this is a race condition, which might impact other inverter types as well.

SzosszeNET, 2024-10-12:
Hmm, I always thought those intervals are meant to be different?
Mine was set I'm pretty sure 1, 5 and 10 and when wiped and reinstalled the integration set them to 5, 10 and 15. Will bump them to 10, 15 and 20 now just to see if it makes a difference. Is there anything on the purpose or best practice for the 3 polling intervals?

bryanajwells, 2024-10-12:
> Hmm, I always thought those intervals are meant to be different? Mine was set I'm pretty sure 1, 5 and 10 and when wiped and reinstalled the integration set them to 5, 10 and 15. Will bump them to [...]

Just checked the source - all 3 have the default of 15 seconds.  Dont know what the difference is though

wills106, 2024-10-13:
Only Sofar and Solinteg take advantage of the different poll speeds.
That's why by default they are all at 15.

mrpp, 2024-11-11:
I too have the issue where setting the discharge times in HA, they don't always update on the Solis when pressing button.solis_modbus_update_charge_discharge_times

SzosszeNET, 2024-12-12:
Sadly this issue still occurs time to time despite upping the polling interval to 45s.
Of course had happened yesterday with the 96p/kWh pricing. Ouch :(

bryanajwells, 2025-01-31:
Hi @wills106, any chance you could look into this please?  It tripping a number of us up and can cause significant issue when the 'read' occurs during setting.

I dont know how to fix the issue within HA (script reads values from sensor, whilst user is changing the values in UI), but one thing which does occur to me is that if the charge/discharge times are modified, then dont update from the inverter read for a configurable time.  read 'timeout' reset each time the time is modified.  When the 'apply' button is pushed this cancels the timer.

I'm able to assist with testing.

closed 2025-09-27 by github-actions[bot]