wills106/homeassistant-solax-modbus #1034: [Bug]:  Sofar - HYD 5KTL-3PH - Only 82 entities created missing e.g. Battery info and PV power

vacuum007, 2024-09-03:
### Describe the bug

Hi,

the connection works in principle and I get updated values on some entities like "Active Power Output Total", but it seems a lot of entities are missing. Especially all battery related data like the SOC.

Help is very much appreciated. 
Thanks for the great work in any case!

Cheers
vacuum007

### Integration Version

2024.08.14

### Homeassistant core version

2024.8.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SH1051

### Connection Method

LSE-3  (Firmware: ME_0D_270A_1.09)

### Detailed Error Log

No entries in log, when disabling "Enable Readout" in configuration.

### Additional context

Inverter Software Version: 0V1100610V1100610V110061
Hardware: V003
"Enable Readout" is not ticked in configuration as then no entity created at all.

These are the created entities:
sensor.sofar_active_power_load_sys
sensor.sofar_active_power_output_l1
sensor.sofar_active_power_output_l1n
sensor.sofar_active_power_output_l2
sensor.sofar_active_power_output_l2n
sensor.sofar_active_power_output_l3
sensor.sofar_active_power_output_total
sensor.sofar_active_power_pcc_l1
sensor.sofar_active_power_pcc_l1n
sensor.sofar_active_power_pcc_l2
sensor.sofar_active_power_pcc_l2n
sensor.sofar_active_power_pcc_l3
sensor.sofar_active_power_pcc_total
sensor.sofar_active_power_pv_ext
sensor.sofar_apparent_power_output_total
sensor.sofar_apparent_power_pcc_total
sensor.sofar_current_output_l1
sensor.sofar_current_output_l1n
sensor.sofar_current_output_l2
sensor.sofar_current_output_l2n
sensor.sofar_current_output_l3
sensor.sofar_current_pcc_l1
sensor.sofar_current_pcc_l1n
sensor.sofar_current_pcc_l2
sensor.sofar_current_pcc_l2n
sensor.sofar_current_pcc_l3
sensor.sofar_export_energy_today
sensor.sofar_export_energy_total
sensor.sofar_fault_1
sensor.sofar_fault_10
sensor.sofar_fault_11
sensor.sofar_fault_12
sensor.sofar_fault_2
sensor.sofar_fault_3
sensor.sofar_fault_4
sensor.sofar_fault_5
sensor.sofar_fault_6
sensor.sofar_fault_7
sensor.sofar_fault_8
sensor.sofar_fault_9
sensor.sofar_grid_frequency
sensor.sofar_hardware_version
sensor.sofar_heatsink_temperature_1
sensor.sofar_heatsink_temperature_2
sensor.sofar_import_energy_today
sensor.sofar_import_energy_total
sensor.sofar_inverter_temperature_1
sensor.sofar_inverter_temperature_2
sensor.sofar_load_consumption_today
sensor.sofar_load_consumption_total
sensor.sofar_module_temperature_1
sensor.sofar_module_temperature_2
sensor.sofar_power_factor_output_l1
sensor.sofar_power_factor_output_l2
sensor.sofar_power_factor_output_l3
sensor.sofar_power_factor_pcc_l1
sensor.sofar_power_factor_pcc_l2
sensor.sofar_power_factor_pcc_l3
sensor.sofar_reactive_power_output_l1
sensor.sofar_reactive_power_output_l2
sensor.sofar_reactive_power_output_l3
sensor.sofar_reactive_power_output_total
sensor.sofar_reactive_power_pcc_l1
sensor.sofar_reactive_power_pcc_l2
sensor.sofar_reactive_power_pcc_l3
sensor.sofar_reactive_power_pcc_total
sensor.sofar_serial_number
sensor.sofar_software_version
sensor.sofar_solar_generation_today
sensor.sofar_solar_generation_total
sensor.sofar_system_state
sensor.sofar_rtc
sensor.sofar_voltage_l1
sensor.sofar_voltage_l2
sensor.sofar_voltage_l3
sensor.sofar_voltage_line_l1
sensor.sofar_voltage_line_l2
sensor.sofar_voltage_line_l3
sensor.sofar_voltage_phase_l1n
sensor.sofar_voltage_phase_l2n
sensor.sofar_waiting_time

ehrenhauser, 2024-09-03:
Same for me but i just got 14 entities, all the others got unavailable. Changed back to 2024.05.7 since this version still works perfect

Zwer2k, 2024-09-06:
The serial number of your inverter starts with SH1051, this is the reason for missing entities. In the plugin_sofar.py, SH1... is not stored as a hybrid inverter.
According to the Sofar documentation, devices with a serial number starting with SH1 are not hybrid inverters. For HYD, the serial number should start with SP1 or SP2.
@ehrenhauser can you please give the first letters of your HYD. Is it also an HYD 5KTL-3ph? 
[image]

ehrenhauser, 2024-09-07:
Hello Zwer2k,
my inverter serial is SP1ES115NBD096 and I have a HYD 15KTL-3PH. Honestly i'm not shure wether i'm doing the updating correct. I just install the update (2024.09.1) and then restart ha as recommended. After that some battery related entities are unavailable and in the integrations i realize a 2nd device with this unavailable entities. deleting the integrations an trying to set up a new one does not work but i don't know wether i'm doing the set up the correct way.

vacuum007, 2024-09-07:
> The serial number of your inverter starts with SH1051, this is the reason for missing entities. In the plugin_sofar.py, SH1... is not stored as a hybrid inverter. According to the Sofar [...]

Thanks, maybe they changed the serial number system?
It should be a 5ktl-3ph, according to the sticker and the Solarman data.
[image]

vacuum007, 2024-09-07:
Thanks, maybe they changed the serial number system in new units? 
According to the sticker and the Solarman data it is a 5ktl-3ph.

[image]

Zwer2k, 2024-09-07:
> Hello Zwer2k,
> thank you very much for your reply. my inverter serial is SP1ES115NBD096 and I have a HYD 15KTL-3PH. Honestly i'm not sure wether i'm doing the updating correct. I just install the update (2024.09.1) and then restart ha as recommended. After that some battery related entities are unavailable and in the integrations i realize a 2nd device with this unavailable entities. deleting the integrations an trying to set up a new one does not work but i don't know wether i'm doing the set up the correct way. 

Please check whether the inverter prefix is ​​set in the new version. First of all, deactivate the battery reading, the function does not work with LSE-3 yet

[image]

Zwer2k, 2024-09-07:
> Thanks, maybe they changed the serial number system in new units? 
> According to the sticker and the Solarman data it is a 5ktl-3ph.
> 
> [image]
> 

Can you please give 2-3 more letters from the serial number (not complete). I want to see if the model can be clearly identified from the serial number in order to display it in the integration.

vacuum007, 2024-09-07:
Sure, Serial starts with: SH1051005KE...

Zwer2k, 2024-09-07:
OK, thank you. Unfortunately this is not entirely clear, 05 appears twice in your serial number, one of which should stand for 5kW. For HYD models >= 10kW, characters 7+8 stand for the kW, here they are different. This is not relevant to the problem, but it would still be interesting to see what the serial number looks like for others in order to be able to map it correctly.
It would be nice if other Sofar inverter owners could post the first 10 characters of the serial number + the model designation of their inverters here so that the identification can be implemented as reliably as possible.

ehrenhauser, 2024-09-07:
Thank you, my problem is solved with this recommendation. I just had to change some names of entities which my dashboard was referring to. Thanks again

vacuum007, 2024-09-07:
> OK, thank you. Unfortunately this is not entirely clear, 05 appears twice in your serial number, one of which should stand for 5kW. For HYD models >= 10kW, characters 7+8 stand for the kW, here they [...]

Would you know what to change in my individual config file to tell the integration SH* is a hybrid inverter? 
Until there is a general solution.

wills106, 2024-09-07:
I thought the SH1 was on the older register set?
Must be used on both?

https://github.com/wills106/homeassistant-solax-modbus/blob/8bd8b7b6a9da3ad910c908a9a50925488ecc778c/custom_components/solax_modbus/plugin_sofar.py#L4026-L4027

Change PV to HYBRID

Zwer2k, 2024-09-07:
> I thought the SH1 was on the older register set? Must be used on both?
> 
> https://github.com/wills106/homeassistant-solax-modbus/blob/8bd8b7b6a9da3ad910c908a9a50925488ecc778c/custom_components/solax_modbus/plugin_sofar.py#L4026-L4027
> 
> Change PV to HYBRID

Ich vermute, dass jemand dies einfach aus der Sofar-Dokumentation übernommen hat. Die Dokumentation ist möglicherweise nicht korrekt. In der Dokumentation steht SOFAR 3...6KTLM-G3. Kennen Sie jemanden, der dieses Modell hat?

Zwer2k, 2024-09-07:
> Would you know what to change in my individual config file to tell the integration SH* is a hybrid inverter? Until there is a general solution.

These lines
https://github.com/wills106/homeassistant-solax-modbus/blob/8bd8b7b6a9da3ad910c908a9a50925488ecc778c/custom_components/solax_modbus/plugin_sofar.py#L4026-L4027

replace with the following
```
elif seriesnumber.startswith("SH1"):
        invertertype = HYBRID | X3 | GEN | BAT_BTS  # HYD5...8KTL-3P
        self.inverter_model = "HYD5...8KTL-3P"```#
```
Option BAT_BTS in combination with option  
‘Read battery modules’ in plugine settings, allows extended readout of the BTS information. However, it does not currently work with LSE-3.

vacuum007, 2024-09-08:
> These lines
>
> https://github.com/wills106/homeassistant-solax-modbus/blob/8bd8b7b6a9da3ad910c908a9a50925488ecc778c/custom_components/solax_modbus/plugin_sofar.py#L4026-L4027
>
> replace with the [...]

This did the trick! Thank you so much :-)

wills106, 2024-09-15:
Closing down as complete.

closed 2024-09-15 by wills106