wills106/homeassistant-solax-modbus #1037: [Bug]: Invalid content of *_data.json file crashes integration.

PatrikTrestik, 2024-09-03:
### Describe the bug

If file exists, but contains garbage or is empty, function _loadLocalData_ crashes irrecoiverably.

### Integration Version

2024.08.14

### Homeassistant core version

all

### Inverter brand

not relevant

### Plugin used

plugin_solax.py

### Serial prefix

all

### Inverter firmware versions

not relevant

### Connection Method

Modbus TCP

### Dongle firmware

N/A

### Detailed Error Log

2024-09-03 12:42:52.125 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x118 : remaining_battery_capacity
2024-09-03 12:42:52.129 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 684, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 869, in async_read_modbus_registers_all
    await self._hass.async_add_executor_job(self.loadLocalData)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 355, in loadLocalData
    loaded = json.load(fp)
             ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

PatrikTrestik, 2024-09-03:
Working on fix. 
Change in exception handling will help.

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this change.

wills106, 2024-09-07:
Closing as corrected.

closed 2024-09-07 by wills106