wills106/homeassistant-solax-modbus #1039: [Bug]: Solis - timed charge entities removed in latest release?

simondrake, 2024-09-03:
### Describe the bug

After updating to `2024.08.14` the following entites become unavailable and are reported as no longer provided by the integration.

I don't see this in the release notes, so is this expected?

```
  - entity: number.solis_timed_charge_start_hours
  - entity: number.solis_timed_charge_start_minutes
  - entity: number.solis_timed_charge_end_hours
  - entity: number.solis_timed_charge_end_minutes
  - entity: number.solis_timed_discharge_start_hours
  - entity: number.solis_timed_discharge_start_minutes
  - entity: number.solis_timed_discharge_end_hours
  - entity: number.solis_timed_discharge_end_minutes
```

### Integration Version

2024.08.14

### Homeassistant core version

2024.8.3

### Inverter brand

Solis

### Plugin used

plugin_growatt.py

### Serial prefix

603105

### Connection Method

USR-TCP232-304 Serial RS485 to TCP/IP Ethernet Server Converter Module

### Additional context

I've reverted back to `2024.08.13` and the entities come back.

bryanajwells, 2024-09-03:
Hi @simondrake, Im using 2024.08.14, the names of the entities is ....hours and ....minutes.  The previous version had them has ...h and ...m
It has an issue with not being able to update the settings https://github.com/wills106/homeassistant-solax-modbus/issues/1028
so reverting to an earlier release is certainly worth it unless you want to manually fix.

simondrake, 2024-09-04:
@bryanajwells  I'm seeing different behaviour. I've been using `_hours` and `_minutes` for as long as I can remember but the latest release makes them unavailable. 

They are replaced with entities such as `_hours_2_2`, `_3_2` etc etc

simondrake, 2024-09-04:
In the attached screenshot - you can see I have `solis_timed_charge_start_hours` and `_2`/`_3` variants, on version `2024.08.13`. 

The unavailable entries are what is introduced in `2024.08.14` - `_4`, `_2_2`, and `_3_2` - the non-numbered `_start_hours` are no longer available on update. 

[image]

@wills106 - is this an expected change?

lazy-pete, 2024-10-23:
having the same problem. rolled back to 2024.10.3 and still unavailable

slate-1, 2024-11-05:
I also have this issue. sn prefix 6031. Solis 

Connection -  Serial RS485 to TCP/IP

Very recently added integration - excellent by the way.

App works as expected for most entities, App accepts charge times (not using discharge times) & updates json file when send button pressed but inverter does not receive updated schedule. Assuming Inverter should show new values, fairly recent user of home assistant & integration but happy to help if I can for debugging/testing.

As a recent installer this aspect has never worked for me, first install contained the _hours, _mins entities. Not .h, .m etc.

Also inverter model not shown - have seen other issues reporting this.

Thank you for the excellent integration. If you need any additional information please let me know.

closed 2024-11-05 by simondrake

bryanajwells, 2024-11-05:
@simondrake  - Are you sure this should be closed.  I'm still experiencing the issue on the latest version.