wills106/homeassistant-solax-modbus #1041: Add Battery Voltage Cell Difference

Salada71, 2024-09-04:
### Description

Add new sensor.
I do not know how to push to branch. I am not programmer.

### Suggested Solution

const.py:
```
def value_function_battery_voltage_cell_difference(initval, descr, datadict):
    return  datadict.get('cell_voltage_high', 0) - datadict.get('cell_voltage_low',0)
```

plugin_solax.py:
```
SolaXModbusSensorEntityDescription(
        name = "Battery Voltage Cell Difference",
        key = "battery_voltage_cell_difference",
        value_function= value_function_battery_voltage_cell_difference,
        native_unit_of_measurement = UnitOfElectricPotential.VOLT,
        device_class = SensorDeviceClass.VOLTAGE,
        state_class = SensorStateClass.MEASUREMENT,
        rounding = 3,
        allowedtypes = AC | HYBRID | GEN4 | GEN5,
    ),
```

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this addition.

wills106, 2024-09-07:
Closing as added in change.

closed 2024-09-07 by wills106