wills106/homeassistant-solax-modbus #1042: [Bug]: POWER_VOLT_AMPERE_REACTIVE depricated from HA Core 2025.9

PistolaPeteUK, 2024-09-05:
### Describe the bug

Just as a heads up I upgraded to HA 2024.9 and got the following warning:

POWER_VOLT_AMPERE_REACTIVE was used from solax_modbus, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'solax_modbus' custom integration
 

### Integration Version

2024.08.14

### Homeassistant core version

2024.09

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603104

### Connection Method

Waveshare TCP adaptor

### Detailed Error Log

Logger: homeassistant.const
Source: helpers/deprecation.py:197
First occurred: 10:36:48 (1 occurrences)
Last logged: 10:36:48

POWER_VOLT_AMPERE_REACTIVE was used from solax_modbus, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'solax_modbus' custom integration

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this change.

wills106, 2024-09-07:
Closing as corrected.

closed 2024-09-07 by wills106