wills106/homeassistant-solax-modbus #1043: Solis - S6-EH3P8K-H-EU -  entities for string 3 & 4 missing

kowi-k, 2024-09-05:
### Describe the bug

Hi, 
the integration is working fine so far, but the entities for string/mppt 3+4 are missing.
Register for eg pv current 3 (33054) reads the correct data. 

[image]

### Integration Version

2024.8.14

### Homeassistant core version

2024.8.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

103305

### Connection Method

waveshare rs485

### Detailed Error Log

log is empty

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with MPPT3 & MPPT4.

kowi-k, 2024-09-06:
Thanks a lot! Works perfect!

wills106, 2024-09-15:
Closing down as complete.

closed 2024-09-15 by wills106