wills106/homeassistant-solax-modbus #1044: [Bug]: battery_charge_max_current limits

Nei1c, 2024-09-07:
### Describe the bug

Hi

The battery_charge_max_current & battery_discharge_max_current entities appear to have maximum values of 30 amps configured within the integration.  The inverter I'm using (SolaX X3 Ultra 30) has a maximum charge and discharge current of 60 amps (30 + 30 on the two battery stacks).  I'd lowered the charge current setting from 60 and now cant set it higher than 30 so my maximum charge rate is half of what it should be.  Can you raise the maximum setting for these two entities to 60?

Thanks

### Integration Version

2024.09.01

### Homeassistant core version

2024.9.1

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H3BC3

### Inverter firmware versions

DSP 0 ARM 16.1

### Connection Method

SolaX Pocket Wifi+LAN

### Dongle firmware

1.003.11

Nei1c, 2024-09-10:
Issue resolved with release of 2024.09.2

Many thanks

adicorra, 2024-12-03:
Hello,

I have the exact same problem with a SOLIS RHI-5K-48ES-5G inverter.
Max value of number solis_battery_charge_current is 62.5A and the 5kW inverter takes 100A.
I lowered the value and now cannot reset it back to more than 62A.

Thanks,

[image]