wills106/homeassistant-solax-modbus #1046: [Bug]: Battery Charge Upper SOC

matteo710, 2024-09-08:
### Describe the bug

I have a configuration with 2 solax inverters in parallel. I cannot set the Battery Charge Upper SOC parameter. After forcing it, it always returns to the initial value.
thaks
Matteo

### Integration Version

2024.9.1

### Homeassistant core version

2024.9.1

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

DSP v1.41 ARM v1.40

### Connection Method

Solax Pocket Wifi 3.0

### Dongle firmware

3.017.02

### Detailed Error Log

2024-09-08 14:11:38.953 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-08 14:11:38.965 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-08 14:11:39.695 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-09-08 14:11:39.779 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-09-08 14:11:42.836 INFO (MainThread) [custom_components.solax_modbus.number] writing S1 battery_charge_upper_soc number register 224 value 100 after div by readscale 1 scale 1
2024-09-08 14:11:48.954 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-08 14:11:48.966 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-08 14:11:49.733 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-09-08 14:11:50.004 DEBUG (MainThread) [custom_components.solax_modbus] device group read done

steps56, 2024-09-08:
This bug has already been called at min. 2 other bug reports here... but sadly never been commented or even fixed ba wills106 😪.
Would be nice/normal/polite, to get any feedback about this bug by wills106.

wills106, 2024-09-11:
> sadly never been commented or even fixed by wills106 😪

I'm sorry but it's difficult to fix bugs on an Inverter I don't own.

steps56, 2024-09-11:
> > sadly never been commented or even fixed by wills106 😪
> 
> I'm sorry but it's difficult to fix bugs on an Inverter I don't own.

I undestand now, thanks for answer.

Additionally one other oddity:
I had sometimes a look into solax.py and there are the 2 temperature sensors listet, but just one ist showing up at your HA interation.
It's the higher value of the 2 shown in my screen here.
This higher value is definitely the Board temp. but labeled as inverter temp (heat sink temp) and showing up at HA.
Showing at Screen ist "Solar Live", an inofficial SolaX Android App.
These 2 Temp. values are:
Left one with lower temp. = heat sink (not shown at HA)
Right one with higher temp= board temp (shown at HA)

May be, ist just some mismatch of modbus number of those 2 entities... or 🤷🏼‍♂️.
It's not mainly important, but would be nice to have in HA like at the App

[image]

richie241, 2024-09-18:
+1
same here - cant update the 'SolaX Inverter Battery Charge Upper SOC'
after a few seconds, it jumps back up 100%

Hardware: X3 Ultra + 2x HV-S21

wills106, 2024-09-21:
Closing down as a duplicate of #713 & #1017

Use #1017 as the main discussion.

closed 2024-09-21 by wills106