wills106/homeassistant-solax-modbus #1047: solis rai-3k-48es-5: buttons for setting Minimum SOC, charge/discharge time and on/off do not work,

stefeis, 2024-09-09:
### Describe the bug

changing the charge/discharge times by using the "update button for charge/discharge does not change the values => after some seconds the old values are visible. same at setting up the "minimum SOC" => no changing. or do I have to "safe" the changes somehow? I also was not able to turn off/on the inverter. the different modes I didn´t try, but I guess they are also not possible to change. I removed S3-stick to cloud, now just connected by internal wlan to my network and HA.

### Integration Version

2024.09.1

### Homeassistant core version

2024.9.1

### Inverter brand

Solis rai-3k-48es-5

### Plugin used

plugin_solis.py

### Serial prefix

010262

### Connection Method

RTU over TCP

### Detailed Error Log

after setting the minimum SOC I also pressed RTC update.
2024-09-09 11:12:20.490 INFO (MainThread) [custom_components.solax_modbus.number] writing Solis battery_minimum_soc number register 43011 value 20 after div by readscale 1 scale 1
2024-09-09 11:12:20.850 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-09 11:12:25.852 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2024-09-09 11:12:28.536 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2024-09-09 11:12:30.415 INFO (MainThread) [custom_components.solax_modbus.button] writing Solis button register 43000 value [('_uint16', 24), ('_uint16', 9), ('_uint16', 9), ('_uint16', 11), ('_uint16', 12), ('_uint16', 30)]

viking2010, 2025-04-02:
@stefeis can you confirm if this is still happening? I have seen this issue a few times with my system, but in my automations, I get them to "press" the button twice over a period of about 30 seconds and this seems to work and negate the issue of values not being set. I wondered if it was related to some sort of latency or wifi issue, so I'm now using a network cable rather than wifi with my S3 stick. It seems to be more stable for me, but still sometimes get the problem.

stefeis, 2025-04-04:
I mounted a totaly new PV-system based on DEYE and will sell the solis rai. therefore I`ve no further information...

viking2010, 2025-04-06:
@wills106 I believe this can be closed off.

stefeis, 2025-04-07:
yes, sorry