wills106/homeassistant-solax-modbus #1050: [Bug]: Battery 2 Temperature is 0

rgrae81, 2024-09-11:
### Describe the bug

Battery 2 Temperature is 0
[image]

Checked Temp at Solax, in my case Bat2 has the same temp as Bat1 but at HA its showing still 0

### Integration Version

2024.09.02

### Homeassistant core version

2024.9.1

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H3BC25

### Inverter firmware versions

DSP 1.33

### Connection Method

Solax Wifi+Lan

### Dongle firmware

1.002.09

wills106, 2024-09-11:
Are they connected in series ie both on Bat1 or on two separate connections?

gt008, 2024-09-12:
Maybe update inverter firmware. I have same/similar set up. H3BC25 2 Batteries (2xH23.2 connected in batt port 1 and 2) but on version 1.4 DSP. I get temp for both batteries. Just colder where I am!! 8degC now!

rgrae81, 2024-09-17:
Batteries are connected to Ultra on BAT Port 1 and BAT Port 2

Maybe Firmware-Update will do it. Where can I get a complete FW-Set inclusive BMS ?

gt008, 2024-09-17:
Easiest is to call Solax. They can update the firmware remotely. Otherwise
need to download from a third party site (search for link) which is not
always 100% up to date and load onto a USB to update manually.

Are you using Solax H5.8 t-bat batteries?

rgrae81, 2024-09-18:
Ok, so i will call solax.
I'm using HV-S3.6.

wills106, 2024-10-12:
Did you get a Firmware update?

rgrae81, 2024-10-24:
Firmware update on Bats or Inverter?
Inverter display
DSP 2.07
ARM 2.03
WR Code: 007.01 009.07 09

Batteries display on Solax Cloud Version 003.02

My Bat2 Temp is on Register 0x132 as here shown
[https://github.com/wills106/homeassistant-solax-modbus/pull/1106](url)

wills106, 2024-11-02:
v3.35 of the docs must have had an error in then them

rgrae81, 2024-11-02:
@wills106 should I send you over 3.37?

wills106, 2024-11-02:
If you can

rgrae81, 2024-11-02:
@wills106 
Here it is.
[(Solax)Hybrid X1&X3-G4 ModbusTCP&RTU V3.36-English_240611.pdf](https://github.com/user-attachments/files/17606735/Solax.Hybrid.X1.X3-G4.ModbusTCP.RTU.V3.36-English_240611.pdf)

rgrae81, 2024-11-02:
Here it is.

[(Solax)Hybrid X1-IES&X3-IES&ULTRA ModbusTCP&RTU V3.37-English_240314.pdf](https://github.com/user-attachments/files/17606737/Solax.Hybrid.X1-IES.X3-IES.ULTRA.ModbusTCP.RTU.V3.37-English_240314.pdf)

closed 2024-11-02 by rgrae81