wills106/homeassistant-solax-modbus #1052: [Bug]: 'IGrowatt' - 'MOD 6KTL3-XH (BP)' On-grid Battery Discharge Stop SOC does not adjust

FISK-PUMA, 2024-09-12:
### Describe the bug

When adjusting "EMS Discharging Stop SOC"  only " off-grid Battery Discharge Stop SOC" changes on my Growatt inverter/battery, "On-grid Battery Discharge Stop SOC" still the same.  (10%)
[image]
[image]

### Integration Version

2024.09.01

### Homeassistant core version

2024.9.1

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

JCM0DB

### Connection Method

RS485-USB

TonyMathiesen, 2024-09-13:
Which battery system are you having? 
I use the MOD 8KTL3-XH (BP) and APX HV battery system, and I only have the one discharging stop SOC in the user interface. 
What does your setup list under property?
[image]

FISK-PUMA, 2024-09-13:
I also have a APX HV battery system. My BMS is a [https://parametre.online/0voz58nhg7659vhuerxj.png] Growatt APX 98020-P1

But firmware in the battery/BMS was updated by growatt last week maybe that is way?

See battery information i picture

BR.
Mads

[cid:f72fa267-c3d0-4676-b705-a2faae9e16a4]
[cid:99691f2f-d2c0-473e-8a2a-172dd16ceb2c]

image.png (view on web)<https://github.com/user-attachments/assets/741f7bc2-91bd-4384-ab19-1cadc93252dd>

TonyMathiesen, 2024-09-13:
Hi Mads. 
Can you try sending the picture again? I can only see a 1x1 pixel.
That could very well be due to a new firmware. I got mine installed a month ago. Will check about the firmware and see if I can get it added - as it makes really good sense to have the option to set an on-/offgrid SOC level.

FISK-PUMA, 2024-09-14:
Hi Tony

I hope pictures are better now. 🙂

Br. Mads

TonyMathiesen, 2024-09-14:
Hi Mads. 
There is no picture attached, maybe it is because it is a reply via mail.
BR Tony.

FISK-PUMA, 2024-09-14:
HI Tony 
I now  using github hope thats better .
[image]
[image]
BR. Mads

TonyMathiesen, 2024-09-14:
Hi Mads. 
Now I can see the pictures 👍🏼 Thanks. 
I can see you are using different version. I will contact Growatt or my installer to get the firmware update, or do you have the files? I believe that I might have found the register that we need to add to the plugin to get both in the integration. 
BR Tony

FISK-PUMA, 2024-09-14:
Hi Tony

No i dont have the files, Growatt did the update, remote.

That sounds good 🙂

BR. Mads

TonyMathiesen, 2024-09-17:
Hi Mads
Growatt did the firmware upgrade remotely yesterday. I have found the register and created a pull request #1056 to add the on-grid level. You can check the pull request if you want to test it. 

In the mail correspondence with Growatt the supporter mentioned:
> Keep in mind that these options are proportional to the Depth of Discharge (DoD) of your battery, set at 90%. Please do not change these values by more than 20%, as it could cause your batteries to over-discharge.

I asked a bit into it, and in general he would not recommend to adjust the stop SOC levels as it also impacts other parameters. One should be cautious to avoid stressing the battery cells, and one should monitor the parameters after every adjustment.

wills106, 2024-09-18:
Do you know what version number of the Firmware is required?
Also do we expose it in HA?

TonyMathiesen, 2024-09-18:
Hi @wills106 
Appears the firmware control version needs to be "ZBDB-0008", before I had "ZBDB-0006".
It is possible to read it out, however the last part seems to be handled differently than what is listed in the documentation. 
[image]

If read out directly from the registers, two first is ascii and the last one is just the number. 
[image]

From app: 
[image]

closed 2024-09-19 by wills106

wills106, 2024-09-19:
Just released [2024.09.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.4) with the change.

FISK-PUMA, 2024-09-20:
Hi Tony

Very nice that you found a solotion.  🙂

Info from Growatt dosent sound that good, right now I only want to adjust the levet by 20% between 30-10%.
 it is becuse i want to save some power on the battery, so that I can use that in the expensive morning hours. Maybe i should find another way to fix this. 😉

Thank you very much for your help🙂
BR.
Mads

PR #1056 (2024-09-19): Fix #1052 For Growatt MOD TL3-XH
Fix #1052 For Growatt MOD TL3-XH: 
Add EMS Discharging Stop SOC (on grid) from register 3067. It does requires newer firmware version. Tested with version "DN1.0/DNAA065101/ZBDB-0008".