wills106/homeassistant-solax-modbus #1057: [Bug]: 'Growatt' - 'MOD10KTL3-X' certain Entities zeroed

Zvolli, 2024-09-18:
### Describe the bug

As agreed with @TonyMathiesen I create a new ticket for this issue. 
I am missing following entities that are zeroed. Growatt web UI and App does have these working fine so it's not a com error between meter and inverter.

[image]

Let me know if you need further information

### Integration Version

2024.09.3

### Homeassistant core version

2024.9.1

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

Serial number: DL1.0ZBAA

### Connection Method

ESP32 MODBUS gateway RS485

### Detailed Error Log

[home-assistant_2024-09-18T05-53-01.122Z.log](https://github.com/user-attachments/files/17039024/home-assistant_2024-09-18T05-53-01.122Z.log)

wills106, 2024-09-19:
Follows on from [conversation here](https://github.com/wills106/homeassistant-solax-modbus/issues/735#issuecomment-2356066498)

Onlinepeter, 2024-09-19:
@wills106 Is this also the issue why we only have 1 control for the MOD xx KTL3-XH inverters - I was expecting more controls to be available
[image]

wills106, 2024-09-19:
@Onlinepeter 
What does your Inverter return for the Serial?
https://github.com/wills106/homeassistant-solax-modbus/blob/b79a65c06457f41b9b6f4b687cb824bf5b39828d/custom_components/solax_modbus/plugin_growatt.py#L5026-L5042

TonyMathiesen, 2024-09-19:
@Zvolli I will look into cleanup the plugin for the MOD TL3-X non-hybrid version as we discussed and then you can test. Will see if I can get some done in the weekend.

TonyMathiesen, 2024-09-19:
@Onlinepeter Have you checked under Configuration, they may be hidden by default.

Onlinepeter, 2024-09-19:
@wills106 sorry I am not sure how to find out.
If I press the link I only get this on my screen
[image]

or this if I look at the device
[image]

Zvolli, 2024-09-19:
Excellent I appreciate your help 😊👍

Mvh. Michael Svoldgaard

tors. 19. sep. 2024 kl. 19.52 skrev TonyMathiesen ***@***.***

Onlinepeter, 2024-09-19:
> @Onlinepeter Have you checked under Configuration, they may be hidden by default 😉

eg Priority that I could expect is a control I see as read only (I can change it shrough Growatts own app
[image]

TonyMathiesen, 2024-09-19:
@Onlinepeter Currently the priority is not supported, however I'm working on a way to get the registers exposed to the integration. It will not so pretty / user friendly but it least it will give the possibility to control via some helpers and an automation. It does require a bit more work, not to impact other users of the integration. 

I only got my system a month ago, so I'm also new to Growatt. 

So I will first try to clean up some the things for TL3-X and TL3-XH in general.

Onlinepeter, 2024-09-19:
@TonyMathiesen thanks for the information.
Please let me know if I can do anything on my side. As long as the no of controls are limited, I can not use it for much, so I can be used as tester if needed :)

TonyMathiesen, 2024-09-22:
@Zvolli Can you send a picture of which values are available from the Web UI or App?

Zvolli, 2024-09-22:
Sure. There are a few screenshots attached

Sendt fra Outlook til iOS<https://aka.ms/o0ukef>

TonyMathiesen, 2024-09-22:
@Zvolli No pictures in the post.. Can you resent using github not mail?

Zvolli, 2024-09-22:
Sure here you go
[image]
[image]
[image]
[image]
[image]
[image]

TonyMathiesen, 2024-09-22:
@Zvolli Any of the parameters that shows the values for import, export, etc.? If we find the parameters, then we might be able to add it 😉

Zvolli, 2024-09-22:
reverseActive_Energy is export to grid
reverseActiveTodayEnergi is daily total export
positiveActiveTodayEnergu is daily total import
Reactive_power could be current load (a bit unsure if it is this one)

TonyMathiesen, 2024-09-22:
Could maybe be some of these.. Not sure. 

[image]

I could try add them as test, and then you could see if they read out anything. Mine does not have this register range, so I cannot test to see.

Zvolli, 2024-09-22:
Let's try it :)

TonyMathiesen, 2024-09-22:
@Zvolli Try this.. https://pastebin.com/0ceqFGnf replace your plugin_growatt.py file with this. Keep the old, then you can quickly revert back again. 
It is with the cleanup which is described in #1066, so it will remove all the not supported entities for TL3-X .. And then I have added the 4 above as a test.. You need to restart HA when changed the file.

Zvolli, 2024-09-22:
ok. 
 TEST_Nominal Output Reactive power seems to be a copy of Output power so we don't need this.
TEST_Output apparent power seems to be same as above but with a bit lower wattage. I can't seem to match this from the UI so it's a bit unknown.
TEST_Reactive power generation is zeroed.
TEST_Real Output Reactive power seems to toggle around 30-50W and does not match anything else I can compare it with

Zvolli, 2024-09-22:
[image]

TonyMathiesen, 2024-09-22:
@Zvolli So that didn't gave us much 😒.. This is the Modbus documentation I have used so far [https://pdfcoffee.com/growatt-inverter-modbus-rtu-protocol-ii-v1-24-english-new-pdf-free.html](url)
The MOD TL3-X is listed directly which register sets are used, but assumes it uses the same as the other TL3-X versions - as it does not appair to have any of the 3xxx registers. 

[image]
Hense I believe we need to look in the input register range 0-124, 125-249 (From page 47). Or the holding register range 0-124, 125-249 (From page 9).

Zvolli, 2024-09-23:
Unfortunately not, and yes we need to look at the 0-124, 125-249 range.

wills106, 2024-09-23:
Released [2024.09.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.5) with the clean-up changes.

TonyMathiesen, 2024-09-23:
@Zvolli I looked through the registers again. It doesn't appear to be any of the ones in that range. I wonder if these data are available in the inverter or if they are only calculated or stored on Growatt's server through the ShineX or how you have it connected.
It looks like the parameters you have on the screenshots are from your smart meter - are there any of the parameter on the inverter that haves the values?

TonyMathiesen, 2024-10-15:
@Zvolli Shouldn't we close this request for now? We can always reopen or create a new if new information arises. 
Have a great day 🙂

rmk-sys, 2024-10-15:
@TonyMathiesen

1. monitor version is ZEBA-5.

2. I'd like to have „grid_power_va“ with the right scaling (1617,5) and unit (VA), and not:

[image]

playing with that entity, didn’t succeed.

Zvolli, 2024-10-15:
Well not really. It’s not solved. If it’s unsolvable then it can be closed. The data is there so it’s a bit weird that the registers are not available

Sendt fra Outlook til iOS<https://aka.ms/o0ukef>

TonyMathiesen, 2024-10-15:
@Zvolli Understand. Then when you have the time, have a look of the parameters on the inverter, to see if some of them have the needed information. The screenshots you shared earlier was with parameters from your smart meter, and currently (to my knowledge) we are only able to communicate with the inverter.

Zvolli, 2024-10-15:
But the inverter collects this and forward it to growatt cloud so it should be possible but maybe only by intercepting data from the data logger

Sendt fra Outlook til iOS<https://aka.ms/o0ukef>

TonyMathiesen, 2024-10-15:
Could very well be, that it is only intercepting from the data logger. You could try contacting Growatt support to ask if it is possible to readout from the modbus on the inverter.

HellixFix, 2025-01-14:
have a MOD 10KTL3-XH(BP) and also can't find "PositiveActiveDayEnergy" - daily sum import from the grid as displayed in the cloud (read from the smartmeter by the inverter)
I manually browsed all registers, but didn't find correct data anywhere

Growatt support here in CZ is very uncommunicative, information is insufficient.....
[image]

HellixFix, 2025-01-16:
> @ HellixFix to my humble oppinion the entity you need is: today_s_grid_import which is provided by Solax Inverter Modbus.

there is little difference - does not count energy when inverter sleeps (as battery is discharged to limit and there is no sun)

[image]

efoverskov, 2025-04-08:
I'm seeing the same issue when inverter is sleeping (MOD10KTL3-XH inverter) - data is sent to growatt cloud but not picked up by this integration. I experimented with Solar Assistant at one point, and it seems they can collect the data correctly using the same modbus interface, which suggests that the data must be available.

closed 2025-09-21 by github-actions[bot]