wills106/homeassistant-solax-modbus #1058: [Bug]: latest update - no entities

szyper-sakrona, 2024-09-18:
### Describe the bug

With latest update my Solax inverter has no active entities. Latest HA version.

### Integration Version

2024.09.3

### Homeassistant core version

Core 2024.9.2 Supervisor 2024.09.1

### Inverter brand

Solax g4

### Plugin used

plugin_solax.py

### Serial prefix

H34A06

### Inverter firmware versions

DSP v1.41 ARM v1.40

### Connection Method

Pocekt wifi plus

### Dongle firmware

latest

wills106, 2024-09-20:
Please read [detailed-error-log](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)

Also [i-have-lost-my-entities](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)

Without an error log there isn't anything we can look at to see what the issue is.

tanith01, 2024-09-23:
Happened to me as well on a Solax G4.

Here's my error logs to hopefully give you guys more info:

**Log Details (ERROR)**
Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7335
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 22:19:28 (3 occurrences)
Last logged: 22:20:24

SolaX: cannot find any serial number(s)

**Log Details (ERROR)**
Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7599
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 22:19:28 (3 occurrences)
Last logged: 22:20:24

unrecognized inverter type - serial number : unknown

**Log Details (ERROR)**
Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:546
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 22:19:18 (10 occurrences)
Last logged: 22:20:39

Cancel send, because not connected!

**Log Details (WARNING)**
Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:69
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 22:19:18 (10 occurrences)
Last logged: 22:20:39

SolaX: attempt to read serialnumber failed at 0x0 data: None
SolaX: attempt to read serialnumber failed at 0x300 data: None
SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 546, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

**Log Details (WARNING)**
Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:81
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 22:19:18 (10 occurrences)
Last logged: 22:20:39

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed
SolaX: reading serial number from address 0x1a10 failed; other address may succeed

tanith01, 2024-09-23:
I fixed my issue by following [i-have-lost-my-entities](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30) this link. (Firmware upgrade of the Pocket Wifi 3.0)

closed 2025-08-28 by wills106