wills106/homeassistant-solax-modbus #1067: [Bug]: 'Growatt' - 'Inverter Mode TL3-Sl' Modbus Error: [Input/Output] ERROR: No response received after 6 retries

rafalkur, 2024-09-22:
### Describe the bug

Hello,
Im using 2024.09.4 and trying to configure it with my growatt (via waveshare eth converter) for the first time. Initialization phase last forever, after configuring again with same settings im getting success box, but device is not added.
in logs:
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.051 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for button
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/button/init.py", line 73, in async_unload_entry
return await component.async_unload_entry(entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.052 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for number
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/number/init.py", line 134, in async_unload_entry
return await component.async_unload_entry(entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.052 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for select
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/select/init.py", line 109, in async_unload_entry
return await component.async_unload_entry(entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.053 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/sensor/init.py", line 136, in async_unload_entry
return await component.async_unload_entry(entry)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 201, in async_unload_entry
raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-09-22 21:55:14.053 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry growatt for solax_modbus
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 819, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/init.py", line 184, in async_unload_entry
hass.data[DOMAIN].pop(entry.options["name"])
KeyError: 'growatt'
2024-09-22 21:55:25.654 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
File "/config/custom_components/solax_modbus/plugin_growatt.py", line 64, in async_read_serialnr
inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=6)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/init.py", line 546, in async_read_holding_registers
resp = await self._client.read_holding_registers(address, count, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 6 retries
2024-09-22 21:55:25.655 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: reading firmware number from address 0xbb9 failed; other address may succeed
2024-09-22 21:55:25.655 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read growatt 0xbb9 firmware number before potential swap: None
2024-09-22 21:55:25.655 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: cannot find firmware version, even not for other Inverter
2024-09-22 21:55:25.655 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : unknown
2024-09-22 21:55:25.655 INFO (MainThread) [custom_components.solax_modbus] next inverter check in 10sec
2024-09-22 21:55:35.656 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: trying to determine inverter type

### Integration Version

2024.09.4

### Homeassistant core version

2024.9.2

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

EHE2A1

### Connection Method

RS485 - ETH, Waweshare adapter

### Additional context

this is first time integration, it never worked before

wills106, 2024-09-23:
Is this a PV only Inverter or Hybrid?

If it's PV only or Hybrid with a discharged battery, I am guessing the Inverter had shut it's self down as the time was 21:55 according to your logs.

The Integration requires the Inverter to be powered via PV for the initial connection to the Inverter.

rafalkur, 2024-09-23:
this is PV only, no batteries. good point with inverter being shut but i was testing it whole day, this is now:

2024-09-23 15:15:27.245 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-09-23 15:15:27.246 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
(...)
2024-09-23 15:23:03.337 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Growatt for solax_modbus
(...)
2024-09-23 15:40:29.403 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt inverter type - firmware version : dhaa0101    
2024-09-23 15:40:39.404 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: trying to determine inverter type
2024-09-23 15:40:39.452 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt 0x9 firmware number before potential swap: dhaa0101

wills106, 2024-09-23:
Released [2024.09.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.5) adding in your Inverter. It may still not work as I don't know what register set this Inverter uses. So may need further work.

rafalkur, 2024-09-23:
upgraded my Growatt to latest version (8914) and logs looks different:

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 72, in async_read_serialnr
    res = decoder.decode_string(14).decode("ascii")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 9: ordinal not in range(128)
2024-09-23 16:04:22.821 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-09-23 16:04:22.854 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-09-23 16:04:22.893 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-09-23 16:04:22.894 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-09-23 16:04:22.894 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-09-23 16:04:32.950 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: ReadHoldingRegistersResponse (7)
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 72, in async_read_serialnr
    res = decoder.decode_string(14).decode("ascii")
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 9: ordinal not in range(128)
2024-09-23 16:04:32.950 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-09-23 16:04:32.984 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-09-23 16:04:33.014 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-09-23 16:04:33.014 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-09-23 16:04:33.014 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-09-23 16:04:43.069 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: ReadHoldingRegistersResponse (7)
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 72, in async_read_serialnr
    res = decoder.decode_string(14).decode("ascii")

rafalkur, 2024-09-23:
[image]

if you mean the setting on the first page, yes, i set growatt each time. with [2024.09.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.5) initialization takes forever

2024-09-23 16:52:22.532 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-09-23 16:52:22.533 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-09-23 16:52:24.445 ERROR (MainThread) [homeassistant.components.binary_sensor] rpi_power: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 724, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1302, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/rpi_power/binary_sensor.py", line 53, in update
    value = self._under_voltage.get()
            ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2024-09-23 16:52:54.507 WARNING (zeroconf-ServiceBrowser-_googlecast._tcp-116) [pychromecast.dial] Failed to determine cast type for host <unknown> (<urlopen error _ssl.c:983: The handshake operation timed out>) (services:{MDNSServiceInfo(name='Onkyo-TX-NR686-3235188ca2ccb389be9fb33ae070cf44._googlecast._tcp.local.')})
2024-09-23 16:55:39.605 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:55:49.654 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:55:59.704 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:56:09.753 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:56:19.803 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:56:29.853 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:56:39.902 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:56:49.952 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:00.001 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:10.051 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:20.100 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:30.150 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:40.200 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:57:50.249 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:00.299 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:10.349 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:20.397 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:30.446 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:40.496 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:58:50.546 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:59:00.595 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
2024-09-23 16:59:10.645 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : dhaa0101    
[... cut ...]
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 172, in async_setup_entry
    await hub.async_init()
  File "/config/custom_components/solax_modbus/__init__.py", line 313, in async_init
    await asyncio.sleep(10)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 665, in sleep
    return await future
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

TonyMathiesen, 2024-09-23:
Maybe a stupid question.. Have you restarted HA after updating to 2024.09.5?

rafalkur, 2024-09-23:
> Maybe a stupid question.. Have you restarted HA after updating to 2024.09.5?

many times

rafalkur, 2024-09-24:
I made a configuration again and collected pcap trace with debug message taken same time. It seems converter and HA are talking on modbus level. IP of converter is 192.168.0.9 and ha 192.168.0.220. Please unzip and open with wireshark to see what growatt is replying to ha.

[modbus-tcp.zip](https://github.com/user-attachments/files/17111972/modbus-tcp.zip)
[home-assistant_solax_modbus_2024-09-24T09-19-24.154Z.log](https://github.com/user-attachments/files/17111974/home-assistant_solax_modbus_2024-09-24T09-19-24.154Z.log)

wills106, 2024-09-24:
If you replace `DHA `with `dha`

https://github.com/wills106/homeassistant-solax-modbus/blob/e132ec0a1282a2b20924c638a1088bb374857d5d/custom_components/solax_modbus/plugin_growatt.py#L5011

Does it start working?

rafalkur, 2024-09-24:
wow, bingo! :-) it shows me 1 device and 43 entities just after changing this parameter and reboot. Thanks a lot

btw, my inverter is 10kW so in the description of this line maybe the range should be 10-22: # PV TL3-SL 15-22kW #1067  ?

wills106, 2024-09-24:
Ok, I didn't realise `startswith()` was case sensitive.

I only had a quick look for a datasheet and noticed it mentioned 15-22kW. I'm guessing the datasheet I found was out of date.

rafalkur, 2024-09-24:
For some reason no parameter values are coming back from inverter, like sensor data or diagnostics:
Grid Current L1 unknown
Grid Current L2 unknown
Grid Current L3 unknown
Grid Frequency unknown
Grid Power L1 unknown
etc ....

same as general parameters:
Firmware Version unknown
Inverter Temperature unknown
IPM Inverter Temperature unknown
Serial Number unknown

[home-assistant_solax_modbus_2024-09-24T11-30-48.338Z.log](https://github.com/user-attachments/files/17113907/home-assistant_solax_modbus_2024-09-24T11-30-48.338Z.log)

wills106, 2024-09-24:
Unfortunately your Inverter isn't listed in the Modbus docs. So it's a bit of guess work to see what works best.

https://github.com/wills106/homeassistant-solax-modbus/blob/e132ec0a1282a2b20924c638a1088bb374857d5d/custom_components/solax_modbus/plugin_growatt.py#L5011

You could try changing `GEN` to `GEN2` or `GEN3` to see if it makes it better or worse. Just remember to keep DHA lowercase.

Other option is for you to reach out to Growatt and see if there are any specific docs for your Inverter.

rafalkur, 2024-09-24:
I played with GEN2 and GEN3. This parameter causes new entities to appear in sensors section but the values are same unknown.  If you would consider adding decoding schema for this Growatt inverter im sharing with you the doc which contains its modbus data format. Thanks for great support.

[Growatt PV Inverter Modbus RS485 RTU Protocol V3 16.pdf](https://github.com/user-attachments/files/17115511/Growatt.PV.Inverter.Modbus.RS485.RTU.Protocol.V3.16.pdf)

wills106, 2024-09-24:
Ok that looks different enough from the Current GEN-GEN3 and SPF Registers.

rafalkur, 2024-10-04:
@wills106  any chance in near future this Growatt will work with this integration?

wills106, 2024-10-12:
Can you give a screen shot of which Entities return values? Or is everything Unknown?

wills106, 2024-10-12:
Can you try [2024.10.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.10.2b1) and see if you get more entities returning values.

wills106, 2024-10-13:
I have just released [2024.10.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.10.2) with a few more entities.

rafalkur, 2024-10-14:
Hello
I have installed latest, 2024.10.3 and i will check tomorrow under light, its dark now.

rafalkur, 2024-10-15:
looks like it doesnt read parameters yet: 
AC Power to Grid L1 unavailable
AC Power to Grid L2 unavailable
AC Power to Grid L3 unavailable
AC Power to Grid Total unavailable
AC Power to User L1 unavailable
AC Power to User L2 unavailable
AC Power to User L3 unavailable
AC Power to User Total unavailable
Battery Charge Power unavailable
Battery Combined Power unavailable

wills106, 2024-10-16:
Can you provide a screenshot of the Integration page where it shows what does and doesn't work?

wills106, 2024-10-16:
> AC Power to Grid L1 unavailable
> AC Power to Grid L2 unavailable
> AC Power to Grid L3 unavailable
> AC Power to Grid Total unavailable
> AC Power to User L1 unavailable
> AC Power to User L2 unavailable
> AC Power to User L3 unavailable
> AC Power to User Total unavailable

Just checked the code, you shouldn't have any of those entities, they are for a different Inverter type.

You are best, removing the Inverter from the Integration page. Restarting HA and adding the Inverter in again.

Then see what does and doesn't work.

rafalkur, 2024-10-17:
Hello,
after removing latest version and installing 2024.10.2b1
[image: image.png]

the available entities still show unknown:
[image: image.png]

serial number is incorrect: Serial number: dhaa0101

nothing about growat or solax keyword in debug.

modbus config:
[image: image.png]
[image: image.png]

cheers

wills106, 2024-10-21:
Replying via email with pictures doesn't work.
You need to add them through the GitHub website.

rafalkur, 2024-10-23:
hi Wills106,
sure, via git:
after removing latest version and installing 2024.10.2b1
[image]

the available entities still show unknown:
[image]

serial number is incorrect: Serial number: dhaa0101

nothing about growat or solax keyword in debug.

modbus config:

[image]
[image]

closed 2025-09-21 by github-actions[bot]