wills106/homeassistant-solax-modbus #1070: [Bug]: 'Solis' - 'RHI-3.6K-48ES-5G' - Unable to connect/reading SN from address 0x80ec failed/serial unknown

viking2010, 2024-09-29:
### Describe the bug

Hi

Since upgrading from v2024.05.07 to the latest version v2024.09.05, as well as doing an upgrade of HA to v2024.09.01, data ceases to be retrieved periodically until I have reloaded the integration.  My inverter SN starts with 160F522, and I am using the Solis Data logger S2-WL-ST with LAN connection. LAN connectivity is solid and no drop outs on the device are detected at the time when it fails to connect and read data from the logger. The multiple errors I am getting are:

- Error doing job: Fatal error: protocol.data_received() call failed. (None)
- Cancel send, because not connected!
- Unable to decode response Modbus Error: Unknown response 96
- Failed to connect [Errno 111] Connect call failed ('192.168.0.12', 502)
- Detected blocking call to open with args
- Unable to connect to Inverter at 192.168.0.12:502
- Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.12:502]
- Solis: reading serial number from address 0x80ec failed; other address may succeed
- solis: cannot find serial number, even not for other Inverter
- unrecognized solis inverter type - serial number : unknown

I will attach the logs to this ticket which will provide further details to all the above errors. 

I understand that you cannot have both modbus and Solis cloud at the same time, but I have spotted that the inverter was being picked up on the Solis Cloud platform as 'online' instead of it showing 'offline' as it would do normally. In an attempt to prevent the Solis cloud picking up the inverter, I have removed the plant and data logger from the Solis Cloud platform and I've had to raise a ticket with Solis to remove the inverter. Weirdly, I can't remove it myself and it's still collecting data even though I've removed the logger and plant from their system. 

I also rolled back from the latest Solax Modbus integration to previous one that was working for me. It initially seemed more stable, but still getting the same issues. For the time being I've left it on v2024.05.07. What I haven't tried is rolling back to a previous version of HA but may do so over the course of the next week, but I suspect this all has something to do with the fact that the Solis Cloud is trying to take priority over the modbus connectivity, but as I am unsure, I thought I'd log this ticket. 

### Integration Version

2024.05.07

### Homeassistant core version

2024.09.01

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

160F52

### Connection Method

 S2-WL-ST (using LAN Connection NOT WiFi)

### Detailed Error Log

Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:147
First occurred: 28 September 2024 at 17:56:22 (2 occurrences)
Last logged: 28 September 2024 at 20:45:01

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1017, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 302, in data_received
    self.datagram_received(data, None)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 336, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 207, in callback_data
    self.framer.processIncomingPacket(data, self._handle_response, slave=0)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/base.py", line 139, in processIncomingPacket
    self.frameProcessIncomingPacket(single, callback, slave, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 91, in frameProcessIncomingPacket
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
[... 90 lines cut ...]
First occurred: 06:17:43 (61 occurrences)
Last logged: 09:59:10

solis: reading serial number from address 0x80ec failed; other address may succeed

This error originated from a custom integration.

===================================================================================

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2431
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 06:17:43 (61 occurrences)
Last logged: 09:59:10

solis: cannot find serial number, even not for other Inverter
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2457
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 06:17:43 (61 occurrences)
Last logged: 09:59:10

unrecognized solis inverter type - serial number : unknown

viking2010, 2024-10-01:
Just an update to the above:

- Reverting to the older previously working version has made no difference. 
- Removing the plant, datalogger and inverter from the solis site has made no change.
- The Data logger still tries to contact the Solis url's despite not having any information being processed by them anymore.
- Adding in a block for the Solis urls that the datalogger tries to connect to seems to cause more issues. When they're blocked, the logger weirdly eventually drops off the network. 

I am going to re-install the latest version of the integration again, but this is either a dodgy data logger which only seems to have occurred after an upgrade, or it's a change that's been made in the HA version that's caused this. 

All very odd, so if anyone has any ideas, please let me know.

viking2010, 2024-10-03:
I have now rolled back to my previous HA OS version from 2 weeks ago, and everything is working fine so far, so it must be something that's changed in HA that's causing this rather than the integration itself. Does anyone have any idea what might have changed that's modbus related in the latest HA version?

For reference, I'm now running HA v2024.8.3 and integration version 2024.05.7. Looks like I'll be stuck on that version for a bit!

viking2010, 2024-10-04:
Further investigation with Solis has found that they did an upgrade of the data logger on 23rd Sept, around the same time I started getting issues. They stated that previous F/W versions would freeze the data logger when local tcp comms were happening at the same time as Solis Cloud.  They say that the latest version resolves this issue. However, given that it's working ok in an older version of HA and an older Solax integration version, does point the finger more at HA or the Integration being the issue. I'm going to clone a copy of my HA VM as a dev server and see if I get the same issues again. Will report back on here when done.

wills106, 2024-10-05:
You have probably got the same issue as #1063 

It's possible the datalogger still isn't fully Modbus compliant. It's known that pyModbus doesn't like non compliant devices.

> General exception: unpack requires a buffer of 2 bytes

Looks like the Integration is receiving an incorrect response size?

Although I have noticed [pyModbus v3.7.1](https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.7.1) has corrected a max read size for registers. I don't know if that's related or not? You could try manually updating HA pyModbus version to that?

viking2010, 2024-10-07:
Hi, I'm going to close off as as mentioned in my last update, I believe the issues have stemmed from a firmware upgrade that Solis made to the data logger. They updated the firmware again which broke the dongle, then eventually managed to downgraded it back to the original firmware so I'm now going to look at upgrading HA OS again and see how it goes.

closed 2024-10-07 by viking2010