wills106/homeassistant-solax-modbus #1078: [FR]: Support Solax DataHub 1000 as RS485 modbus interface

fxstein, 2024-10-04:
### Description

Add support for the Solax DataHub 1000 as a primary RS485 to Ethernet bridge with additional functionality.

In addition of not needing the Solax WLAN adapter or any other aftermarket RS485 adapter, the DataHub provides additional functionality and the ability to connect to all Solax devices through a single hub. That includes multiple inverters and EV chargers.

### Suggested Solution

The Solax DataHub 1000 can be access via Modbus through its Ethernet port. 

https://kb.solaxpower.com/solution/detail/2c9fa4148a514f09018ab04027290860

By offsetting the modbus addresses it allows direct access to all attached Solax devices through a single interface.

This could allow the integration to find, identify and auto create all the attached devices through a single configuration flow. This would simplify and harden the setup for such deployments. 

### Alternatives

This is an alternative to the current aftermarket modbus adapters being used with the added benefit of being able to access all Solax devices through a single connection

### Additional Context

This would simplify parallel mode deployments or installs with one or more EV chargers.

wills106, 2024-10-05:
Do you have access to the Modbus documentation for the DataHub?
As that table listed is in a different order compared to reading from the Inverter directly.

fxstein, 2024-10-05:
I am working on getting the detailed documentation for the Datahub. Will report back as I make progress.

JMyrng, 2024-11-02:
I have tested the direct connection with the Datahub via Modbus. The information that you see for each connected inverter is in this table, nothing additional. 

[image]

With the hybrid, activePower also includes battery power, so there is no way to distinguish between PV and battery. 

This means no information about 
- SOC
- Battery Power
- individual stings

Generally much less information than directly via Modbus on the inverter. Unfortunately. 

The DataHub lacks a proper API or connection via MQTT.

closed 2025-09-19 by github-actions[bot]