wills106/homeassistant-solax-modbus #1080: Solis - start_hours sometimes resets to zero and I cannot track down why

simondrake, 2024-10-05:
### Describe the bug

At this point I’ve no idea what is happening, or under what circumstances, and after weeks of trying different things I’m completely at a loss so raising it here as a last resort. 

I keep seeing weird behaviour when setting start hours/minutes and end hours/minutes, to charge my batteries at the most optimal times. 

For example, last night I set them to charge between 02:30 (start_hours = 02, start_minutes = 30) and 04:00 (end_hours = 04, end_minutes = 00) then hit the update charge/discharge times button. 

A few minutes go past, I open HA again, and the start time has reset to 00:00, meaning it would have charged at the wrong times. 

It doesn’t happen every time, but often enough that it’s annoying and has resulted in charging at the wrong times. 

I’ve tried the three different charge/discharge entities (_2 & _3), have looked through the logs for anything related (there’s nothing), tried spamming the update charge/discharge button several times in a row, and tried doing it all very quickly, and somewhat slowly, but I cannot find a pattern. 

I’ve even tried removing and reinstalling the integration from scratch. 

I feel like this started when I started using predbat, but I’ve since removed the addon and all config and it’s still happening. 

I’ve looked through the open issue but can’t see anybody with the same issue. 

Do you have any idea why this would happen or where to look to try and narrow it down? 

Feel like I’m losing my sanity with this issue 😂

### Integration Version

2024.09.05

### Homeassistant core version

2024.9.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603105

### Connection Method

USR-TCP232-304 Serial RS485 to TCP/IP Ethernet Server Converter Module

### Detailed Error Log

None, unfortunately

simondrake, 2024-10-09:
@bryanajwells Yeah that looks to be the same. 

I think I've worked around my issue using helpers/scripts, was just going to leave it another day or two before confirming.

bryanajwells, 2024-10-09:
> @bryanajwells Yeah that looks to be the same.
> 
> I think I've worked around my issue using helpers/scripts, was just going to leave it another day or two before confirming.

It would be interesting to see your workaround as I'm being bitten by this one regularly. 
I t
Had thought it had gone away until I got the electric bill and say usage at 11pm which shouldn't have been there. 2330 would have been ok.

simondrake, 2024-10-09:
> It would be interesting to see your workaround as I'm being bitten by this one regularly.

Sure thing. I'll post it here and probably the linked thread a bit later.

wills106, 2024-10-12:
Closing down to keep conversation on #1029

closed 2024-10-12 by wills106