wills106/homeassistant-solax-modbus #1086: [FR]: 'Overview of request'

kprzeb, 2024-10-09:
### Description

Add suport inverter type X3-MIC/PRO SN MC80****

### Suggested Solution

Hello It is posible add suport Solax X3-MIC/PRO sn MC80**** mayby protocol is similar to other solax. 
Thenks

wills106, 2024-10-12:
What's the next two characters?

wills106, 2024-10-12:
I have released [2024.10.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.10.1) with a few variants of MC80...

kprzeb, 2024-10-12:
Hello most of the number is MC802TF319****

kprzeb, 2024-10-12:
[image]
Unfortunately, I think something is wrong. Is there anything I can do to help with development?

wills106, 2024-10-12:
Is this Inverter X3-MIC or X3-MIC-G2?

kprzeb, 2024-10-12:
This is his marking

wills106, 2024-10-12:
If you are trying to send an image. You can only upload it from the GitHub website. Replying by email doesn't upload an image.

wills106, 2024-10-12:
If it's wider than it's tall then it's Gen1
If it's taller than it's wide then it's Gen2

This is the Gen2 https://www.solaxpower.com/products/x3-mic/

kprzeb, 2024-10-12:
> If you are trying to send an image. You can only upload it from the GitHub website. Replying by email doesn't upload an image.

oh sorry I didn't know about it. This is his marking
[image]

wills106, 2024-10-12:
You are best looking at the shape.

kprzeb, 2024-10-12:
> If it's wider than it's tall then it's Gen1 If it's taller than it's wide then it's Gen2
> 
> This is the Gen2 https://www.solaxpower.com/products/x3-mic/

So its GEN 1
[image]

schaffnix84, 2024-10-16:
Gibt es hier schon eine Lösung? Seit dem Update 2024.10.1 sind bei mir beim Solax x3 mic gen2 auch keine Daten mehr vorhanden.

wills106, 2024-11-02:
@kprzeb Have a look at this comment: https://github.com/wills106/homeassistant-solax-modbus/issues/1108#issuecomment-2452933522

You could try changing `MU802T` to `MC802T` in both locations.

Also are you using the Pocket WiFi 2 to connect via Modbus or using the RS485 connection with a device like a Waveshare?
The Pocket WiFi 2 isn't [officially supported](https://homeassistant-solax-modbus.readthedocs.io/en/latest/compatible-adaptors/#solax-pocketwifi-2)