wills106/homeassistant-solax-modbus #1088: [FR]: 'setable Active Power Rate on Growatt (SPH)'

hgnirke, 2024-10-12:
### Description

In the case the battery is full and house consumption still to low to lower grid feed in or limit to 800W balcony solar maximum there is this regulator "Active Power Rate" in web settings @server.growatt.com. 

Looking in the modbus RTU documentation (attached) it  shows it is register 3 in a 0-100% Level

[growatt-inverter-modbus-rtu-protocolii-v120-english.pdf](https://github.com/user-attachments/files/17350209/growatt-inverter-modbus-rtu-protocolii-v120-english.pdf)

### Suggested Solution

something like:
 
    GrowattModbusSensorEntityDescription(
        name = "Active Power Rate",
        key = "active_power_rate",
        native_unit_of_measurement = PERCENTAGE,
        register = 3,
        allowedtypes = GEN2,
        entity_registry_enabled_default = False,
        icon = "mdi:mdi:dip-switch",
    ),

wills106, 2024-10-13:
I have just released [2024.10.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.10.2) with Active and Reactive Power Limits

hgnirke, 2024-10-13:
Just updated and checked. 
Works perfect as expected.
Thank you!

It's great to be a part of your project.
I really like it to give input that will be processed so fast.
If you would like me to test things in my setting or having new ideas regarding Growatt SPH feel free to contact.

wills106, 2024-10-21:
Closing down as complete.

If you find anything else that needs adding is feel free submit another FR

closed 2024-10-21 by wills106