wills106/homeassistant-solax-modbus #1090: [FR]: 'SOFAR setting SOC and other battery config'

chepreronius, 2024-10-13:
### Description

Completely missing configuration of SOC battery, safety limit, current A to the battery and discharge current A

### Suggested Solution

Add to the almost perfect plugin this missing options be nice. Without this I can't configure automatically charge discharge from spot prices. Thanks buddy, the reward will not pass you by.

cschlipf, 2024-11-17:
See #703

You can't set SoC. You may mean DoD (depth of discharge).

Sofar requires too many registers to be set along with many very critical registers. Therefore we decided to not pursue this anymore. As @anton4 stated above, you can set other values and use HA automations to achieve what you want.

cschlipf, 2024-11-17:
Duplicate of #703

kay0815, 2025-02-21:
...just for understanding. I also thought setting the DoD would be usefull, my use case would be to set the DoD to 35% if the solar forecast for the next day is week, and set it to 15% in the opposite case. Is there any way to achive that with the values that are setable via HA?

cschlipf, 2025-02-21:
I use the passive mode to disable/enable discharge. See the documentation: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-energy-storage-modes/#passive-mode

You just react on battery capacity change and in case you have a weak forecast, you disable discharge below 35%.

kay0815, 2025-02-22:
ok, thank you

cschlipf, 2025-03-13:
@wills106 feel free to close this duplicate

closed 2025-03-13 by wills106