wills106/homeassistant-solax-modbus #1102: [Bug]: MIC X3 GEN2 MC204T no data After Update 10.1

schaffnix84, 2024-10-17:
### Describe the bug

I use the mic x3 gen2 and the x3 hybrid g4. With version 2024.09.4 everything works. As soon as I update to 2024.10.1, I don't get any more data from the mic. HA restarts don't help either. What could be the reason? I deleted the mic and installed it new. No datas 

### Integration Version

2024.09.4

### Homeassistant core version

2024.10.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

MC204T

### Inverter firmware versions

DSP v1.117 ARM v1.113

### Connection Method

Waveshare

### Dongle firmware

Not used

### Detailed Error Log

No datas read

### Additional context

The Software 9.x Works fine

wills106, 2024-10-21:
Does it give any errors in the log?

schaffnix84, 2024-10-21:
Hello!

This ist, what I found about the Problem.

I made the Update an copied the Protokoll:

 

Home Assistant Core

2024-10-21 13:02:43.737 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ ((float(states('sensor.solaxg_measured_power'))) / 1000) *

ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ ((float(states('sensor.solaxg_measured_power'))) / 1000) *

homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ ((float(states('sensor.solaxg_measured_power'))) / 1000) *

2024-10-21 13:02:43.743 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ ((float(states('sensor.solaxg_measured_power'))) / 1000) *

}}' but no default was specified') while processing template 'Template<template=({{ ((float(states('sensor.solaxg_measured_power'))) / 1000) *

}}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.solaxg_arbeit_in_eur_pro_stunde'

2024-10-21 13:02:43.744 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (float(states('sensor.solax_total_yield')) + float(states('sensor.solaxg_total_yield'))) - (float(states('sensor.einspeisung_sm_gesamt'))) - (float(states('sensor.solax_eps_yield_total')))}}) renders=2>

ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (float(states('sensor.solax_total_yield')) + float(states('sensor.solaxg_total_yield'))) - (float(states('sensor.einspeisung_sm_gesamt'))) - (float(states('sensor.solax_eps_yield_total')))}}' but no default was specified

homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (float(states('sensor.solax_total_yield')) + float(states('sensor.solaxg_total_yield'))) - (float(states('sensor.einspeisung_sm_gesamt'))) - (float(states('sensor.solax_eps_yield_total')))}}' but no default was specified

2024-10-21 13:02:43.749 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ (float(states('sensor.solax_total_yield')) + float(states('sensor.solaxg_total_yield'))) - (float(states('sensor.einspeisung_sm_gesamt'))) - (float(states('sensor.solax_eps_yield_total')))}}' but no default was specified') while processing template 'Template<template=({{ (float(states('sensor.solax_total_yield')) + float(states('sensor.solaxg_total_yield'))) - (float(states('sensor.einspeisung_sm_gesamt'))) - (float(states('sensor.solax_eps_yield_total')))}}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.eigennutzung_2wr'

2024-10-21 13:02:43.775 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}) renders=2>

ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}' but no default was specified

homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}' but no default was specified

2024-10-21 13:02:43.782 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}' but no default was specified') while processing template 'Template<template=({{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.einsparung_solaxgarage_in_prozent'

2024-10-21 13:02:43.805 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ float(states('sensor.solaxg_today_s_yield')) + float(states('sensor.solax_today_s_yield')) }}) renders=2>

ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ float(states('sensor.solaxg_today_s_yield')) + float(states('sensor.solax_today_s_yield')) }}' but no default was specified

homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ float(states('sensor.solaxg_today_s_yield')) + float(states('sensor.solax_today_s_yield')) }}' but no default was specified

2024-10-21 13:02:43.809 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ float(states('sensor.solaxg_today_s_yield')) + float(states('sensor.solax_today_s_yield')) }}' but no default was specified') while processing template 'Template<template=({{ float(states('sensor.solaxg_today_s_yield')) + float(states('sensor.solax_today_s_yield')) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.today_s_yield_gesamt_pv'

[... cut ...] 0 * 100 }}' but no default was specified') while processing template 'Template<template=({{ float(states('sensor.einsparung_solaxg_in_euro_korrigiert')) / 3850 * 100 }}) renders=8>' for attribute '_attr_native_value' in entity 'sensor.einsparung_solaxgarage_in_prozent'

2024-10-21 13:02:44.753 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaXG: reading serial number from address 0x0 failed; other address may succeed

 

I hope it helps!

 

Thank you, for your great work!!

schaffnix84, 2024-10-21:
Edit:The Waveshare from the MIC does Not Flash with the blue light After the UpdateMit freundlichen Grüßen Johannes Schaffer

michaelbeelman, 2024-10-21:
I seem to have a similar issue with my X3-MIC with this info:

X3-MIC
door SolaX Power
Firmware: DSP v1.None ARM v1.None
Hardware: Gen1
Serienummer: MC502TF4172046

Since the update from 2024.05 to 2024.10.4 I'm no longer receiving data through the RS485>USB connection. Any clues?

wills106, 2024-11-02:
I added in a couple of new registers that are either missing, or not present due to firmware version.

https://github.com/wills106/homeassistant-solax-modbus/commit/cf216d70ac78d92a4f81a09fc06fb392d84d71b9

Try the black list method on one of the newer entities at a time, like in this PR https://github.com/wills106/homeassistant-solax-modbus/pull/1118/files but you shouldn't need it on everything. Probably "lock_state" or "Export Power Limit"

wills106, 2024-11-02:
Have you tried to manually read each of the holding registers here:
https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6114-L6160

Using a tool such as Modbus Poll, to see if any fail?
If they all return values you could try putting in a 'newblock' before each one in turn, to see if it allows the rest to be read:

https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6567

michaelbeelman, 2024-11-02:
Hi,

I just tried all these individual holding registers, but they all return illegal data address. I don’t think these registers exist on my model and firmware combination.

***@***.***:~$ mbpoll -a 1 -b 9600 -P none -t 4 -r 849 -c 1 -0 -1 /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

mbpoll 1.4-12 - FieldTalk(tm) Modbus(R) Master Simulator

Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll

This program comes with ABSOLUTELY NO WARRANTY.

This is free software, and you are welcome to redistribute it

under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU

Slave configuration...: address = [1]

                        start reference = 849, count = 1

Communication.........: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0,       9600-8N1

                        t/o 1.00 s, poll rate 1000 ms

Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...

Read output (holding) register failed: Illegal data address

***@***.***:~$ mbpoll -a 1 -b 9600 -P none -t 4 -r 850 -c 1 -0 -1 /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

mbpoll 1.4-12 - FieldTalk(tm) Modbus(R) Master Simulator

Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll

This program comes with ABSOLUTELY NO WARRANTY.

This is free software, and you are welcome to redistribute it

under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU

Slave configuration...: address = [1]

                        start reference = 850, count = 1

Communication.........: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0,       9600-8N1

                        t/o 1.00 s, poll rate 1000 ms

Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...

Read output (holding) register failed: Illegal data address

***@***.***:~$ mbpoll -a 1 -b 9600 -P none -t 4 -r 871 -c 1 -0 -1 /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

mbpoll 1.4-12 - FieldTalk(tm) Modbus(R) Master Simulator

Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll

This program comes with ABSOLUTELY NO WARRANTY.

This is free software, and you are welcome to redistribute it

under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU

Slave configuration...: address = [1]

                        start reference = 871, count = 1

Communication.........: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0,       9600-8N1

                        t/o 1.00 s, poll rate 1000 ms

Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...

Read output (holding) register failed: Illegal data address

***@***.***:~$ mbpoll -a 1 -b 9600 -P none -t 4 -r 881 -c 1 -0 -1 /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

mbpoll 1.4-12 - FieldTalk(tm) Modbus(R) Master Simulator

Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll

This program comes with ABSOLUTELY NO WARRANTY.

This is free software, and you are welcome to redistribute it

under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU

Slave configuration...: address = [1]

                        start reference = 881, count = 1

Communication.........: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0,       9600-8N1

                        t/o 1.00 s, poll rate 1000 ms

Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...

Read output (holding) register failed: Illegal data address

I did notice an issue in Home Assistant where the Energy Dashboard is showing a negative number for solar production since the upgrade.

***@***.*** ***@***.***

While the value for ‘SolaX Inverter Total Yield’ seems just fine:

***@***.***

Do you maybe have any idea what causes this issue and how to solve it?

Best regards,
Michael

https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6114-L6160

https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6567

schaffnix84, 2024-11-03:
Hallo!

 

I tried the blacklist with the 2 "lock_state" and "Export Power Limit".

 

    SolaxModbusNumberEntityDescription(

        name = "Active Power Limit",

        key = "active_power_limit",

        register = 0x638,

        fmt = "i",

        native_min_value = 0,

        native_max_value = 30000,

        native_step = 100,

        native_unit_of_measurement = UnitOfPower.WATT,

        device_class = NumberDeviceClass.POWER,

        allowedtypes = MIC | GEN2 | X3,

        blacklist=('MC210T',),

 

I tried it with all entities like in this example. But it did not work.

I am just a user of this Integration. I am sorry, but I do not understand enough to read Adresses by my selfe. I tried it, but I don´t ´know what I exactly to do. Sorry.

I restore my system to the Version of September ☹

 

Thank you. 

 

 <https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6114-L6160> https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6114-L6160

 <https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6567> https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L6567

schaffnix84, 2024-11-04:
I copied and inserted the complete Solax.py from the Blacklist. Unfortunately, it did not help. I can't really get along with the software for reading the modbus. Always only get timeout. Unfortunately, I don't know much about programming. I am now back to the old version. Everything works with that. Maybe someone who can help better will come in touch. I always work with complete instructions to set up something like that.

marianbezeg, 2024-11-07:
Hello,

I am detecting same issue for MC208Txxxxx 
There is only partial info and its so unstable.

DSP v1.119 ARM v1.117

`2024-11-07 16:42:15.761 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.245:502
2024-11-07 16:42:15.761 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.245:502]
2024-11-07 16:42:15.764 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-07 16:42:15.771 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.245', 502)
2024-11-07 16:42:15.771 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.245:502
2024-11-07 16:42:15.772 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.245:502]
2024-11-07 16:42:15.774 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-11-07 16:42:15.774 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-11-07 16:42:15.774 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-11-07 16:42:25.781 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.245', 502)
2024-11-07 16:42:25.781 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.245:502
2024-11-07 16:42:25.787 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.245', 502)
2024-11-07 16:42:25.787 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.245:502
2024-11-07 16:42:25.788 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 153, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.245:502]
2024-11-07 16:42:25.790 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
[... cut ...]
2024-11-07 19:07:10.280 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.111', 502)
2024-11-07 19:07:10.281 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.1.111:502
2024-11-07 19:07:10.483 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.111', 502)
2024-11-07 19:07:10.688 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.111', 502)`

wills106, 2024-11-10:
> ### Inverter firmware versions
> DSP v1.117 ARM v1.113

Is that what it actually reports or should it be DSP v1.17 & ARM v1.13?
I introduced some new sensors in 2024.10.x and it looks like your probably on an older Firmware version than is required for them to be present. Another user updated the firmware and the Integration is working again https://github.com/wills106/homeassistant-solax-modbus/pull/1118#issuecomment-2466155182

schaffnix84, 2024-11-10:
Oh man! This is the solution! The firmware update was the solution. I asked support for an update a few weeks ago, but it was never done.Thank you very much! Now I'm up to date everywhere and everything works! Thank you for your great work! Johannes

I introduced some new sensors in 2024.10.x and it looks like your probably on an older Firmware version than is required for them to be present. Another user updated the firmware and the Integration is working again #1118 (comment)

wills106, 2024-11-10:
I'll close this issue down then.

closed 2024-11-10 by wills106