wills106/homeassistant-solax-modbus #1103: [Bug]: Growatt - MIN 10000TL-X2 Unrecognized inverter type

StoneJunction, 2024-10-18:
### Describe the bug

I am connecting to the inverter with a Waveshare Modbus Interface however it takes a long time to setup and the logs are showing a "unrecognized inverter type" error. The serial number in the log is a strange set of non ascii characters:
[image]

### Integration Version

2024.10.3

### Homeassistant core version

2024.10.1

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AM1.0

### Connection Method

Waveshare RS485 TO POE ETH (B)

### Detailed Error Log

Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 11:32:55 (1 occurrences)
Last logged: 11:32:55

Error setting up entry GrowattViaModbus for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 188, in async_setup_entry
    await hub.async_init()
  File "/config/custom_components/solax_modbus/__init__.py", line 329, in async_init
    await asyncio.sleep(10)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 665, in sleep
    return await future
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7747
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 11:19:49 (78 occurrences)
Last logged: 11:32:48

unrecognized inverter type - serial number :

wills106, 2024-10-21:
Just added your Inverter in [2024.10.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.10.4)

StoneJunction, 2024-10-21:
Working great. Thanks

StoneJunction, 2024-10-21:
tested

closed 2024-10-21 by StoneJunction