wills106/homeassistant-solax-modbus #1107: Adding max_ac_charge_current as changeable

sceife, 2024-10-21:
### Description

I would like to use my existing solar system to charge excess energy into an external battery.
In order not to waste energy, it is important that only as much energy flows into the external battery as how much excess solar energy is available.
I would like to control it via an entity max_ac_charge_current that I can currently only see but cannot change.

### Suggested Solution

I can change the value directly on the Growatt SPF inverter. It would be great if it could be done directly in the app.
Changeable like the AC Input Mode, Buzzer or active_power_limit

flopp999, 2025-04-02:
@sceife can you post a picture of the setting from web or phone, and maybe I can have a look at it.

sceife, 2025-04-02:
[image]

Its a picture from Homeassiatant. At my Inverter i can change the value directly. After hat change i See the value also in Homeassiatant.

I hope that helps, thanks for support

flopp999, 2025-04-02:
Do you know what the name is in Growatt-world, at the inverter?
Is may be easier for me to find it in modbus

sceife, 2025-04-02:
[image]

I found this in the manuel of the 3500 ES.

flopp999, 2025-04-02:
Is the value in HA correct? If you set it to 24 in inverter, it will be 24 in HA?
What is the exact model you have of Growatt?

sceife, 2025-04-02:
Yes thats correct. Thats the reason i wrote the Initial message/wish to change the value.

I have the SPF 3500 ES.  The function is identically to the SPF 5000 ES

flopp999, 2025-04-02:
In this PDF you will see register 38 on page 11. 
38 FloatChargeCurr Float Charge Current W 0~80 0.1A 
https://watts247.com/manuals/gw/GrowattModBusProtocol.pdf
In Solax integration this is the code to read the value from your inverter
https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_growatt.py#L2283

What I don't understand is that the value must be divided by 10, according to the PDF but I don't see that in Solax. Maybe it is done somewhere that I can't find.

Can you send modbus commands to you inverter from a computer?
If you have maybe we can test to send a value and see if that change the current settings.

sceife, 2025-04-02:
If you have one comment i can try it tommorow.
At the moment the Inverter is offline because there is no Sun.

If you have a manipulated plugin.py i can change it too.

flopp999, 2025-04-04:
@sceife can you check what you get om modbus holding address 38?

sceife, 2025-04-04:
There was no time sorry, i try it tommorow.
I try to inform how to do

sceife, 2025-04-06:
[image]

At the Inverter i saw 14 Ampere and in Homeassiatant via modbus Register 38 also 14 A too.

So everthing works

closed 2025-09-18 by github-actions[bot]