wills106/homeassistant-solax-modbus #1109: [Bug]: 'S6-GR1P4K" not Accessable

fastsonic, 2024-10-23:
### Describe the bug

try to connect over HA to my S6-GR1P4K with an Elfin Wifi2Modbus Adapter.
My RHI5K-48 working fine with same Elfin Adapter. 
Also EVCC can directly connect to this Adapter and become Data.
Missing some in the Configuration ?

### Integration Version

2024.10.4

### Homeassistant core version

2024.10.0

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

 180203023

### Connection Method

RS485 over Elfin Wifi Adapter

### Detailed Error Log

2024-10-23 16:10:26.669 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: reading serial number from address 0x80ec failed; other address may succeed
2024-10-23 16:10:26.669 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: cannot find serial number, even not for other Inverter
2024-10-23 16:10:26.669 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis_2 inverter type - serial number : unknown
2024-10-23 16:10:36.769 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: reading serial number from address 0x80ec failed; other address may succeed
2024-10-23 16:10:36.769 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: cannot find serial number, even not for other Inverter
2024-10-23 16:10:36.769 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis_2 inverter type - serial number : unknown
2024-10-23 16:10:46.968 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: reading serial number from address 0x80ec failed; other address may succeed
2024-10-23 16:10:46.968 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: cannot find serial number, even not for other Inverter
2024-10-23 16:10:46.968 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis_2 inverter type - serial number : unknown
2024-10-23 16:10:57.168 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: reading serial number from address 0x80ec failed; other address may succeed
2024-10-23 16:10:57.168 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: cannot find serial number, even not for other Inverter
2024-10-23 16:10:57.169 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis_2 inverter type - serial number : unknown
2024-10-23 16:11:05.092 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.wolf_wp_l1_under_voltage_threshold: 75 (range 76.0 - 240.0)
2024-10-23 16:11:05.093 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.wolf_wp_l1_over_current_threshold: 65 (range 1.0 - 64.0)
2024-10-23 16:11:07.377 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: reading serial number from address 0x80ec failed; other address may succeed
2024-10-23 16:11:07.377 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis_2: cannot find serial number, even not for other Inverter
2024-10-23 16:11:07.377 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis_2 inverter type - serial number : unknown

wills106, 2024-11-03:
Have you tried solis_old?
That uses a different register set.

Regarding EVCC, if you have access to the Modbus documentation for that I could look at supporting it.

fastsonic, 2024-11-07:
there is a  normal documentation on the Github vor Evcc.
Maybe there are all Infos you need ?
https://github.com/evcc-io/evcc
Therea a 2 Typs Both working with the Inverter only the Flag for charge discharge battery is different. but for the Inverter not importand.

closed 2025-09-17 by github-actions[bot]