wills106/homeassistant-solax-modbus #1113: [Bug]: Home assistant doesn’t load Solax integration

iMatte81, 2024-10-25:
### Describe the bug

After updating Home assistant core on docker version 2024.10.3 s9lax integration stop to load.

### Integration Version

2024.10.4

### Homeassistant core version

2024.10.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

NA

### Inverter firmware versions

NA

### Connection Method

Solax pocket wifi

### Dongle firmware

NA

### Detailed Error Log

Registratore: homeassistant.setup
Fonte: setup.py:334 
Prima occorrenza: 22:28:44 (1 occorrenze) 
Ultima registrazione: 22:28:44

Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 60, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1019, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 60, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.12/site-packages/pymodbus/transaction.py)

Lekoglo, 2024-10-25:
Same problem here, since the Update it is no longer working / providing any data.

wills106, 2024-11-02:
What pyModbus version are you running?

https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#how-to-check-a-version-of-a-python-module-eg-pymodbus

iMatte81, 2024-11-02:
My version is 3.7.4

iMatte81, 2024-11-13:
Any news?

I have rewrite the comunication with nodered, I have only a problem, I can set the active power target but I'm not able to put the inverter in Enable Power Control. I don't find the modbus address of this register.

Could you give me an hand?

Thanks

wills106, 2024-11-16:
Does it work if you manually downgrade pyModbus to 3.6.8 or 3.6.9?

It might be difficult to support 3.7.x as HA is still on pre 3.7x version. Looks like they have changed how you import framers.

iMatte81, 2024-11-16:
It doesn't work but I think it is a my problem, because after downgrade to 3.6.9 after a reboot the version become always 3.7.4. When I'll have time I'll reinstall home assistant and I'll check if the problem is solved.

iMatte81, 2024-11-16:
No, all other modbus entity are read by nodered

wills106, 2024-11-16:
I wonder if Node-RED is pulling it to 3.7.4?

I'll try and get another instance of HA going and see if I can get it to work with 3.7.4

iMatte81, 2024-11-16:
> I wonder if Node-RED is pulling it to 3.7.4? (Edit: Node seems to be Java, so shouldn't cause it)
> 
> I'll try and get another instance of HA going and see if I can get it to work with 3.7.4

Home assistant runs in another docker container

bernie-gitmans, 2024-12-10:
I'm facing a similar issue, the interface is not working anymore

iMatte81, 2024-12-11:
> I'm facing a similar issue, the interface is not working anymore

At the end, I have used nodered and I have rewritten all the code using the same registers and the same names, in this way all the entities remain the same of the Solax integration.

wills106, 2024-12-23:
Closing down as #1172 is more visible.

closed 2024-12-23 by wills106

wills106, 2024-12-24:
Just released [2025.01.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.0b1) adding in support for pyModbus 3.7.4, but you need to be running HA 2025.1.0b

wills106, 2025-01-05:
Just released [2025.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.2)