wills106/homeassistant-solax-modbus #1119: [Bug]: X1-RetroFit mis categorised as a Gen4 inverter

JBayangosB, 2024-10-31:
### Describe the bug

Hi, 

According to Solax documentation my inverter is actually a Gen3

### Integration Version

2024.10.3

### Homeassistant core version

2024.10.3

### Inverter brand

Solax power

### Plugin used

plugin_solax.py

### Serial prefix

PRI502

### Inverter firmware versions

Firmware: DSP v1.42 ARM v1.35

### Connection Method

Ethernet

### Dongle firmware

n/a

wills106, 2024-11-02:
Does your Inverter look like:
https://www.solaxpower.com/products/x1-ac/

Or:
https://www.solaxpower.com/products/x1-retro-fit/

JBayangosB, 2024-11-03:
It looks more like this one: 

https://www.solaxpower.com/products/single-phase-hybrid/

Which also aligns with what I've observed with the app and when speaking to Evergen who were managing my battery.

[image]

In hindsight this makes sense because I never had success using the force discharge options as they would always reset when I applied them which I believe are gen4 features.

Lemme know if I can provide more details 👌

wills106, 2024-11-10:
Just noticed you mentioned your connecting over ethernet. Does the PRI have built in ethernet like the Hybrid Gen3 then? As the X1-AC didn't have built in ethernet, only RS485. So you would have to use something like a Waveshare or PocketWiFi when the PocketWiFi decides to work.

wills106, 2024-11-13:
Strange how it identifies as a X1-Hybrid-G3 am I correct in thinking you have no PV connectors on this Inverter then?

JBayangosB, 2024-11-14:
I had a look at it and it seems like it's possible for it to accept PV, but my system is tied to the AC. I do have a solar system but it is not directly connected to this inverter

wills106, 2024-11-16:
Just released [2024.11.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.11.2) with this change.

closed 2024-11-16 by wills106